aaronwmorris/indi-allsky #2008: Image Stops updating

Leogeddes, 2025-03-20:
Hi Aaron,

I've had an issue that has developed where atleast once per day the image stops updating and it takes most of the time, a few reboots to get it working again.

Error is below and support info below

System and indi all sky is up to date.

Cheers Leo

indi-allky log errors
```
2025-03-20T20:32:43.192800+13:00 allsky [ERROR] Capture-1-1128/MainThread libcamera._processMetadata() [436]: libcamera sensor black level key not found
2025-03-20T20:32:43.193006+13:00 allsky [ERROR] Capture-1-1128/MainThread capture.saferun() [543]: 8.0043s EXPOSURE RECEIVED IN 0.4048s.  POSSIBLE CAMERA PROBLEM.
2025-03-20T20:32:43.196239+13:00 allsky [ERROR] Image-1-1131/MainThread image.processImage() [334]: Frame not found: /tmp/tmp6fz8yxac.dng
2025-03-20T20:33:08.195605+13:00 allsky [ERROR] Capture-1-1128/MainThread libcamera._processMetadata() [334]: Metadata file not found: [Errno 2] No such file or directory: '/tmp/tmp86q6s_g4.json'
2025-03-20T20:33:08.195789+13:00 allsky [ERROR] Capture-1-1128/MainThread libcamera._processMetadata() [357]: libcamera camera temperature key not found
2025-03-20T20:33:08.195852+13:00 allsky [ERROR] Capture-1-1128/MainThread libcamera._processMetadata() [436]: libcamera sensor black level key not found
2025-03-20T20:33:08.195962+13:00 allsky [ERROR] Capture-1-1128/MainThread capture.saferun() [543]: 8.0043s EXPOSURE RECEIVED IN 0.4037s.  POSSIBLE CAMERA PROBLEM.
2025-03-20T20:33:08.198961+13:00 allsky [ERROR] Image-1-1131/MainThread image.processImage() [334]: Frame not found: /tmp/tmpduz4y19k.dng
2025-03-20T20:33:49.227639+13:00 allsky [ERROR] Capture-1-1121/MainThread libcamera._processMetadata() [334]: Metadata file not found: [Errno 2] No such file or directory: '/tmp/tmpi0lpqooc.json'
2025-03-20T20:33:49.227862+13:00 allsky [ERROR] Capture-1-1121/MainThread libcamera._processMetadata() [357]: libcamera camera temperature key not found
2025-03-20T20:33:49.227936+13:00 allsky [ERROR] Capture-1-1121/MainThread libcamera._processMetadata() [436]: libcamera sensor black level key not found
2025-03-20T20:33:49.230128+13:00 allsky [ERROR] Capture-1-1121/MainThread capture.saferun() [543]: 8.0043s EXPOSURE RECEIVED IN 2.2590s.  POSSIBLE CAMERA PROBLEM.
2025-03-20T20:33:49.254502+13:00 allsky [ERROR] Image-1-1123/MainThread image.processImage() [334]: Frame not found: /tmp/tmp_je9jkub.dng
2025-03-20T20:36:15.049189+13:00 allsky [ERROR] Capture-1-1121/MainThread libcamera._processMetadata() [334]: Metadata file not found: [Errno 2] No such file or directory: '/tmp/tmpj02e259s.json'
2025-03-20T20:36:15.049347+13:00 allsky [ERROR] Capture-1-1121/MainThread libcamera._processMetadata() [357]: libcamera camera temperature key not found
2025-03-20T20:36:15.049424+13:00 allsky [ERROR] Capture-1-1121/MainThread libcamera._processMetadata() [436]: libcamera sensor black level key not found
2025-03-20T20:36:15.049818+13:00 allsky [ERROR] Capture-1-1121/MainThread capture.saferun() [543]: 8.0043s EXPOSURE RECEIVED IN 0.4562s.  POSSIBLE CAMERA PROBLEM.
2025-03-20T20:36:15.052775+13:00 allsky [ERROR] Image-1-1123/MainThread image.processImage() [334]: Frame not found: /tmp/tmpxk2ee9fh.dng
2025-03-20T20:36:40.056018+13:00 allsky [ERROR] Capture-1-1121/MainThread libcamera._processMetadata() [334]: Metadata file not found: [Errno 2] No such file or directory: '/tmp/tmpoe2f7kgt.json'
2025-03-20T20:36:40.056163+13:00 allsky [ERROR] Capture-1-1121/MainThread libcamera._processMetadata() [357]: libcamera camera temperature key not found
2025-03-20T20:36:40.056220+13:00 allsky [ERROR] Capture-1-1121/MainThread libcamera._processMetadata() [436]: libcamera sensor black level key not found
2025-03-20T20:36:40.056556+13:00 allsky [ERROR] Capture-1-1121/MainThread capture.saferun() [543]: 8.0043s EXPOSURE RECEIVED IN 0.4563s.  POSSIBLE CAMERA PROBLEM.
2025-03-20T20:36:40.059687+13:00 allsky [ERROR] Image-1-1123/MainThread image.processImage() [334]: Frame not found: /tmp/tmpqda86_0c.dng
2025-03-20T20:37:05.072050+13:00 allsky [ERROR] Capture-1-1121/MainThread libcamera._processMetadata() [334]: Metadata file not found: [Errno 2] No such file or directory: '/tmp/tmpgno2rlua.json'
2025-03-20T20:37:05.072196+13:00 allsky [ERROR] Capture-1-1121/MainThread libcamera._processMetadata() [357]: libcamera camera temperature key not found
2025-03-20T20:37:05.072255+13:00 allsky [ERROR] Capture-1-1121/MainThread libcamera._processMetadata() [436]: libcamera sensor black level key not found
2025-03-20T20:37:05.072594+13:00 allsky [ERROR] Capture-1-1121/MainThread capture.saferun() [543]: 8.0043s EXPOSURE RECEIVED IN 0.4565s.  POSSIBLE CAMERA PROBLEM.
2025-03-20T20:37:05.075605+13:00 allsky [ERROR] Image-1-1123/MainThread image.processImage() [334]: Frame not found: /tmp/tmpcvkoktbj.dng
2025-03-20T20:40:53.396435+13:00 allsky [ERROR] Capture-1-1128/MainThread capture.saferun() [543]: 36.3832s EXPOSURE RECEIVED IN 10.6316s.  POSSIBLE CAMERA PROBLEM.
2025-03-20T20:40:53.960146+13:00 allsky [ERROR] Image-1-1131/MainThread moonOverlay.apply() [188]: Moon overlay X offset places moon outside image boundary
```

#################################
###     end support info      ###`

Leogeddes, 2025-03-20:
`/home/admin/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 3882912 kB

System: Raspberry Pi 4 Model B Rev 1.5

Linux allsky 6.6.74+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux
Hostname
allsky
customer.acklnzl1.pop.starlinkisp.net 

Time
Thu 20 Mar 20:56:01 NZDT 2025

System timezone
Pacific/Auckland

Uptime
 20:56:01 up 18 min,  1 user,  load average: 0.43, 0.55, 0.49

Memory
               total        used        free      shared  buff/cache   available
Mem:         3882912     1248408     2053664        6896      660360     2634504
Swap:        1048572           0     1048572

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
udev             1668152        0   1668152   0% /dev
tmpfs             388292     6088    382204   2% /run
/dev/mmcblk0p2  60844680 40438912  17295164  71% /
tmpfs            1941456      180   1941276   1% /dev/shm
tmpfs               5120       16      5104   1% /run/lock
tmpfs              76800        8     76792   1% /home/admin/allsky/tmp
/dev/mmcblk0p1    522230    77474    444756  15% /boot/firmware
tmpfs             388288       44    388244   1% /run/user/1000

systemctl runlevel
multi-user.target

sysctl info
vm.swappiness = 1

Thermal info
cpu-thermal
32128

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether d8:3a:dd:59:a2:e6 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.31/24 brd 192.168.1.255 scope global dynamic noprefixroute eth0
[... 1461 lines cut ...]
2025-03-20T20:37:05.075605+13:00 allsky [ERROR] Image-1-1123/MainThread image.processImage() [334]: Frame not found: /tmp/tmpcvkoktbj.dng
2025-03-20T20:40:53.396435+13:00 allsky [ERROR] Capture-1-1128/MainThread capture.saferun() [543]: 36.3832s EXPOSURE RECEIVED IN 10.6316s.  POSSIBLE CAMERA PROBLEM.
2025-03-20T20:40:53.960146+13:00 allsky [ERROR] Image-1-1131/MainThread moonOverlay.apply() [188]: Moon overlay X offset places moon outside image boundary
2025-03-20T20:41:02.902703+13:00 allsky [ERROR] Image-1-1131/MainThread moonOverlay.apply() [188]: Moon overlay X offset places moon outside image boundary
2025-03-20T20:42:00.906469+13:00 allsky [ERROR] Image-1-1131/MainThread moonOverlay.apply() [188]: Moon overlay X offset places moon outside image boundary
2025-03-20T20:42:32.605866+13:00 allsky [ERROR] Capture-1-1128/MainThread capture.saferun() [543]: 60.0000s EXPOSURE RECEIVED IN 39.3361s.  POSSIBLE CAMERA PROBLEM.
2025-03-20T20:42:40.221203+13:00 allsky [ERROR] Image-1-1131/MainThread moonOverlay.apply() [188]: Moon overlay X offset places moon outside image boundary
2025-03-20T20:43:43.657029+13:00 allsky [ERROR] Image-1-1131/MainThread moonOverlay.apply() [188]: Moon overlay X offset places moon outside image boundary
2025-03-20T20:44:54.318835+13:00 allsky [ERROR] Image-1-1131/MainThread moonOverlay.apply() [188]: Moon overlay X offset places moon outside image boundary
2025-03-20T20:45:56.893913+13:00 allsky [ERROR] Image-1-1131/MainThread moonOverlay.apply() [188]: Moon overlay X offset places moon outside image boundary
2025-03-20T20:46:59.963875+13:00 allsky [ERROR] Image-1-1131/MainThread moonOverlay.apply() [188]: Moon overlay X offset places moon outside image boundary
2025-03-20T20:48:03.232376+13:00 allsky [ERROR] Image-1-1131/MainThread moonOverlay.apply() [188]: Moon overlay X offset places moon outside image boundary
2025-03-20T20:49:06.298326+13:00 allsky [ERROR] Image-1-1131/MainThread moonOverlay.apply() [188]: Moon overlay X offset places moon outside image boundary
2025-03-20T20:50:09.008414+13:00 allsky [ERROR] Image-1-1131/MainThread moonOverlay.apply() [188]: Moon overlay X offset places moon outside image boundary
2025-03-20T20:51:11.837255+13:00 allsky [ERROR] Image-1-1131/MainThread moonOverlay.apply() [188]: Moon overlay X offset places moon outside image boundary
2025-03-20T20:52:14.609200+13:00 allsky [ERROR] Image-1-1131/MainThread moonOverlay.apply() [188]: Moon overlay X offset places moon outside image boundary
2025-03-20T20:53:20.999421+13:00 allsky [ERROR] Image-1-1131/MainThread moonOverlay.apply() [188]: Moon overlay X offset places moon outside image boundary
2025-03-20T20:54:23.916496+13:00 allsky [ERROR] Image-1-1131/MainThread moonOverlay.apply() [188]: Moon overlay X offset places moon outside image boundary
2025-03-20T20:55:30.263929+13:00 allsky [ERROR] Image-1-1131/MainThread moonOverlay.apply() [188]: Moon overlay X offset places moon outside image boundary
```

###########################h######
###     end support info      ###
#################################
`

aaronwmorris, 2025-03-20:
Intesresting.  You appear to be running the latest libcamera code, so that is not a problem.

Perhaps it is a hardware problem.  You should try to reseat the CSI cable for the camera.  Reseat the cable on both the SBC and the camera.

Leogeddes, 2025-03-20:
Thanks Aaron ill try to take it down this weekend and reseat the cable. I'll report back.

Cheers

Leogeddes, 2025-04-01:
Hey Aaron, reseating the cable did not resolve the issue. Cheers Leo

aaronwmorris, 2025-04-02:
This is a bit of a mystery to me.  These types of errors are common when files cannot be written to the /tmp filesystem, but you do not have a separate /tmp file system.

Leogeddes, 2025-04-02:
Maybe things are corrupt. I've had a few system errors I've had to resolve manually lately like apt source errors etc missing keys. Maybe I'll try a fresh install

Leogeddes, 2025-04-02:
Just finished a fresh install and libcamera-hello now produces results previously it was throwing a segmentation error. Reinstalling indi-allsky now.

Leogeddes, 2025-04-02:
Ok so I can confirm the sensor is working correctly via libcamera with the fresh install however, in indi allky I'm now receiving the following error.

2025-04-02T21:48:34.004646+13:00 allsky [INFO] Capture-1-996/MainThread capture.saferun() [308]: Exposure state: BUSY

Support info:
/home/admin/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 3888500 kB

System: Raspberry Pi 4 Model B Rev 1.5

Linux allsky 6.12.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux
Hostname
allsky
customer.acklnzl1.pop.starlinkisp.net customer.acklnzl1.pop.starlinkisp.net 

Time
Wed  2 Apr 21:51:16 NZDT 2025

System timezone
Pacific/Auckland

Uptime
 21:51:16 up 7 min,  1 user,  load average: 0.00, 0.09, 0.07

Memory
               total        used        free      shared  buff/cache   available
Mem:         3888500      965768     2511236        2000      475236     2922732
Swap:         524284           0      524284

Filesystems
Filesystem     1K-blocks    Used Available Use% Mounted on
udev             1673024       0   1673024   0% /dev
tmpfs             388852    1200    387652   1% /run
/dev/mmcblk0p2  60838900 9259652  48468684  17% /
tmpfs            1944248     176   1944072   1% /dev/shm
tmpfs               5120      16      5104   1% /run/lock
/dev/mmcblk0p1    522230   69420    452810  14% /boot/firmware
tmpfs             388848      44    388804   1% /run/user/1000

systemctl runlevel
multi-user.target

sysctl info
vm.swappiness = 60

Thermal info
cpu-thermal
30667

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
[... 1445 lines cut ...]
2025-04-02T21:20:24.152647+13:00 allsky [ERROR] Capture-1-979/MainThread libcamera.getCcdExposureStatus() [314]: libcamera-still error: b'rpicam-still: symbol lookup error: /lib/aarch64-linux-gnu/libcamera.so.0.3: undefined symbol: _ZN7libpisp22compute_optimal_strideER24pisp_image_format_config\n'
2025-04-02T21:20:24.152873+13:00 allsky [ERROR] Capture-1-979/MainThread libcamera._processMetadata() [334]: Metadata file not found: [Errno 2] No such file or directory: '/tmp/tmp8ld1nyuz.json'
2025-04-02T21:20:24.152959+13:00 allsky [ERROR] Capture-1-979/MainThread libcamera._processMetadata() [357]: libcamera camera temperature key not found
2025-04-02T21:20:24.153031+13:00 allsky [ERROR] Capture-1-979/MainThread libcamera._processMetadata() [371]: libcamera sensor AWB key not found
2025-04-02T21:20:24.153097+13:00 allsky [ERROR] Capture-1-979/MainThread libcamera._processMetadata() [436]: libcamera sensor black level key not found
2025-04-02T21:20:24.154492+13:00 allsky [ERROR] Image-1-981/MainThread image.processImage() [292]: Frame not found: /tmp/tmpwd13gysv.jpg
2025-04-02T21:24:47.225705+13:00 allsky [ERROR] Capture-1-993/MainThread capture._initialize() [1091]: CCD night gain above maximum, changing to 22
2025-04-02T21:24:50.226254+13:00 allsky [ERROR] Capture-1-993/MainThread capture._initialize() [1100]: CCD moon mode gain above maximum, changing to 22
2025-04-02T21:24:53.227400+13:00 allsky [ERROR] Capture-1-993/MainThread capture._initialize() [1105]: CCD day gain below minimum, changing to 1
2025-04-02T21:27:44.336620+13:00 allsky [ERROR] Capture-2-1035/MainThread capture._initialize() [1091]: CCD night gain above maximum, changing to 22
2025-04-02T21:27:47.337107+13:00 allsky [ERROR] Capture-2-1035/MainThread capture._initialize() [1100]: CCD moon mode gain above maximum, changing to 22
2025-04-02T21:27:50.338132+13:00 allsky [ERROR] Capture-2-1035/MainThread capture._initialize() [1105]: CCD day gain below minimum, changing to 1
2025-04-02T21:29:25.174753+13:00 allsky [ERROR] Capture-1-969/MainThread capture._initialize() [1091]: CCD night gain above maximum, changing to 22
2025-04-02T21:29:28.175684+13:00 allsky [ERROR] Capture-1-969/MainThread capture._initialize() [1100]: CCD moon mode gain above maximum, changing to 22
2025-04-02T21:29:31.176014+13:00 allsky [ERROR] Capture-1-969/MainThread capture._initialize() [1105]: CCD day gain below minimum, changing to 1
2025-04-02T21:37:19.739514+13:00 allsky [ERROR] Capture-1-966/MainThread capture._initialize() [1091]: CCD night gain above maximum, changing to 22
2025-04-02T21:37:22.740083+13:00 allsky [ERROR] Capture-1-966/MainThread capture._initialize() [1100]: CCD moon mode gain above maximum, changing to 22
2025-04-02T21:37:25.741149+13:00 allsky [ERROR] Capture-1-966/MainThread capture._initialize() [1105]: CCD day gain below minimum, changing to 1
2025-04-02T21:44:45.378959+13:00 allsky [ERROR] Capture-1-996/MainThread capture._initialize() [1100]: CCD moon mode gain above maximum, changing to 22
2025-04-02T21:44:48.379859+13:00 allsky [ERROR] Capture-1-996/MainThread capture._initialize() [1105]: CCD day gain below minimum, changing to 1
```

#################################
###     end support info      ###
#################################

Leogeddes, 2025-04-02:
Sorry Aaron I think night awb was the culprit for the camera busy issue. Will leave it running tonight and will give you an update tomorrow. But I think it's sorted. Cheers

Leogeddes, 2025-04-05:
Alright, so it ran fine for a day then I had other issues not related to indi allsky. Ive removed the sdcard, reinstalled with a new card and there has been no issues since. I think this was all caused by a failing sdcard causing corruption. Will look into an NVME at a later date. Cheers for your time Aaron.

aaronwmorris, 2025-04-05:
Great news!

closed 2025-04-24 by aaronwmorris