aaronwmorris/indi-allsky #2009: Startrails: Option to maskout the area surrounding the image circle

aaronwmorris, 2025-03-20:
Aaron, would it be possible to offer an option to maskout the area surrounding the image circle on timelapse images?

_Originally posted by @jimmauck in https://github.com/aaronwmorris/indi-allsky/discussions/2005#discussioncomment-12558461_

aaronwmorris, 2025-03-20:
@omeganon Just to mention this, you can use the `Logo Overlay` to mask parts of your image.  The name implies adding a logo, but you can use it as a mask, too.

omeganon, 2025-03-20:
I was thinking specifically for use during timelapse generation to only hide those dynamic text blocks (camera/weather/planets/etc info). Logo Overlay wouldn't be useful there would it?

aaronwmorris, 2025-04-03:
Merged #2044 with the image circle mask feature for the startrails and startrails timelapse.