aaronwmorris/indi-allsky #2019: Swap almost full, is it normal ?

JeyJeyC, 2025-03-24:
Hi :)

I use INDI Allsky on a Pi 5 - 8 Go of RAM + a SSD 1 To.
The last reboot was 17 days ago.
I use only INDI Allsky on this PI.

I have an alert on the WEB inerface who says the SWAP is over 90%.

Is it normal ?
Is there anything to check? (if yes, I don't know what / how to do it)

[image]

Thanks :)

aaronwmorris, 2025-03-24:
With the default linux settings, this is fairly common.  If you add this tuning and reboot, swap should remain lower...

```
echo "vm.swappiness = 1" | sudo tee /etc/sysctl.d/90-indi-allsky.conf

sudo sysctl --system
```

JeyJeyC, 2025-03-24:
Thank you.

What do you recommend for Indi, to do it or let it like that?

aaronwmorris, 2025-03-24:
I would definitely recommend this setting.  This is one of the optimizations that the setup.sh script will apply if you elect to apply them.

JeyJeyC, 2025-03-25:
It's strange, because de 2 times I updated INDI + re-execute the setup, I always applied optimisations.

So is the optimisation not working?

aaronwmorris, 2025-03-25:
Its possible your swap is so small that it is filling up quickly.  I believe the default is 256MB.

You could just disable swap entirely.

omeganon, 2025-03-25:
This swap use looks normal to me. The signal for bad swap use would be the graph above. Used would be very high and Cache would be very low or none.

Using memory as cache only happens when the OS is not under any memory pressure, and all the active memory needed by running programs is fully satisfied. It will then use that extra memory to 'cache' disk files for faster access. That cache is actively reclaimed as needed for running programs.

Things that are moved to swap in this scenario are those memory areas that are not seeing any active use at all and having them in active memory could actually degrade performance by reducing the amount of memory that could be used for that disk caching. The OS will do the right thing and move them from swap to active memory if they are actively accessed.

closed 2025-04-24 by aaronwmorris