aaronwmorris/indi-allsky #2030: Where exactly are mini-timelapses?

Antyrael, 2025-03-30:
In the Media pulldown menu is an option to view Mini-Timelapses.
I can't find any settings related to Mini-Timelapses, nor do I see any files that might indicate they're mini.
How do I configure these or why aren't there any?

(Almost) everything else works like a charm though, I love this AllSky implementation!

omeganon, 2025-03-30:
These are created by you on demand. They’re meant to create a shortened Timelapse around an event that you capture in an image. From Media -> Images, find an interesting event, then click the mini Timelapse button at the bottom. You can control the amount of time before and after the event to create the mini Timelapse. 

https://github.com/aaronwmorris/indi-allsky/wiki/Mini-Timelapses

Antyrael, 2025-03-30:
Ah, I see. Thank you for your explanation!

closed 2025-03-30 by Antyrael