aaronwmorris/indi-allsky #2031: Gallery feature request

omeganon, 2025-03-30:
Related to #2007, but relevant for overexposed or any other image, it would be great to be able to exclude images from the Timelapse generation directly from the Gallery view; I.e. having an exclude icon next to download and zoom. 

Currently, the fastest way to exclude images is to find them in the gallery, note the timestamp of the image, then go find it in Images to exclude.

aaronwmorris, 2025-03-30:
I have not had as much success customizing the Gallery view.  I am sure this is a possibility, I just have not figured out how to do it.

There has to be more than just the action of excluding, you also have to indicate when the exclusion is successful.  There really needs to be a way to remove the exclusion as well.

https://photoswipe.com/

omeganon, 2025-03-30:
Understood.

omeganon, 2025-03-31:
Awesome, thanks for the quick turnaround. I’ll update later today.

omeganon, 2025-04-04:
Do you want to keep this open for the positioning of the icon? I took a look but can't understand where that's defined.

aaronwmorris, 2025-04-04:
I do not understand the positioning problem either.

If this works I am good to close this out for now.

omeganon, 2025-04-04:
I can confirm it works as wanted.

closed 2025-04-04 by omeganon