aaronwmorris/indi-allsky #2037: Grainy/Dusty images with Neptune C IMX178

stepix-fpv, 2025-04-01:
Hello,
I have some kind of grainy/dusty results with image from my neptune C. Darks are not pplied here but the same is with dark applied, even if the dots became balck.
Any idea?!

stepix-fpv, 2025-04-01:
/home/stepix/indi-allsky/misc/support_info.sh: riga 62: attenzione: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 3882920 kB

System: Raspberry Pi 4 Model B Rev 1.1

Linux raspberrypi 6.6.74+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux
Hostname
raspberrypi
raspberrypi.homenet.telecomitalia.it 

Time
mar 1 apr 2025, 08:40:04, CEST

System timezone
Europe/Rome

Uptime
 08:40:04 up 16:16,  2 users,  load average: 0,83, 0,34, 0,17

Memory
               total        used        free      shared  buff/cache   available
Mem:         3882920     1300808     1233820      221236     1647004     2582112
Swap:        1048572      859764      188808

Filesystems
File system    1K-blocchi    Usati Disponib. Uso% Montato su
udev              1670404        0   1670404   0% /dev
tmpfs              388292     1316    386976   1% /run
/dev/mmcblk0p2  120213332 49701892  64385232  44% /
tmpfs             1941460      184   1941276   1% /dev/shm
tmpfs                5120       16      5104   1% /run/lock
/dev/mmcblk0p1     522230    70506    451724  14% /boot/firmware
tmpfs              388292       48    388244   1% /run/user/1000

sysctl info
vm.swappiness = 60

Thermal info
cpu-thermal
51608

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether dc:a6:32:2e:64:90 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether dc:a6:32:2e:64:91 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.142/24 brd 192.168.1.255 scope global dynamic noprefixroute wlan0
       valid_lft 13273sec preferred_lft 13273sec
    inet6 fe80::670d:484f:7065:8303/64 scope link noprefixroute 
[... 1324 lines cut ...]
    "CERT_BYPASS": true,
    "USERNAME": "",
    "PASSWORD": "REDACTED",
    "PASSWORD_E": "REDACTED",
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "AIRCRAFT_LABEL_TEMPLATE": "{id:s} {distance:0.1f}km {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:15,300 (Left)\n# anchor:la (Left Justified)\n# color:200,200,200\nAircraft"
  },
  "SATELLITE_TRACK": {
    "ENABLE": false,
    "DAYTIME_TRACK": false,
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "SAT_LABEL_TEMPLATE": "{title:s} {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:-15,200 (Right)\n# anchor:ra (Right Justified)\n# color:200,200,200\nSatellites"
  }
}

```
#################################
###     end support info      ###
#################################

aaronwmorris, 2025-04-01:
ASIFitsview automatically stretches the image, so that may be a bit exaggerated.

stepix-fpv, 2025-04-01:
It is true, but I have other darks from other camera and seems "better". I have to investigate more on this issue.

stepix-fpv, 2025-04-03:
Turned off the dome heater and get more tha 10°C reduction on the camera temperature.....situation now is better
[image]

closed 2025-08-24 by aaronwmorris