aaronwmorris/indi-allsky #2048: ftp no more working

pierrebilb, 2025-04-05:
Hi,
since a few days (march 29), ftp file transfert is no more working; here is the log:

```
Apr  5 07:19:57 orangepi3b [ERROR] MainProcess-1958/MainThread allsky._fileUploadWorkerStart() [573]: Upload worker exception:     raise ConnectionFailure(str(e)) from e
Apr  5 07:19:57 orangepi3b [ERROR] MainProcess-1958/MainThread allsky._fileUploadWorkerStart() [573]: Upload worker exception: indi_allsky.filetransfer.exceptions.ConnectionFailure: timed out
Apr  5 07:19:57 orangepi3b [ERROR] MainProcess-1958/MainThread allsky._fileUploadWorkerStart() [573]: Upload worker exception: 
Apr  5 07:19:57 orangepi3b [ERROR] MainProcess-1958/MainThread allsky._fileUploadWorkerStart() [573]: Upload worker exception: During handling of the above exception, another exception occurred:
Apr  5 07:19:57 orangepi3b [ERROR] MainProcess-1958/MainThread allsky._fileUploadWorkerStart() [573]: Upload worker exception: 
Apr  5 07:19:57 orangepi3b [ERROR] MainProcess-1958/MainThread allsky._fileUploadWorkerStart() [573]: Upload worker exception: Traceback (most recent call last):
Apr  5 07:19:57 orangepi3b [ERROR] MainProcess-1958/MainThread allsky._fileUploadWorkerStart() [573]: Upload worker exception:   File "/home/orangepi/indi-allsky/indi_allsky/uploader.py", line 108, in run
Apr  5 07:19:57 orangepi3b [ERROR] MainProcess-1958/MainThread allsky._fileUploadWorkerStart() [573]: Upload worker exception:     self.saferun()
Apr  5 07:19:57 orangepi3b [ERROR] MainProcess-1958/MainThread allsky._fileUploadWorkerStart() [573]: Upload worker exception:   File "/home/orangepi/indi-allsky/indi_allsky/uploader.py", line 139, in saferun
Apr  5 07:19:57 orangepi3b [ERROR] MainProcess-1958/MainThread allsky._fileUploadWorkerStart() [573]: Upload worker exception:     self.processUpload(u_dict)
Apr  5 07:19:57 orangepi3b [ERROR] MainProcess-1958/MainThread allsky._fileUploadWorkerStart() [573]: Upload worker exception:   File "/home/orangepi/indi-allsky/indi_allsky/uploader.py", line 432, in processUpload
Apr  5 07:19:57 orangepi3b [ERROR] MainProcess-1958/MainThread allsky._fileUploadWorkerStart() [573]: Upload worker exception:     client.close()
Apr  5 07:19:57 orangepi3b [ERROR] MainProcess-1958/MainThread allsky._fileUploadWorkerStart() [573]: Upload worker exception:   File "/home/orangepi/indi-allsky/indi_allsky/filetransfer/python_ftp.py", line 57, in close
Apr  5 07:19:57 orangepi3b [ERROR] MainProcess-1958/MainThread allsky._fileUploadWorkerStart() [573]: Upload worker exception:     self.client.quit()
Apr  5 07:19:57 orangepi3b [ERROR] MainProcess-1958/MainThread allsky._fileUploadWorkerStart() [573]: Upload worker exception:   File "/usr/lib/python3.11/ftplib.py", line 659, in quit
Apr  5 07:19:57 orangepi3b [ERROR] MainProcess-1958/MainThread allsky._fileUploadWorkerStart() [573]: Upload worker exception:     resp = self.voidcmd('QUIT')
Apr  5 07:19:57 orangepi3b [ERROR] MainProcess-1958/MainThread allsky._fileUploadWorkerStart() [573]: Upload worker exception:            ^^^^^^^^^^^^^^^^^^^^
Apr  5 07:19:57 orangepi3b [ERROR] MainProcess-1958/MainThread allsky._fileUploadWorkerStart() [573]: Upload worker exception:   File "/usr/lib/python3.11/ftplib.py", line 285, in voidcmd
Apr  5 07:19:57 orangepi3b [ERROR] MainProcess-1958/MainThread allsky._fileUploadWorkerStart() [573]: Upload worker exception:     self.putcmd(cmd)
Apr  5 07:19:57 orangepi3b [ERROR] MainProcess-1958/MainThread allsky._fileUploadWorkerStart() [573]: Upload worker exception:   File "/usr/lib/python3.11/ftplib.py", line 207, in putcmd
Apr  5 07:19:57 orangepi3b [ERROR] MainProcess-1958/MainThread allsky._fileUploadWorkerStart() [573]: Upload worker exception:     self.putline(line)
Apr  5 07:19:57 orangepi3b [ERROR] MainProcess-1958/MainThread allsky._fileUploadWorkerStart() [573]: Upload worker exception:   File "/usr/lib/python3.11/ftplib.py", line 202, in putline
Apr  5 07:19:57 orangepi3b [ERROR] MainProcess-1958/MainThread allsky._fileUploadWorkerStart() [573]: Upload worker exception:     self.sock.sendall(line.encode(self.encoding))
Apr  5 07:19:57 orangepi3b [ERROR] MainProcess-1958/MainThread allsky._fileUploadWorkerStart() [573]: Upload worker exception:     ^^^^^^^^^^^^^^^^^
Apr  5 07:19:57 orangepi3b [ERROR] MainProcess-1958/MainThread allsky._fileUploadWorkerStart() [573]: Upload worker exception: AttributeError: 'NoneType' object has no attribute 'sendall'
Apr  5 07:19:57 orangepi3b [ERROR] MainProcess-1958/MainThread allsky._fileUploadWorkerStart() [573]: Upload worker exception: 
```
I have no idea of what's going wrong as i didn't change anything about file transfert...
Thanks for helping me..

aaronwmorris, 2025-04-05:
Your connections are timing out.  My guess is your hosting provider has blocked your traffic.

pierrebilb, 2025-04-06:
Hi,
my internet provider changed my IP last week after 8 years with fixed IP!
Thanks a lot Aaron and very grateful for all your work and help!
With best regards
Pierre

closed 2025-04-06 by pierrebilb