aaronwmorris/indi-allsky #2052: Sensor settings not stored

ecalandrini, 2025-04-06:
Hi! I'm trying to add some sensor to the raspberry. when i save their settings in the sensor page and reload I loose everyting.

aaronwmorris, 2025-04-06:
Can you give me a little more context?  Can you post your Support Info?

ecalandrini, 2025-04-07:
here you have the Support info.

the issue is when I try to add the sensors. i have the MLX90614 and the
BME280, if I add them and I save and reload, they disappear.

thanks for your help,
Eugenio

> Message ID: ***@***.***>
>

/home/heugy/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 3888500 kB

System: Raspberry Pi 4 Model B Rev 1.5

Linux raspberrypi 6.12.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux
Hostname
raspberrypi
raspberrypi.home 

Time
Mon  7 Apr 21:48:24 CEST 2025

System timezone
Europe/Madrid

Uptime
 21:48:24 up 8 min,  0 user,  load average: 2.22, 1.10, 0.53

Memory
               total        used        free      shared  buff/cache   available
Mem:         3888500     1104472     2210812       15612      662668     2784028
Swap:         204796           0      204796

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
udev             1673676        0   1673676   0% /dev
tmpfs             388852    12396    376456   4% /run
/dev/mmcblk0p2  30109280 15156448  13404544  54% /
tmpfs            1944248      184   1944064   1% /dev/shm
tmpfs               5120       16      5104   1% /run/lock
tmpfs            1024000        0   1024000   0% /var/ramfs
/dev/mmcblk0p1    522230    67058    455172  13% /boot/firmware
tmpfs             388848        4    388844   1% /run/user/1001
tmpfs             388848        4    388844   1% /run/user/1000

systemctl runlevel
multi-user.target

sysctl info
vm.swappiness = 1

Thermal info
cpu-thermal
36511

system python: Python 3.11.2

[... 1222 lines cut ...]
      0,
      0
    ]
  },
  "UPLOAD_WORKERS": 2,
  "FILETRANSFER": {
    "CLASSNAME": "pycurl_sftp",
    "HOST": "",
    "PORT": 0,
    "USERNAME": "",
    "PASSWORD": "REDACTED",
    "PASSWORD_E": "REDACTED",
    "PRIVATE_KEY": "",
    "PUBLIC_KEY": "",
    "CONNECT_TIMEOUT": 10,
    "TIMEOUT": 60,
    "CERT_BYPASS": true,
    "REMOTE_IMAGE_NAME": "image.{ext}",
    "REMOTE_PANORAMA_NAME": "panorama.{ext}",
    "REMOTE_IMAGE_FOLDER": "/home/allsky/upload/allsky",
    "REMOTE_PANORAMA_FOLDER": "/home/allsky/upload/allsky",
    "REMOTE_RAW_FOLDER": "/home/allsky/upload/allsky/export",
    "REMOTE_FITS_FOLDER": "/home/allsky/upload/allsky/fits",
    "REMOTE_METADATA_NAME": "latest_metadata.json",
    "REMOTE_METADATA_FOLDER": "/

aaronwmorris, 2025-04-07:
The config is cutoff, but there is a possibility...  Can you show the output of the following?

```
source virtualenv/indi-allsky/bin/activate

./config list
```

The config that is loaded is the latest one.  If you have a config that is dated in the future, that can prevent a saved config from being used.

ecalandrini, 2025-04-08:
here there is the config list result. is there a way to remove configs?

thanks for your help!

+----+----------------------------+---------+------------+---------------------------------------+
| ID |        Create Date         | User ID |   Level    |                  Note                 |
+----+----------------------------+---------+------------+---------------------------------------+
| 62 | 2025-04-06 17:58:06.683634 |    1    | 20250216.0 | Load config: /tmp/tmp.4ddBlUVMr7.json |
| 61 | 2025-04-06 17:28:03.910909 |    2    | 20250216.0 |                                       |
| 60 | 2025-04-06 17:24:21.510377 |    2    | 20250216.0 |                                       |
| 59 | 2025-04-06 09:39:46.249843 |    1    | 20250216.0 | Load config: /tmp/tmp.4uCprNaXMX.json |
| 58 | 2025-04-06 09:28:32.830142 |    2    | 20250216.0 |                                       |
| 57 | 2025-04-06 09:23:48.101109 |    2    | 20250216.0 |                                       |
| 56 | 2025-04-06 09:00:47.047282 |    2    | 20250216.0 |                                       |
| 55 | 2025-04-06 08:45:21.475696 |    2    | 20250216.0 |                                       |
| 54 | 2025-04-05 22:43:45.727619 |    2    | 20250216.0 |                                       |
| 53 | 2025-04-05 22:10:50.416098 |    2    | 20250216.0 |                                       |
| 52 | 2025-04-05 21:28:00.583424 |    2    | 20250216.0 |                                       |
| 51 | 2025-04-05 21:23:15.797868 |    2    | 20250216.0 |                                       |
| 50 | 2025-04-05 21:19:27.309309 |    2    | 20250216.0 |                                       |
| 49 | 2025-04-05 20:54:30.460537 |    1    | 20250216.0 | Load config: /tmp/tmp.9f66XExbz9.json |
| 48 | 2025-04-05 20:27:04.570324 |    1    | 20250216.0 | Load config: /tmp/tmp.OfMhJ4rGQX.json |
| 47 | 2024-10-10 19:30:49.363761 |    2    | 20240827.0 |                                       |
| 46 | 2024-10-10 19:26:14.514898 |    2    | 20240827.0 |                                       |
| 45 | 2024-10-10 17:05:04.757142 |    2    | 20240827.0 |                                       |
| 44 | 2024-10-10 16:53:36.675825 |    2    | 20240827.0 |                                       |
| 43 | 2024-10-10 16:53:32.107058 |    2    | 20240827.0 |                                       |
| 42 | 2024-10-10 16:51:57.236432 |    2    | 20240827.0 |                                       |
| 41 | 2024-10-10 16:28:36.891748 |    2    | 20240827.0 |                                       |
| 40 | 2024-10-10 16:21:18.994039 |    2    | 20240827.0 |                                       |
| 39 | 2024-10-06 16:11:07.638589 |    2    | 20240827.0 |                                       |
| 38 | 2024-10-06 16:02:39.119297 |    2    | 20240827.0 |                                       |
| 37 | 2024-10-06 15:29:12.504207 |    2    | 20240827.0 |                                       |
| 36 | 2024-10-06 15:28:29.274957 |    2    | 20240827.0 |                                       |
| 35 | 2024-10-06 15:23:38.303070 |    2    | 20240827.0 |                                       |
| 34 | 2024-10-06 15:04:45.401697 |    2    | 20240827.0 |                                       |
| 33 | 2024-10-06 15:02:31.820386 |    2    | 20240827.0 |                                       |
| 32 | 2024-10-06 15:01:41.160568 |    2    | 20240827.0 |                                       |
| 31 | 2024-10-06 15:00:11.025355 |    2    | 20240827.0 |                                       |
| 30 | 2024-10-06 14:58:48.620006 |    2    | 20240827.0 |                                       |
| 29 | 2024-10-06 00:12:01.003056 |    2    | 20240827.0 |                                       |
| 28 | 2024-10-05 20:07:58.886062 |    2    | 20240827.0 |                                       |
| 27 | 2024-10-05 19:55:12.395460 |    2    | 20240827.0 |                                       |
| 26 | 2024-10-05 19:41:54.953204 |    2    | 20240827.0 |                                       |
| 25 | 2024-10-04 16:33:41.471211 |    2    | 20240827.0 |                                       |
| 24 | 2024-10-04 16:02:09.949350 |    2    | 20240827.0 |                                       |
| 23 | 2024-09-30 21:26:17.748057 |    2    | 20240827.0 |                                       |
| 22 | 2024-09-30 21:24:12.812880 |    2    | 20240827.0 |                                       |
| 21 | 2024-09-30 21:20:19.386716 |    2    | 20240827.0 |                                       |
| 20 | 2024-09-30 21:18:56.428380 |    2    | 20240827.0 |                                       |
| 19 | 2024-09-30 21:16:37.801977 |    1    | 20240827.0 | Load config: /tmp/tmp.kjBx706kPE.json |
| 18 | 2024-09-30 19:54:36.763622 |    2    | 20240827.0 |                                       |
| 17 | 2024-09-30 19:44:50.306917 |    2    | 20240827.0 |                                       |
| 16 | 2024-09-30 19:44:48.270535 |    2    | 20240827.0 |                                       |
| 15 | 2024-09-30 19:42:56.314447 |    2    | 20240827.0 |                                       |
| 14 | 2024-09-30 19:40:00.783930 |    2    | 20240827.0 |                                       |
| 13 | 2024-09-30 19:20:40.062991 |    1    | 20240827.0 | Load config: /tmp/tmp.VrSjqyRqLD.json |
| 12 | 2024-09-30 18:07:58.679615 |    2    | 20240806.0 |                                       |
[... cut ...]
| 5  | 2024-08-13 15:41:05.447343 |    2    | 20240806.0 |                                       |
| 4  | 2024-08-13 15:40:26.811127 |    1    | 20240806.0 | Load config: /tmp/tmp.3ugYEh27xJ.json |
| 3  | 2024-08-13 15:10:45.264724 |    1    | 20240806.0 | Load config: /tmp/tmp.Tjwcs5X5G2.json |
| 2  | 2024-08-12 21:34:45.246237 |    1    | 20240806.0 | Load config: /tmp/tmp.2EVuWjHQHU.json |
| 1  | 2024-08-12 21:27:32.152318 |    1    | 20240806.0 |             Initial config            |
+----+----------------------------+---------+------------+---------------------------------------+

aaronwmorris, 2025-04-08:
There does not seem to be an issue with the config dates.  I did just add the ability to delete individual configs in #2058 .

I cannot think of any reason why this would happen.  Does `./misc/sensor_test.py` return valid data for the sensors?

ecalandrini, 2025-04-09:
I don't get data from sensors, only from openweather API:
[WARNING]: Initializing [OpenWeather] OpenWeather API Sensor
[WARNING]: No sensor B - Initializing sensor simulator
[WARNING]: No sensor C - Initializing sensor simulator
[WARNING]: No sensor D - Initializing sensor simulator
[INFO]: [OpenWeather] OpenWeather API - temp: 13.8c, humidity: 79%,
pressure: 1019.0hPa, clouds: 0%
[INFO]: Sensor 0: {'dew_point': 10.229345614341428, 'frost_point':
8.447485336998568, 'heat_index': 25.53397458405475, 'wind_degrees': 0,
'data': (13.81, 13.31, 79, 1019, 0, 0.51, 0.0, 0.0, 0.0,
10.229345614341428)}
[INFO]: Sensor 1: {'data': ()}
[INFO]: Sensor 2: {'data': ()}
[INFO]: Sensor 3: {'data': ()}

Should I try to remove all the config files and start from scratch?

Thanks

aaronwmorris, 2025-04-09:
Yeah, you might try flushing the configs and starting a new config from scratch.  This wiki page can guide you through that:

https://github.com/aaronwmorris/indi-allsky/wiki/Reset-configuration

ecalandrini, 2025-04-09:
It didn't work. I had to set several time my location and the camera type.
Still when I insert a sensor and I do save and reload, the sensor disappear.

ecalandrini, 2025-04-09:
> On a hunch, can you re-run setup and reconfigure the indiserver without
> indi-gpsd? See if that might help.

Now in config->sensor I only see openweather but in config/admin I can
choose the sensor measurements. but in the chart I see 0s for the sensors,
I can't see even the name of the panel.
[image: image.png]

aaronwmorris, 2025-04-10:
You could try to remove and recreate your virtualenv.

```
# make sure virutalenv is deactivated
deactivate

rm -fR virtualenv/indi-allsky

pip cache purge
```

# rerun setup.sh

ecalandrini, 2025-04-11:
> rerun setup.sh and reboot

at the end i decided to flash the sd card and install everything from
scratch. now it's working like a charm, sorry for bothering you.

aaronwmorris, 2025-04-11:
It is no big deal.  Feel free to reopen if it happens again!

closed 2025-04-11 by aaronwmorris