aaronwmorris/indi-allsky #2053: ERROR: These packages do not match the hashes from the requirements file

fishscene, 2025-04-07:
cd ~
git clone https://github.com/aaronwmorris/indi-allsky.git
cd indi-allsky/
./misc/build_indi.sh ## Run this before setup.sh
./setup.sh

Select: "indi"
Optional Python modules? Yes
GPIO Python modules? No

```
Collecting google-cloud-storage (from -r /home/pi/indi-allsky/requirements/requirements_optional.txt (line 5))
  Using cached https://www.piwheels.org/simple/google-cloud-storage/google_cloud_storage-3.1.0-py2.py3-none-any.whl (174 kB)
Collecting google-api-python-client (from -r /home/pi/indi-allsky/requirements/requirements_optional.txt (line 6))
  Downloading https://www.piwheels.org/simple/google-api-python-client/google_api_python_client-2.166.0-py2.py3-none-any.whl (13.2 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╺━━━━ 11.6/13.2 MB 196.9 kB/s eta 0:00:09
ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    google-api-python-client from https://www.piwheels.org/simple/google-api-python-client/google_api_python_client-2.166.0-py2.py3-none-any.whl#sha256=39e697f6ecff5c429af4c7a1341d068ac9a1baf04f0060124a08a5f52d0c4ee9 (from -r /home/pi/indi-allsky/requirements/requirements_optional.txt (line 6)):
        Expected sha256 39e697f6ecff5c429af4c7a1341d068ac9a1baf04f0060124a08a5f52d0c4ee9
             Got        f3b648a20a312c7e0a3bbbd2dba9a0ad9a30235d32e72658ef94450e2edf8c7e
```

Every time I run "./setup.sh", it errors out at a different part of the download, around the 10-11MB mark. Here's another sample:

```
Downloading https://www.piwheels.org/simple/google-api-python-client/google_api_python_client-2.166.0-py2.py3-none-any.whl (13.2 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╺━━━━━━ 10.9/13.2 MB 245.0 kB/s eta 0:00:10
ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes.
```

..any idea on how to get past this?

fishscene, 2025-04-07:
NVM... after 6 or 7 attempted builds, it seems to have just... worked?
Probably just a transient issue, but I'm hoping it doesn't indicate the rPi is having trouble talking over WiFi.
Sorry to bug y'all.

aaronwmorris, 2025-04-07:
These errors happen from time to time.  It might be a bad mirror.  Sometimes it has been reported due to a poor wifi signal.

Retrying, like you did, will eventually clear up the problem.