aaronwmorris/indi-allsky #2066: 2 versions of the same camera after upgrading to 2025.04.01

oldjiberjaber, 2025-04-12:
I seem to have 2 Indi cameras now after the upgrade to 04.01? 
One is has the historic images (ZWO CCD ASI462MC) and one is new images (ZWO ASI462MC). 
Is there anyway I can merge the two together so I retain the historic images and new under one camera? 

I can find some ASI config files in .indi but it's not obvious what I need to edit?

Cheers

---------------------------------------------------------------
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 11
Arch: x86_64
Bits: 64

CPUs: 2
Memory: 5006320 kB

System: Generic PC

Linux Shed 5.10.0-34-amd64 #1 SMP Debian 5.10.234-1 (2025-02-24) x86_64 GNU/Linux
Hostname
Shed.lan
Shed 

Time
Sat 12 Apr 11:05:57 BST 2025

System timezone
Europe/London

Uptime
 11:05:57 up 23:06,  0 users,  load average: 0.45, 0.46, 0.30

Memory
               total        used        free      shared  buff/cache   available
Mem:         5006320     1457188      144252      519288     3404880     2778748
Swap:         999420        3096      996324

Filesystems
Filesystem     1K-blocks      Used Available Use% Mounted on
udev             2445536         0   2445536   0% /dev
tmpfs             500632     52540    448092  11% /run
/dev/sdb1      238269176 171898692  54194256  77% /
tmpfs            2503160    462120   2041040  19% /dev/shm
tmpfs               5120         0      5120   0% /run/lock
/dev/sda1      960301264 744773200 166673680  82% /mnt/tempdata
tmpfs             500632        76    500556   1% /run/user/1001

systemctl runlevel
multi-user.target

sysctl info
vm.swappiness = 1

Thermal info

system python: Python 3.9.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
[... 1411 lines cut ...]
Apr 12 11:01:21 Shed [ERROR] Image-1-3144/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 12 11:01:36 Shed [ERROR] Image-1-3144/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 12 11:01:51 Shed [ERROR] Image-1-3144/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 12 11:02:06 Shed [ERROR] Image-1-3144/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 12 11:02:21 Shed [ERROR] Image-1-3144/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 12 11:02:36 Shed [ERROR] Image-1-3144/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 12 11:02:51 Shed [ERROR] Image-1-3144/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 12 11:03:06 Shed [ERROR] Image-1-3144/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 12 11:03:22 Shed [ERROR] Image-1-3144/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 12 11:03:37 Shed [ERROR] Image-1-3144/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 12 11:03:52 Shed [ERROR] Image-1-3144/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 12 11:04:07 Shed [ERROR] Image-1-3144/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 12 11:04:22 Shed [ERROR] Image-1-3144/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 12 11:04:37 Shed [ERROR] Image-1-3144/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 12 11:04:52 Shed [ERROR] Image-1-3144/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 12 11:05:07 Shed [ERROR] Image-1-3144/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 12 11:05:22 Shed [ERROR] Image-1-3144/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 12 11:05:37 Shed [ERROR] Image-1-3144/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 12 11:05:52 Shed [ERROR] Image-1-3144/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 12 11:06:07 Shed [ERROR] Image-1-3144/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
```

#################################
###     end support info      ###
#################################

aaronwmorris, 2025-04-13:
This sometimes happens when upgrading indilib.  indi-allsky uses the camera name as the identifier and sometimes the name is changed between releases.

Merged #2069 which includes a new script I wrote just for this situation.  This script is `./misc/merge_cameras.py`.  It basically just changes the camera for all of the images and videos and adds the new name to the existing camera so that it is detected properly.

You just need to figure out the ID numbers of the camera.  They are probably `1` and `2`, but you never know.  Once you have the cameras merged, you can hide the camera that is no longer needed.

https://github.com/aaronwmorris/indi-allsky/wiki/SQL-queries

oldjiberjaber, 2025-04-13:
Fab - thanks, that worked though I should have realised to identify the cameras I could use the first SQL query in your link rather than just guess :)

closed 2025-04-13 by oldjiberjaber