aaronwmorris/indi-allsky #2096: File transfer to network storage

AstroAndy46, 2025-04-26:
Hi

I'm looking for some help as I am getting the file transfer upload wrong.

I have a network drive which is accessed from a storage server on my PC all my drives are network drives mapped back to the storage server, the issue I'm having is I'm getting a error massage saying "host name invalid".

I have put the host name and tried the ip address in but I still get the same error message, also for the username and password would that have to be the server details or my PC login details as I log on to my server?

I would be very grateful for any help as I still very new to this.

Kind regards
Andy

aaronwmorris, 2025-04-27:
Do you receive the hostname invalid error when you try to enter the hostname in the filetransfer config?  What hostname are you trying to enter?

AstroAndy46, 2025-04-27:
Good morning Aaron

I'm trying to put the information on the file transfer tab, yes when I go to save the config that is when the error comes up.
The host names I have tried are;

https://"server name"
https://"server ip address"
//"server name"
//"server ip address"
"server name"
"server ip address"

I have used different protocols.

Here is a small part of the error log.

Apr 27 07:54:55 AllSkyCameraPi4 [INFO] Image-5-121915/MainThread miscDb.addThumbnail() [1170]: Adding thumbnail to DB: /var/www/html/allsky/images/ccd_284686a9-32af-481f-b8a2-45a247a74ce7/exposures/20250427/day/27_07/thumbnails/9a779832-90c6-4052-a03d-4279ead87cdf.jpg

Apr 27 07:54:55 AllSkyCameraPi4 [ERROR] MainProcess-803/Upload-10 uploader.processUpload() [230]: Hostname not set for file transfer

Apr 27 07:54:58 AllSkyCameraPi4 [INFO] Capture-5-121913/MainThread capture.saferun() [307]: Camera last ready: 4.8s

Apr 27 07:54:58 AllSkyCameraPi4 [INFO] Capture-5-121913/MainThread capture.saferun() [308]: Exposure state: OK

Apr 27 07:54:58 AllSkyCameraPi4 [INFO] Capture-5-121913/MainThread capture.detectNight() [1489]: Sun alt: 18.5, Moon alt: 22.7, phase 0.5%

Apr 27 07:54:58 AllSkyCameraPi4 [INFO] Capture-5-121913/MainThread indi.getCcdTemperature() [896]: Camera temperature: 21.4

Apr 27 07:55:02 AllSkyCameraPi4 [INFO] Sensor-5-121917/MainThread tempSensorSht3x.update() [26]: [SHT31-D] SHT3x - temp: 15.9c, humidity: 59.3%

Apr 27 07:55:02 AllSkyCameraPi4 [INFO] Sensor-5-121917/MainThread lightSensorTsl2591.update() [34]: [TSL2591] TSL2591 - lux: 237.7824, visible: 1507401, ir: 23, full: 1507424

Apr 27 07:55:02 AllSkyCameraPi4 [INFO] Sensor-5-121917/MainThread sensor.saferun() [194]: Dew Point: 8.0, Frost Point: 5.9, Heat Index: 27.4

Apr 27 07:55:02 AllSkyCameraPi4 [INFO] Sensor-5-121917/MainThread sensor.saferun() [198]: Sensor SQM: 6.64311

Apr 27 07:55:02 AllSkyCameraPi4 [INFO] Sensor-5-121917/MainThread sensor.check_fan_thresholds() [605]: Fan threshold current: 41.4, target: 35.0, delta: 6.4 (100%)

Apr 27 07:55:07 AllSkyCameraPi4 [INFO] Capture-5-121913/MainThread capture.shoot() [1684]: Taking 0.00029637 s exposure (gain 0)

Apr 27 07:55:07 AllSkyCameraPi4 [INFO] Capture-5-121913/MainThread indi.set_number() [1363]: Setting Duration (s) = 0.00029636648832820356

Apr 27 07:55:07 AllSkyCameraPi4 [INFO] Capture-5-121913/MainThread capture.saferun() [659]: Total time since last exposure 15.0313 s

Apr 27 07:55:08 AllSkyCameraPi4 [INFO] Capture-5-121913/MainThread capture.saferun() [540]: Exposure received in 1.4151 s (1.4148)

Apr 27 07:55:08 AllSkyCameraPi4 [ERROR] Image-5-121915/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon

I hope that helps, thank you for your help with this Aaron.

Regards
Andy

AstroAndy46, 2025-04-27:
Good evening, Aaron

I have used the ip address for the server with my login details but it still does not, here is what I have got in the feilds.

My server name is CVV-STORAGE1

ip address of server "???.???.???.???"
username"username"
password"password"

This is the correct folder names;
/Astrophotography Folder 1/All Sky Camera Recorded Data/INDIallsky/TEST/RAW{day_date:%d%m%Y}/{timeofday:s}/{ts:%H}

The above has errors so I have tried this below;
/Astrophotography_Folder_1/All_Sky_Camera_Recorded_Data/INDIallsky/TEST/RAW{day_date:%d%m%Y}/{timeofday:s}/{ts:%H}

My storage server manages my folder which are on a Dell Powervalt MD3220i & MD3200i

I hope this makes sense.

Thank you for your help Aaron

Regards
Andy

aaronwmorris, 2025-04-27:
I would just use the IP address instead of the hostname.

Once you save the hostname/IP, username, password, and protocol details, there is a very new script I have in the misc folder to easily test file transfers... `./misc/upload_test.py filetransfer"  This will allow you to test the file transfer without having to start the whole program.

AstroAndy46, 2025-04-28:
Hi Aaron

First of all thank you for all the help you are giving me.

I have tried the test script and this is what it has returned;

[image]

Where am I going wrong with this?

Regards
Andy

aaronwmorris, 2025-04-28:
No worries!  Can you post your Support Info here?  I need to see your file transfer related configuration.

AstroAndy46, 2025-04-28:
Hi

Where do I find that?

aaronwmorris, 2025-04-28:
It will be under Tools -> Support Info

AstroAndy46, 2025-04-28:
Hi

Please see below;

/home/pi/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 11
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 7999812 kB

System: Raspberry Pi 4 Model B Rev 1.4

Linux AllSkyCameraPi4 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
Hostname
AllSkyCameraPi4

Time
Mon 28 Apr 15:24:09 BST 2025

System timezone
Europe/London

Uptime
 15:24:09 up 8 days,  3:38,  0 users,  load average: 0.39, 0.41, 0.49

Memory
               total        used        free      shared  buff/cache   available
Mem:         7999812     1433036      437388      194420     6129388     6247696
Swap:         102396           0      102396

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/root       61324388 49356636   9442304  84% /
devtmpfs         3735680        0   3735680   0% /dev
tmpfs            3999904      180   3999724   1% /dev/shm
tmpfs            1599964   177480   1422484  12% /run
tmpfs               5120        4      5116   1% /run/lock
/dev/mmcblk0p1    261106    31380    229726  13% /boot
tmpfs             799980        0    799980   0% /run/user/1000

systemctl runlevel
multi-user.target

sysctl info
vm.swappiness = 1

Thermal info
cpu-thermal
67679

system python: Python 3.9.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
[... 1455 lines cut ...]
Apr 28 15:22:05 AllSkyCameraPi4 [ERROR] MainProcess-803/Upload-38 uploader.processUpload() [474]: Connection failure: Connection timed out after 10004 milliseconds
Apr 28 15:22:08 AllSkyCameraPi4 [ERROR] Image-19-156526/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 28 15:22:20 AllSkyCameraPi4 [ERROR] MainProcess-803/Upload-37 uploader.processUpload() [474]: Connection failure: Connection timed out after 10006 milliseconds
Apr 28 15:22:23 AllSkyCameraPi4 [ERROR] Image-19-156526/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 28 15:22:35 AllSkyCameraPi4 [ERROR] MainProcess-803/Upload-38 uploader.processUpload() [474]: Connection failure: Connection timed out after 10006 milliseconds
Apr 28 15:22:38 AllSkyCameraPi4 [ERROR] Image-19-156526/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 28 15:22:50 AllSkyCameraPi4 [ERROR] MainProcess-803/Upload-37 uploader.processUpload() [474]: Connection failure: Connection timed out after 10006 milliseconds
Apr 28 15:22:53 AllSkyCameraPi4 [ERROR] Image-19-156526/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 28 15:23:05 AllSkyCameraPi4 [ERROR] MainProcess-803/Upload-38 uploader.processUpload() [474]: Connection failure: Connection timed out after 10002 milliseconds
Apr 28 15:23:08 AllSkyCameraPi4 [ERROR] Image-19-156526/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 28 15:23:20 AllSkyCameraPi4 [ERROR] MainProcess-803/Upload-37 uploader.processUpload() [474]: Connection failure: Connection timed out after 10003 milliseconds
Apr 28 15:23:23 AllSkyCameraPi4 [ERROR] Image-19-156526/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 28 15:23:35 AllSkyCameraPi4 [ERROR] MainProcess-803/Upload-38 uploader.processUpload() [474]: Connection failure: Connection timed out after 10006 milliseconds
Apr 28 15:23:38 AllSkyCameraPi4 [ERROR] Image-19-156526/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 28 15:23:50 AllSkyCameraPi4 [ERROR] MainProcess-803/Upload-37 uploader.processUpload() [474]: Connection failure: Connection timed out after 10006 milliseconds
Apr 28 15:23:53 AllSkyCameraPi4 [ERROR] Image-19-156526/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 28 15:24:05 AllSkyCameraPi4 [ERROR] MainProcess-803/Upload-38 uploader.processUpload() [474]: Connection failure: Connection timed out after 10005 milliseconds
Apr 28 15:24:08 AllSkyCameraPi4 [ERROR] Image-19-156526/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
```

#################################
###     end support info      ###
#################################

Thank you for all time, which part of the world are you in.

aaronwmorris, 2025-04-28:
> Hi Aaron
> 
> First of all thank you for all the help you are giving me.
> 
> I have tried the test script and this is what it has returned;
> 
> [image]
> 
> Where am I going wrong with this?
> 
> Regards Andy

I believe this just means indi-allsky was unable to make the SFTP connection to 192.168.70.4.  Can you test connecting via SFTP on the command line?

```
ssh Administrator@192.168.70.4
```

aaronwmorris, 2025-04-28:
I think the problem is the port is blocked for SSH/SFTP transfers.

AstroAndy46, 2025-04-28:
How do I open port 22?

aaronwmorris, 2025-04-28:
I am assuming the storage machine is a Windows device.  I am guessing you would need to update the windows firewall settings to allow port 22 from your local network.

I have also merged #2098 which will allow spaces in folder names for the file transfers.

AstroAndy46, 2025-04-28:
Hi

I have added that line into the forms.py

I have set port 22 on the server which is running WSS 2016 but still no luck, I'm thinking on installing OpenSSH and see if that a difference.

I'll let you know if it works.

AstroAndy46, 2025-04-28:
Hi Aaron

I have managed to talk between the indi-allsky and the server but now there is a folder issue, I'll have a look tomorrow as it is getting late, I'll send you a message if I cannot get it working tomorrow.

AstroAndy46, 2025-04-29:
Good morning, Aaron

Hope all is well.

I'm still getting an issue but now it is URL rejected: Malformed input to a URL function, I have attached the log below;

/home/pi/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 11
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 7999812 kB

System: Raspberry Pi 4 Model B Rev 1.4

Linux AllSkyCameraPi4 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
Hostname
AllSkyCameraPi4

Time
Tue 29 Apr 08:35:36 BST 2025

System timezone
Europe/London

Uptime
 08:35:36 up 8 days, 20:49,  1 user,  load average: 0.38, 0.44, 0.39

Memory
               total        used        free      shared  buff/cache   available
Mem:         7999812      890996     1874228      182604     5234588     6801552
Swap:         102396           0      102396

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/root       61324388 49348196   9450744  84% /
devtmpfs         3735680        0   3735680   0% /dev
tmpfs            3999904      168   3999736   1% /dev/shm
tmpfs            1599964   165680   1434284  11% /run
tmpfs               5120        4      5116   1% /run/lock
/dev/mmcblk0p1    261106    31380    229726  13% /boot
tmpfs             799980        0    799980   0% /run/user/1000

systemctl runlevel
multi-user.target

sysctl info
vm.swappiness = 1

Thermal info
cpu-thermal
41868

system python: Python 3.9.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
[... 1458 lines cut ...]
Apr 29 08:32:15 AllSkyCameraPi4 [ERROR] Image-29-170723/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 29 08:32:17 AllSkyCameraPi4 [ERROR] MainProcess-803/Upload-57 uploader.processUpload() [474]: Connection failure: URL rejected: Malformed input to a URL function
Apr 29 08:32:31 AllSkyCameraPi4 [ERROR] Image-29-170723/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 29 08:32:32 AllSkyCameraPi4 [ERROR] MainProcess-803/Upload-58 uploader.processUpload() [474]: Connection failure: URL rejected: Malformed input to a URL function
Apr 29 08:32:46 AllSkyCameraPi4 [ERROR] Image-29-170723/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 29 08:32:47 AllSkyCameraPi4 [ERROR] MainProcess-803/Upload-57 uploader.processUpload() [474]: Connection failure: URL rejected: Malformed input to a URL function
Apr 29 08:33:01 AllSkyCameraPi4 [ERROR] Image-29-170723/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 29 08:33:02 AllSkyCameraPi4 [ERROR] MainProcess-803/Upload-58 uploader.processUpload() [474]: Connection failure: URL rejected: Malformed input to a URL function
Apr 29 08:33:16 AllSkyCameraPi4 [ERROR] Image-29-170723/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 29 08:33:18 AllSkyCameraPi4 [ERROR] MainProcess-803/Upload-57 uploader.processUpload() [474]: Connection failure: URL rejected: Malformed input to a URL function
Apr 29 08:33:31 AllSkyCameraPi4 [ERROR] Image-29-170723/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 29 08:33:32 AllSkyCameraPi4 [ERROR] MainProcess-803/Upload-58 uploader.processUpload() [474]: Connection failure: URL rejected: Malformed input to a URL function
Apr 29 08:33:46 AllSkyCameraPi4 [ERROR] Image-29-170723/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 29 08:33:47 AllSkyCameraPi4 [ERROR] MainProcess-803/Upload-57 uploader.processUpload() [474]: Connection failure: URL rejected: Malformed input to a URL function
Apr 29 08:34:01 AllSkyCameraPi4 [ERROR] Image-29-170723/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 29 08:34:02 AllSkyCameraPi4 [ERROR] MainProcess-803/Upload-58 uploader.processUpload() [474]: Connection failure: URL rejected: Malformed input to a URL function
Apr 29 08:34:16 AllSkyCameraPi4 [ERROR] Image-29-170723/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 29 08:34:17 AllSkyCameraPi4 [ERROR] MainProcess-803/Upload-57 uploader.processUpload() [474]: Connection failure: URL rejected: Malformed input to a URL function
Apr 29 08:34:58 AllSkyCameraPi4 [ERROR] Image-30-170787/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
Apr 29 08:35:05 AllSkyCameraPi4 [ERROR] MainProcess-803/Upload-60 uploader.processUpload() [474]: Connection failure: URL rejected: Malformed input to a URL function
```

#################################
###     end support info      ###
#################################

aaronwmorris, 2025-04-29:
Merged #2100 for fixes to support spaces in the paths.  I should have tested this before I released it, but it should be working now.

AstroAndy46, 2025-04-30:
Good morning, Aaron

Thank you for sending that over last night.

I'm out for all of today so I'll make the changes to the files tomorrow and let you know what happens.

AstroAndy46, 2025-05-01:
Good morning, Aaron

I have made the changes you have said and the transfer happens with out any failed transfer warning but there is a warning that does come up but it does say transfer complete.

I have attached the log as the images are not transferring to the folder with I can not understand why, please can you help.

/home/pi/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 11
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 7999812 kB

System: Raspberry Pi 4 Model B Rev 1.4

Linux AllSkyCameraPi4 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
Hostname
AllSkyCameraPi4

Time
Thu  1 May 09:49:25 BST 2025

System timezone
Europe/London

Uptime
 09:49:25 up 12:08,  0 users,  load average: 0.36, 0.46, 0.49

Memory
               total        used        free      shared  buff/cache   available
Mem:         7999812      879896     3882480      167180     3237436     6828072
Swap:         102396           0      102396

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/root       61324388 51566572   7232368  88% /
devtmpfs         3735680        0   3735680   0% /dev
tmpfs            3999904      180   3999724   1% /dev/shm
tmpfs            1599964   150248   1449716  10% /run
tmpfs               5120        4      5116   1% /run/lock
/dev/mmcblk0p1    261106    31380    229726  13% /boot
tmpfs             799980        0    799980   0% /run/user/1000

systemctl runlevel
multi-user.target

sysctl info
vm.swappiness = 1

Thermal info
cpu-thermal
48686

system python: Python 3.9.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
[... 1467 lines cut ...]
May  1 09:47:05 AllSkyCameraPi4 [ERROR] Image-10-10578/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
May  1 09:47:07 AllSkyCameraPi4 [WARNING] MainProcess-814/Upload-20 pycurl_sftp.put() [180]: PycURL quoted commands encountered an error (safe to ignore)
May  1 09:47:20 AllSkyCameraPi4 [ERROR] Image-10-10578/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
May  1 09:47:22 AllSkyCameraPi4 [WARNING] MainProcess-814/Upload-19 pycurl_sftp.put() [180]: PycURL quoted commands encountered an error (safe to ignore)
May  1 09:47:35 AllSkyCameraPi4 [ERROR] Image-10-10578/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
May  1 09:47:37 AllSkyCameraPi4 [WARNING] MainProcess-814/Upload-20 pycurl_sftp.put() [180]: PycURL quoted commands encountered an error (safe to ignore)
May  1 09:47:49 AllSkyCameraPi4 [ERROR] Image-10-10578/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
May  1 09:47:52 AllSkyCameraPi4 [WARNING] MainProcess-814/Upload-19 pycurl_sftp.put() [180]: PycURL quoted commands encountered an error (safe to ignore)
May  1 09:48:05 AllSkyCameraPi4 [ERROR] Image-10-10578/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
May  1 09:48:07 AllSkyCameraPi4 [WARNING] MainProcess-814/Upload-20 pycurl_sftp.put() [180]: PycURL quoted commands encountered an error (safe to ignore)
May  1 09:48:20 AllSkyCameraPi4 [ERROR] Image-10-10578/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
May  1 09:48:22 AllSkyCameraPi4 [WARNING] MainProcess-814/Upload-19 pycurl_sftp.put() [180]: PycURL quoted commands encountered an error (safe to ignore)
May  1 09:48:35 AllSkyCameraPi4 [ERROR] Image-10-10578/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
May  1 09:48:37 AllSkyCameraPi4 [WARNING] MainProcess-814/Upload-20 pycurl_sftp.put() [180]: PycURL quoted commands encountered an error (safe to ignore)
May  1 09:48:50 AllSkyCameraPi4 [ERROR] Image-10-10578/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
May  1 09:48:52 AllSkyCameraPi4 [WARNING] MainProcess-814/Upload-19 pycurl_sftp.put() [180]: PycURL quoted commands encountered an error (safe to ignore)
May  1 09:49:05 AllSkyCameraPi4 [ERROR] Image-10-10578/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
May  1 09:49:07 AllSkyCameraPi4 [WARNING] MainProcess-814/Upload-20 pycurl_sftp.put() [180]: PycURL quoted commands encountered an error (safe to ignore)
May  1 09:49:20 AllSkyCameraPi4 [ERROR] Image-10-10578/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
May  1 09:49:22 AllSkyCameraPi4 [WARNING] MainProcess-814/Upload-19 pycurl_sftp.put() [180]: PycURL quoted commands encountered an error (safe to ignore)
```

#################################
###     end support info      ###
#################################

aaronwmorris, 2025-05-01:
Its hard to say what the problem is, but I suspect it might be due to using relative paths for the file transfer paths.  From my notes, the pycurl sftp did not support relative paths.  Make sure the paths all begin with `/` to specify the fully qualified path.

AstroAndy46, 2025-05-02:
Good morning, Aaron

Hope all is well.

I'm still stick, my network path to my folder is;
\\192.168.70.4\Astrophotography Folder 1\All Sky Camera Recorded Data\INDIallsky\Test\Images

The path address in INDI-allsky is;
sftp://192.168.70.4/Astrophotography%20Folder%201/All%20Sky%20Camera%20Recorded%20Data/INDIallsky/Test/Images

I have attached the log file below,

/home/pi/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 11
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 7999812 kB

System: Raspberry Pi 4 Model B Rev 1.4

Linux AllSkyCameraPi4 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
Hostname
AllSkyCameraPi4

Time
Fri  2 May 09:50:48 BST 2025

System timezone
Europe/London

Uptime
 09:50:48 up 1 day, 12:09,  0 users,  load average: 0.54, 0.74, 0.55

Memory
               total        used        free      shared  buff/cache   available
Mem:         7999812      877132     1880780      183568     5241900     6814452
Swap:         102396           0      102396

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/root       61324388 51776604   7022336  89% /
devtmpfs         3735680        0   3735680   0% /dev
tmpfs            3999904      176   3999728   1% /dev/shm
tmpfs            1599964   166632   1433332  11% /run
tmpfs               5120        4      5116   1% /run/lock
/dev/mmcblk0p1    261106    31380    229726  13% /boot
tmpfs             799980        0    799980   0% /run/user/1000

systemctl runlevel
multi-user.target

sysctl info
vm.swappiness = 1

Thermal info
cpu-thermal
48686

system python: Python 3.9.2

indiserver: /usr/local/bin/indiserver

[... 1481 lines cut ...]
May  2 09:49:19 AllSkyCameraPi4 [WARNING] MainProcess-814/Upload-52 pycurl_sftp.put() [180]: PycURL quoted commands encountered an error (safe to ignore)
May  2 09:49:32 AllSkyCameraPi4 [ERROR] Image-26-30538/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
May  2 09:49:34 AllSkyCameraPi4 [WARNING] MainProcess-814/Upload-51 pycurl_sftp.put() [180]: PycURL quoted commands encountered an error (safe to ignore)
May  2 09:49:47 AllSkyCameraPi4 [ERROR] Image-26-30538/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
May  2 09:49:50 AllSkyCameraPi4 [WARNING] MainProcess-814/Upload-52 pycurl_sftp.put() [180]: PycURL quoted commands encountered an error (safe to ignore)
May  2 09:50:02 AllSkyCameraPi4 [ERROR] Image-26-30538/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
May  2 09:50:05 AllSkyCameraPi4 [WARNING] MainProcess-814/Upload-51 pycurl_sftp.put() [180]: PycURL quoted commands encountered an error (safe to ignore)
May  2 09:50:17 AllSkyCameraPi4 [ERROR] Image-26-30538/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
May  2 09:50:20 AllSkyCameraPi4 [WARNING] MainProcess-814/Upload-52 pycurl_sftp.put() [180]: PycURL quoted commands encountered an error (safe to ignore)
May  2 09:50:32 AllSkyCameraPi4 [ERROR] Image-26-30538/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
May  2 09:50:35 AllSkyCameraPi4 [WARNING] MainProcess-814/Upload-51 pycurl_sftp.put() [180]: PycURL quoted commands encountered an error (safe to ignore)
May  2 09:50:47 AllSkyCameraPi4 [ERROR] Image-26-30538/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
May  2 09:50:50 AllSkyCameraPi4 [WARNING] MainProcess-814/Upload-52 pycurl_sftp.put() [180]: PycURL quoted commands encountered an error (safe to ignore)
```

#################################
###     end support info      ###
#################################

I really can not think where I'm going wrong.

Thank you Aaron for all your time with this problem.

Regards
Andy

AstroAndy46, 2025-05-02:
I have tried the /. and still no luck, here is the log.

/home/pi/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 11
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 7999812 kB

System: Raspberry Pi 4 Model B Rev 1.4

Linux AllSkyCameraPi4 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
Hostname
AllSkyCameraPi4

Time
Fri  2 May 15:34:17 BST 2025

System timezone
Europe/London

Uptime
 15:34:17 up 1 day, 17:53,  1 user,  load average: 0.43, 0.60, 0.47

Memory
               total        used        free      shared  buff/cache   available
Mem:         7999812      653592     1086904      168148     6259316     7053412
Swap:         102396           0      102396

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/root       61324388 52764144   6034796  90% /
devtmpfs         3735680        0   3735680   0% /dev
tmpfs            3999904      172   3999732   1% /dev/shm
tmpfs            1599964   151216   1448748  10% /run
tmpfs               5120        4      5116   1% /run/lock
/dev/mmcblk0p1    261106    31380    229726  13% /boot
tmpfs             799980        0    799980   0% /run/user/1000

systemctl runlevel
multi-user.target

sysctl info
vm.swappiness = 1

Thermal info
cpu-thermal
63296

system python: Python 3.9.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether dc:a6:32:d7:f6:56 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
[... 1464 lines cut ...]
May  2 15:31:16 AllSkyCameraPi4 [WARNING] MainProcess-814/Upload-52 pycurl_sftp.put() [180]: PycURL quoted commands encountered an error (safe to ignore)
May  2 15:31:29 AllSkyCameraPi4 [ERROR] Image-26-30538/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
May  2 15:31:33 AllSkyCameraPi4 [WARNING] MainProcess-814/Upload-51 pycurl_sftp.put() [180]: PycURL quoted commands encountered an error (safe to ignore)
May  2 15:31:44 AllSkyCameraPi4 [ERROR] Image-26-30538/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
May  2 15:31:47 AllSkyCameraPi4 [WARNING] MainProcess-814/Upload-52 pycurl_sftp.put() [180]: PycURL quoted commands encountered an error (safe to ignore)
May  2 15:31:59 AllSkyCameraPi4 [ERROR] Image-26-30538/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
May  2 15:32:02 AllSkyCameraPi4 [WARNING] MainProcess-814/Upload-51 pycurl_sftp.put() [180]: PycURL quoted commands encountered an error (safe to ignore)
May  2 15:32:14 AllSkyCameraPi4 [ERROR] Image-26-30538/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
May  2 15:32:17 AllSkyCameraPi4 [WARNING] MainProcess-814/Upload-52 pycurl_sftp.put() [180]: PycURL quoted commands encountered an error (safe to ignore)
May  2 15:32:29 AllSkyCameraPi4 [ERROR] Image-26-30538/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
May  2 15:32:32 AllSkyCameraPi4 [WARNING] MainProcess-814/Upload-51 pycurl_sftp.put() [180]: PycURL quoted commands encountered an error (safe to ignore)
May  2 15:32:44 AllSkyCameraPi4 [ERROR] Image-26-30538/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
May  2 15:32:47 AllSkyCameraPi4 [WARNING] MainProcess-814/Upload-52 pycurl_sftp.put() [180]: PycURL quoted commands encountered an error (safe to ignore)
May  2 15:32:59 AllSkyCameraPi4 [ERROR] Image-26-30538/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
May  2 15:33:02 AllSkyCameraPi4 [WARNING] MainProcess-814/Upload-51 pycurl_sftp.put() [180]: PycURL quoted commands encountered an error (safe to ignore)
May  2 15:33:14 AllSkyCameraPi4 [ERROR] Image-26-30538/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
May  2 15:33:17 AllSkyCameraPi4 [WARNING] MainProcess-814/Upload-52 pycurl_sftp.put() [180]: PycURL quoted commands encountered an error (safe to ignore)
May  2 15:33:29 AllSkyCameraPi4 [ERROR] Image-26-30538/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
May  2 15:33:32 AllSkyCameraPi4 [WARNING] MainProcess-814/Upload-51 pycurl_sftp.put() [180]: PycURL quoted commands encountered an error (safe to ignore)
May  2 15:34:17 AllSkyCameraPi4 [ERROR] Image-27-35663/MainThread processing.update_astrometric_data() [2102]: HST next pass error: that satellite seems to stay always below your horizon
```

#################################
###     end support info      ###
#################################

aaronwmorris, 2025-05-02:
I ran some tests using a Windows 11 SFTP server.  I was only able to make this work by using file paths that started with the drive path, such as `C:/Foo Bar/upload`.

In the case of Windows, you should NOT start the remote folder with `/`.  I apologize for that recommendation, I made some assumptions.

AstroAndy46, 2025-05-02:
Good evening, Aaron

It is now working, thank you for all help with this issue.

Just one last thing, do you think the Hubble data will work as the readings never change.

Hope you have a good Bank Holiday weekend.

Andy

aaronwmorris, 2025-05-03:
Great news!

As for the Hubble, that is not all that unusual.  The error you are seeing is due to the Hubble not being above the horizon for you for the next 24 hours.  I believe Hubble's orbit keeps it closer to the equator so there are extended periods where it is not visible at higher latitudes.

AstroAndy46, 2025-05-03:
Good morning, Aaron

I understand I might take it remove  it and just have relative data for my location.

Have a good weekend and once again thank you for all you help.

Regards
Andy

closed 2025-05-03 by AstroAndy46