aaronwmorris/indi-allsky #2112: Sensor TSL2591

marco050969, 2025-05-08:
Hi Aaron,

After Update to last version my Sensor don´t works with Gain Setting over 25(default). 
Before Update it runs with Gain 428 (high) perfect...
See Error:  
2025-05-08T19:26:32.083682+02:00 Testi [ERROR] Sensor-9-75225/MainThread sensor.update_sensors() [549]: SensorReadException: Overflow reading light channels!, Try to reduce the gain of#012 the sensor using adafruit_tsl2591.GAIN_LOW
Need you more Information? 

Cheers

Marco

marco050969, 2025-05-08:
I don’t have change anything, because i don’t know, who are change this to adjusted integration settings..in previous build it runs

aaronwmorris, 2025-05-08:
There have been no changes to the code regarding this.  The module has not had an updates that I can see.  Is it possible your sky is brighter lately?

marco050969, 2025-05-08:
Nope…