aaronwmorris/indi-allsky #2117: Session stops short for some unknown reason

ikeysolomon, 2025-05-12:
Hi Aaron, 

              Please see attached txt file for support info dump. I am experiencing my session stopping short for an unknown reason? I am saving to USB and there is still plenty of room on the USB. However, looking at the graph for the USB under file systems it seems to have dropped the USB? 

[image]

I started this morning and it dropped out around 1700 h I think? I am in Australia. 

It did this last night, I woke up this morning expecting to see the transition from night to day but it stopped short, so I ran today during the day to try and replicate. 

I am using: 

- a PlayerOne IMX178 Neptune C.
- The USB was a 64GB cruzer blade FAT32 format. 
-  a rpi5 so there should be plenty of resources available? 

Your help would be most appreciated. 

Kind Regards, 
Sam

[support_indi.txt](https://github.com/user-attachments/files/20158636/support_indi.txt)

aaronwmorris, 2025-05-12:
The missing USB drive does appear to be the problem.  It looks like it just disappeared.

omeganon, 2025-05-12:
I wonder if this is an effect of power management? The default for Bookworm is to power down idle USB devices, but it's not clear to me if this also applies to mounted drives.

`cat /sys/module/usbcore/parameters/autosuspend`

Probably returns `2` which means autosuspend is enabled.

It might be worth disabling autosuspend by adding `usbcore.autosuspend=-1` to the end of the line in /boot/firmware/cmdline.txt. 

e.g. --
`console=serial0,115200 console=tty1 root=PARTUUID=0b4b817f-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=US usbcore.autosuspend=-1`

aaronwmorris, 2025-05-12:
It cannot hurt to try it, but I doubt it was the USB power management.

ikeysolomon, 2025-05-13:
Hi, this time we got through 1 whole day and 1 night before the USB was mysteriously "lost". Please see the support info and the last 2000 lines of the log. 

[homeastroberryindi-allskymiscsuppor.txt](https://github.com/user-attachments/files/20195185/homeastroberryindi-allskymiscsuppor.txt)
[2025-05-14T061311.904426+1000 astro.txt](https://github.com/user-attachments/files/20195184/2025-05-14T061311.904426%2B1000.astro.txt)

omeganon, 2025-05-13:
The output of the commands 

`dmesg`

and

`grep -i usb /var/log/kern.log`

might be clueful.

aaronwmorris, 2025-05-13:
@ikeysolomon @omeganon That was going to be the next ask.  The next time it happens, the output of `sudo dmesg` will tell us what is happening to the USB stick.

ikeysolomon, 2025-05-16:
Hey guys back again,

                          This time the error didn't "drop" the USB, it's just no longer taking pictures. I don't know if it is even related or not?
Please see attached files for hunting. Have included output from above commands as well this time even though perhaps this instance is another thing going on?

[output-dmesg.txt](https://github.com/user-attachments/files/20240069/output-dmesg.txt)
[output-GrepUsbVarLogKern.txt](https://github.com/user-attachments/files/20240070/output-GrepUsbVarLogKern.txt)
[output-logIndi.txt](https://github.com/user-attachments/files/20240068/output-logIndi.txt)
[output-SupportInfoIndi.txt](https://github.com/user-attachments/files/20240071/output-SupportInfoIndi.txt)

Kind regards,

Sam

ikeysolomon, 2025-05-19:
As this seems to be the USBs fault I will close the ticket. 
However, upon doing so I would like to discuss a feature(s) request. Reference the use of USBs with Indi. Please advise if you would like me to raise a ticket if you feel anything could be in scope for the direction you want to take?

Could the setup mechanism for using USBs be integrated more naturally into the normal setup shell script? If a user knows from the beginning installation they want to use a USB as their chosen storage mechanism, then can there be a prompt and a script that deals with this from the get go? 

This is with reference to: https://github.com/aaronwmorris/indi-allsky/discussions/1274  

Current implementation requires running setup twice. 

Similarly same concept with FTP and establishing a web interface for live streamed images? Surely some people know they want to do certain things and integrating it at the beginning would provide a better UX and they could be guided through each step rather than trying to find out after the fact?   

Further if I have USBs fail in future. Can the path entry field for image storage also have a button that launches a windows style graphic UI where the user can graphically navigate directories to the USB? This would be a bit more user friendly. I know I am potentially asking a lot but I think there would be a benefit here in performing 2 actions at the same time. Finding out the path to the new USB and simultaneously supplying this to the program.  

PS - have you considered a series of videos showing setup? They could be as simple as screen recordings with captions. 

Thank you. 
Kind Regards, 

Sam

closed 2025-05-19 by ikeysolomon