aaronwmorris/indi-allsky #2118: Dew Heater Config With TitanAstro Skycam One HAT

OceanStateAstro, 2025-05-12:
Hello, 

I have used the settings suggested by TitanAstro to connect and set up the sensor and dew heater to integrate with Indiallsky. It seems that the dew heater when turned on runs only at 100%. I noticed initially I used the dew port sensor slot set to temp instead of humidity. I ran a session last night and the dew heater was not turned on and read 0% all night. It is only powered when you set the sensor slot to "temperature" and only runs at 100% when activated. 

I reached out to TitanAstro and was redirected to post the issue here. 

[image]
[image]
[image]

aaronwmorris, 2025-05-12:
I think the problem is you have the `Dew Point Sensor Slot` set to the same value as the temperature, so it is not a valid target to compare the temperature.  You should set this to the dew point slot or set a manual target.

OceanStateAstro, 2025-05-12:
> I think the problem is you have the `Dew Point Sensor Slot` set to the same value as the temperature, so it is not a valid target to compare the temperature. You should set this to the dew point slot or set a manual target.

Okay I had that set before and it didn't seem to do anything so it switched it back to temperature to clear the dome this morning. Would switching the output work?

aaronwmorris, 2025-05-12:
The challenge may be the dew point reading.  If you have a sealed enclosure, the temperature and humdity reading would be the values INSIDE the dome, not outside the dome.  Therefore the dew point inside the dome may never reach the threshold.

You may have to use a manual target to have it activate.

OceanStateAstro, 2025-05-12:
Okay this makes sense, my sensor is enclosed. Is there any tricks to
setting manually? Or is it just enter values?

OceanStateAstro, 2025-05-13:
I have changed the Temp and humidity for the dew heater to read from the OpenWeatherMapAPI and the dew heater with thresholds on still runs at 100% when on. 

See video from last night here [https://youtu.be/5GOiu6gtwIk?si=CF89dYbrS-XW336u](url)

aaronwmorris, 2025-05-13:
Do you know what the dew point that is being returned by OpenWeatherMap?  49C seems high.

Can you post your Support Info here?

OceanStateAstro, 2025-05-14:
how do I pull support info?

aaronwmorris, 2025-05-14:
It is at the bottom of the Tools menu (assuming you are logged in).

aaronwmorris, 2025-05-14:
It looks like the output was truncated.

OceanStateAstro, 2025-05-14:
What does this mean?

aaronwmorris, 2025-05-14:
The Support Info was cut off.  There was more data in the output.

aaronwmorris, 2025-05-14:
It is still truncated, but enough is there I see the sensor info.

aaronwmorris, 2025-05-14:
Okay, I would recommend upgrading to a newer release.  I am not sure if I fixed any bugs related to this in a newer release, but I would go ahead an upgrade.

https://github.com/aaronwmorris/indi-allsky/wiki/Updating-indi-allsky

closed 2025-10-23 by aaronwmorris