aaronwmorris/indi-allsky #2125: USB keeps on resetting and glitched pictures with ASI120MM (Raspberry pi 4)

MightySpaceman, 2025-05-17:
I just got my ASI120MM working properly after a very long process, ending in somehow finding a flasher with firmware files from 2017 to update the firmware to the compatibility version (ZWO doesn't really like to hand out actual firmware files anymore).

It works, but it's buggy. It does everything well in terms of actual capture, but indi-allsky is constantly having to reset the USB connection on a minute-by-minute basis:

```
May 17 20:32:32 allsky python3[817]: [INFO] Capture-1-817/Dummy-1 indi.newMessage() [349]: new Message 2025-05-17T10:32:32: [WARNING] Exposure failed after 4 attempts. Attempting USB reset...
May 17 20:32:32 allsky python3[817]: [INFO] Capture-1-817/Dummy-1 indi.newMessage() [349]: new Message 2025-05-17T10:32:32: [INFO] Attempting USB reset for device ZWO ASI120MM...
May 17 20:32:32 allsky python3[817]: [INFO] Capture-1-817/Dummy-1 indi.newMessage() [349]: new Message 2025-05-17T10:32:32: [INFO] Finding USB port for device ZWO ASI120MM...
May 17 20:32:32 allsky python3[817]: [INFO] Capture-1-817/Dummy-1 indi.newMessage() [349]: new Message 2025-05-17T10:32:32: [ERROR] Failed to reset USB device
May 17 20:32:32 allsky python3[817]: [INFO] Capture-1-817/Dummy-1 indi.newMessage() [349]: new Message 2025-05-17T10:32:32: [INFO] Reopening camera after reset...
May 17 20:32:32 allsky python3[817]: [INFO] Capture-1-817/Dummy-1 indi.newMessage() [349]: new Message 2025-05-17T10:32:32: [INFO] Reinitializing camera...
May 17 20:32:32 allsky python3[817]: [INFO] Capture-1-817/Dummy-1 indi.newMessage() [349]: new Message 2025-05-17T10:32:32: [INFO] Attempting exposure again after USB reset...
May 17 20:32:33 allsky python3[817]: [INFO] Capture-1-817/Dummy-1 indi.newMessage() [349]: new Message 2025-05-17T10:32:33: [INFO] Taking a 6.25 seconds frame...
```

Many images are also glitched:

[image]

Sometimes it just chooses not to work outright, needing a reboot. Just a bit ago indiserver was complaining about a USB buffer overflow (see [here](https://asciinema.org/a/6EaMQiN3W07lm9wTcUgE4G2AF) but I think I re-ran the ./setup script and it's been replaced by just this repeating over and over.

I have looked at other issues here and it could be a USB cable problem. I have tried another cable and it didn't seem to make a difference, although it seems to be of less quality than the first one I have been using so I am not sure. Also, the previous owner of this camera used it in an allsky cam with a Pi 3 (or maybe older) with no issue that I know of using the same cable. I also have tested it on my laptop with ASICap and it seems to work smoothly at a full FPS. Still, I am going to buy a new shielded cable and some ferrite cores to put around it.

Of note, I have a udev rule to increase the USB buffer for the device that I made to try and fix the previous problem:

```
SUBSYSTEM=="usb", ATTR{idVendor}=="03c3", ATTR{idProduct}=="120a", ENV{LIBUSB_FORCE_SHORT_TRANSFER}="1", ENV{USBDEVFS_BUFFERSIZE}="200000000"
```
And running `cat /sys/module/usbcore/parameters/usbfs_memory_mb` returns 256.

[Support Info](https://github.com/user-attachments/files/20264473/support_info.txt)

aaronwmorris, 2025-05-17:
For the glitched images, you should set the USB Bandwidth to 100.

See https://bbs.zwoastro.com/d/11923-firmware-bug-asi120mm-mini/10

```json
{
    "PROPERTIES": {
      "CCD_CONTROLS": {
        "BandWidth": 100
      }
    },
    "SWITCHES": {
      "CCD_VIDEO_FORMAT": {
        "on": [
          "ASI_IMG_RAW16"
        ],
        "off": [
          "ASI_IMG_RAW8"
        ]
      }
    }
}
```

MightySpaceman, 2025-05-18:
It's the morning now, and everything seems to be working fine, no timeouts of glitched frames, at least for the day. Weird.

> For the glitched images, you should set the USB Bandwidth to 100.
> 
> See https://bbs.zwoastro.com/d/11923-firmware-bug-asi120mm-mini/10
> 
> {
>     "PROPERTIES": {
>       "CCD_CONTROLS": {
>         "BandWidth": 100
>       }
>     },
>     "SWITCHES": {
>       "CCD_VIDEO_FORMAT": {
>         "on": [
>           "ASI_IMG_RAW16"
>         ],
>         "off": [
>           "ASI_IMG_RAW8"
>         ]
>       }
>     }
> }

If it starts acting up again I'll try this. This issue should probably stay open because I have a feeling this stability won't last long.

MightySpaceman, 2025-05-18:
I just found that there are many images outright missing from the gallery even though the web viewer says they are there, and there are a bunch of things messed up in the UI. I'm going to do a fresh install of both raspbian and indi-allsky and see if that helps.

MightySpaceman, 2025-05-18:
It worked fine during the day, but now that it is night I am having trouble again. I've tried all configs - with the increased bandwidth, the 16 bit fix, both at the same time, and neither. The camera just takes one or two pictures, then keeps on having an exposure timeout with a BUSY or ALERT state:

```
May 18 18:19:13 allsky python3[927]: [INFO] Capture-1-927/MainThread capture.saferun() [307]: Camera last ready: 11.0s
May 18 18:19:13 allsky python3[927]: [INFO] Capture-1-927/MainThread capture.saferun() [308]: Exposure state: BUSY
May 18 18:19:13 allsky python3[927]: [INFO] Capture-1-927/MainThread capture.detectNight() [1489]: Sun alt: -15.4, Moon alt: -20.7, phase 72.1%
May 18 18:19:13 allsky python3[927]: [INFO] Capture-1-927/MainThread indi.getCcdTemperature() [892]: Camera temperature: 9.5
May 18 18:19:18 allsky python3[927]: [INFO] Capture-1-927/Dummy-1 indi.newMessage() [349]: new Message 2025-05-18T08:19:18: [ERROR] Exposure failed after 3 attempts.
May 18 18:19:18 allsky python3[927]: [INFO] Capture-1-927/Dummy-1 indi.newMessage() [349]: new Message 2025-05-18T08:19:18: [ERROR] Exposure failed after 3 attempts.
May 18 18:19:24 allsky python3[927]: [INFO] Capture-1-927/MainThread capture.saferun() [307]: Camera last ready: 22.2s
May 18 18:19:24 allsky python3[927]: [INFO] Capture-1-927/MainThread capture.saferun() [308]: Exposure state: ALERT
May 18 18:19:24 allsky python3[927]: [INFO] Capture-1-927/MainThread capture.detectNight() [1489]: Sun alt: -15.4, Moon alt: -20.7, phase 72.1%
May 18 18:19:24 allsky python3[927]: [INFO] Capture-1-927/MainThread indi.getCcdTemperature() [892]: Camera temperature: 9.5

```

omeganon, 2025-05-19:
When I experienced this, it was 2 things --

* 'bad' USB cable. Replacement cable helped a lot.
* The power supply I was using didn't provide enough power to run the PI, camera, and dew heater at the same time, causing camera drop-outs.

MightySpaceman, 2025-05-19:
I tried a 30w power supply instead of the default raspberry pi one and it... seems to run a little better? But it just delays the issue, taking 10 or so exposures before failing. Setting the USB bandwidth to anything but 40 seems to also render it fully inoperable.

MightySpaceman, 2025-05-20:
Testing it with the non-compatible firmware, the camera has an exposure timeout immediately.

closed 2025-10-23 by aaronwmorris