aaronwmorris/indi-allsky #2132: daytime capture colors are off

muyttendaele, 2025-05-22:
The trees on the right side are magenta instead of expected green.

[image]

The daytime settings ...
[image]

I'm using an ZWO-ASI678MC camera with included lens.

Thanks!

muyttendaele, 2025-05-22:
Just realized that this is due to lack of IR filter in that camera.

aaronwmorris, 2025-05-23:
You pretty much figured it out on your own.  The astronomy shops sell replacement windows for ZWO cameras, so you could feasibly purchase an IR-Cut window for the ASI678MC.

muyttendaele, 2025-05-24:
I've decided to ebay the 678MC and buy 676MC. It has an IR filter and slightly better resolution.

I'm really liking indi-allsky so fqr. Thanks for creating and supporting it!

closed 2025-05-24 by muyttendaele