aaronwmorris/indi-allsky #2133: Feature Request: Separate framerate settings for day/night timelapse

wantutri27, 2025-05-23:
Since daytime capture creates more images, is it possible to have a separate fps setting when generating daytime timelapse?

Thanks!

aaronwmorris, 2025-05-24:
I will look at different daytime and night time timelapse settings (framerate, bitrate, etc).

wantutri27, 2025-05-24:
> If you set the same Exposure Period for both, the frame rates will be the same.
> 
> E.g. I have both set to 15s for me, which means 4 images per minute for both.
> 
> Related, I have my max exposure time set to 15s as well.

I have my period set to 15s too, but max exposure during the night is 60s so I manually create daytime timelapse

> I will look at different daytime and night time timelapse settings (framerate, bitrate, etc).

Thanks! Looking forward to this.

aaronwmorris, 2025-05-31:
Merged #2152 which creates new unique daytime timelapse settings.

wantutri27, 2025-05-31:
> Merged [#2152](https://github.com/aaronwmorris/indi-allsky/pull/2152) which creates new unique daytime timelapse settings.

Thanks! Will check it later after the day. But I noticed that after the update, all the images are overexposed compared to before. Even when it's overcast for the past week, the day images were not overexposed (ADU set at 75 +/-3)
[image]

aaronwmorris, 2025-05-31:
Did you upgrade your packages when you upgraded?

I am not exactly sure what would cause the images to become darker.  The only possibility I have considered is maybe your dark frames were created with a different version of libcamera and they are causing too much data to be subtracted.  Can you disable applying dark frames (Image tab `Apply Dark Calibration Frames`) and see if that helps?

wantutri27, 2025-05-31:
> Did you upgrade your packages when you upgraded?
>
> I am not exactly sure what would cause the images to become darker. The only possibility I have considered is maybe your dark frames were created [...]

Yes, I frequently update all my packages. I built my dark frames around end of March but want to point out that everything is normal when using 2be3283. I will switch back to the latest and try also the 2nd to the last commit. Will be back in a while to update

aaronwmorris, 2025-06-01:
There have been no changes to the ADU calculations, nor anything that would affect exposure.

I suspect that libcamera might have changed the behavior of the camera in the latest release.  The dark frames you have may no longer be valid based on the new behavior.

aaronwmorris, 2025-06-01:
Regarding the latest update which changes the mask base... I suppose it is possible.

Can you post your mask base as a file in a zip file, I would like to see what it looks like.

aaronwmorris, 2025-06-02:
After thinking about it, I reversed the logic in #2153.  Any non-zero (non-black part of the mask) will be included in processing.  This change basically caused your mask to exclude pretty much the whole image except for the bright white areas.

The original intent was for the "white" areas to be included, but there is no point in making it quite that strict.

Merged #2160.

wantutri27, 2025-06-02:
> After thinking about it, I reversed the logic in [#2153](https://github.com/aaronwmorris/indi-allsky/pull/2153). Any non-zero (non-black part of the mask) will be included in processing. This change [...]

Thanks! This issue is solved, and the images are properly exposed now. The day and night timelapse settings are the same, and they work as intended. I will close this issue.
[image]

closed 2025-06-02 by wantutri27