aaronwmorris/indi-allsky #2155: white balance at night

muyttendaele, 2025-06-01:
My night captures have a very green tint to them. I'm using INDI capture with a ZWO ASI676. Any thoughts on how to adjust WB here?  Thanks.

[image]

muyttendaele, 2025-06-01:
I have not. Will give that a try and report back.

muyttendaele, 2025-06-01:
I set the SCNR algo to max neutral. The results are much better but still feels like a bit of green/yellow tint.
[image]

My experience is mainly with terrestrial photography so I'm not very familiar with SCNR but I thought it was more of a correction applied to deepsky objects. At first glance the WB of these images seemed off. When I take them into Adobe Camera Raw and set the whitepoint I do get a reasonable result. 

Here is a sample from the night of the original post (above is un-corrected, below is after setting white-point on the window trim)
[image]

Here is a sample from last night where I applied SCNR (again above is un-corrected, below is after setting white-point on the window trim)
[image]

aaronwmorris, 2025-06-02:
You have the option of turning on RAW capture for a few frames (then turn it off).  You can then use the FITS processor to play with the image processing settings, SCNR, color, stretching, etc in realtime.

muyttendaele, 2025-06-02:
That FITS feature is nice. Are you aware of any way to convert the .fit files to DNG format so that I can edit in the adobe tools?

closed 2025-08-24 by aaronwmorris