aaronwmorris/indi-allsky #2185: Support sensor AS3935 - lightning detector

pavelkonir, 2025-06-18:
I would like to request the integration of a lightning detection sensor based on the AS3935 chip into the INDI-AllSky system.

The purpose of this sensor is to detect lightning activity in the vicinity of the all-sky camera and provide relevant data such as distance and intensity of strikes.

An existing Python library for Raspberry Pi is available here:
https://github.com/pcfens/RaspberryPi-AS3935

Please consider adding support for this sensor.

aaronwmorris, 2025-06-25:
@illigtr Yes, it can probably be run in polling mode.  indi-allsky normally polls the sensors every 15 seconds, in this mode I think the lightning sensor would only return info related to the last detected lightning strike (since the last poll).

illigtr, 2025-06-25:
That may be enough, from what I remember the registers keep the data from the last event, so we should be able to extract distance and intensity of the last strike(s) of the last (15 second) poll. :)

timchurches, 2025-06-25:
No hurry on this from my POV. But polling even 15 seconds would miss multiple lightning events in that interval, only getting the last one. Running a separate process (rather than just an asyncio coroutine) that waits for interrupts from the sensor and pushes them onto a queue (or even writes them to a log file) which is then read by the main indi-allsky process might be a solution, that could be useful for other sensors as well?

pavelkonir, 2025-06-26:
I'm thinking of making a python script that will be run based on an interrupt, read values ​​from the chip and through mqtt it would go to allsky. I think that could run on a raspberry pi next to allsky. I just want i2c on other ports so that communication is not affected and asynchronous processing from lightning works

aaronwmorris, 2025-08-24:
That was a challenge I had to work through as well.  i2cdetect normally only shows 0x07 through 0x77.  I updated the wiki with instructions to increase the range to 0x00.  Both of mine would consistently show up as 0x00 but the sensor would never connect via the software.  Trying to change the address never worked, but I gave up after not too much time.

I had these sensors working on a ESP8266 about 5 years ago, but I think they were in SPI mode.  I believe I actually had the modules a couple of years before that, but I was never able to detect lightning.  One anecdote I distinctly remember was in a bout of testing, I ran the calibration routine twice during initialization and it started working.

illigtr, 2025-08-24:
Thanks for the hint of double-calibration. I'll keep that in mind. So far, in testing, I have been able to read the frequency and set the oscillators, I also have been able to detect lightning with both units, one as i2c, the other SPI. Of course this is just testing under Arduino IDE with an old trustworthy UNO board. Libraries were from DF-ROBOT and Sparkfun.  Next step is to understand why alerts for lightning don't appear in ESPHOME, I'll have to look at what library they might be using and then, finally, I will test both sensors with indi-allsky.

illigtr, 2025-09-01:
@aaronwmorris Ok, I'm finally getting around to installing one of my as3935 with my indi-allsky. I'll try my trusty DR-ROBOT, even though it is only i2c, if not, I'll switch to SPI with SJMCU-3935. A few quick questions:

In sensor setup, I put the as3935 as Sensor C. I noticed that one must assign the first Pin/Port for every device, even if the device doesn't use a pin/port. Reason for this?

Also, perhaps a label issue, seems there are two labeled Pin/Port 2. I am guessing that one of these is to set the interrupt pin from the as3935? Can you confirm?

[image]

I seem to be getting an undefined interrupt 2 in the logs. I'll check this next, perhaps I don't have GPIO.pin installed.

illigtr, 2025-09-01:
@aaronwmorris 

Here is that specific error in the  /var/log/indi-allsky/indi-allsky.log

```
2025-09-01T17:17:12.590411-04:00 indi-allsky [ERROR] MainProcess-1124/MainThread allsky._startSensorWorker() [539]: Sensor worker exception: Traceback (most recent call last):
2025-09-01T17:17:12.590580-04:00 indi-allsky [ERROR] MainProcess-1124/MainThread allsky._startSensorWorker() [539]: Sensor worker exception:   File "/home/pi/indi-allsky/indi_allsky/sensor.py", line 129, in run
2025-09-01T17:17:12.590678-04:00 indi-allsky [ERROR] MainProcess-1124/MainThread allsky._startSensorWorker() [539]: Sensor worker exception:     self.saferun()
2025-09-01T17:17:12.590758-04:00 indi-allsky [ERROR] MainProcess-1124/MainThread allsky._startSensorWorker() [539]: Sensor worker exception:   File "/home/pi/indi-allsky/indi_allsky/sensor.py", line 139, in saferun
2025-09-01T17:17:12.590836-04:00 indi-allsky [INFO] Capture-3-1514/Dummy-5 indi.newDevice() [224]: new device pylibcamera Main
2025-09-01T17:17:12.590919-04:00 indi-allsky [ERROR] MainProcess-1124/MainThread allsky._startSensorWorker() [539]: Sensor worker exception:     self.init_sensors()  # sensors before dew heater and fan
2025-09-01T17:17:12.590992-04:00 indi-allsky [ERROR] MainProcess-1124/MainThread allsky._startSensorWorker() [539]: Sensor worker exception:     ^^^^^^^^^^^^^^^^^^^
2025-09-01T17:17:12.591094-04:00 indi-allsky [ERROR] MainProcess-1124/MainThread allsky._startSensorWorker() [539]: Sensor worker exception:   File "/home/pi/indi-allsky/indi_allsky/sensor.py", line 472, in init_sensors
2025-09-01T17:17:12.591170-04:00 indi-allsky [ERROR] MainProcess-1124/MainThread allsky._startSensorWorker() [539]: Sensor worker exception:     self.sensors[2] = c_sensor(
2025-09-01T17:17:12.591236-04:00 indi-allsky [ERROR] MainProcess-1124/MainThread allsky._startSensorWorker() [539]: Sensor worker exception:                       ^^^^^^^^^
2025-09-01T17:17:12.591307-04:00 indi-allsky [ERROR] MainProcess-1124/MainThread allsky._startSensorWorker() [539]: Sensor worker exception:   File "/home/pi/indi-allsky/indi_allsky/devices/sensors/lightningSensorAs3935.py", line 183, in __init__
2025-09-01T17:17:12.591386-04:00 indi-allsky [ERROR] MainProcess-1124/MainThread allsky._startSensorWorker() [539]: Sensor worker exception:     raise SensorException('Pin 2 (Interrupt) not defined')
2025-09-01T17:17:12.591508-04:00 indi-allsky [ERROR] MainProcess-1124/MainThread allsky._startSensorWorker() [539]: Sensor worker exception: indi_allsky.devices.exceptions.SensorException: Pin 2 (Interrupt) not defined
2025-09-01T17:17:12.591607-04:00 indi-allsky [ERROR] MainProcess-1124/MainThread allsky._startSensorWorker() [539]: Sensor worker exception: 
```

aaronwmorris, 2025-09-01:
Indeed, `Pin/Port 2` was supposed to be 1 for the first pin.  That is just a typo in the form description and should not have any negative consequences.  (I will push a change out later for this).

I just decided to standardize pin 2 for the interrupt pin for both SPI and I2C devices even though pin 1 is not used for I2C.

Can you perform a full restart (not just a reload) or just a full reboot?  Maybe it is using an old config for the pin 2 configuration.

illigtr, 2025-09-01:
Actually I did a few things since the initial post today. I updated the OS, I ran the indi-allsky upgrade, and I rebooted. As mentioned, the sensor does appear to work, I was just curious about the error in the logs,  which persist.

illigtr, 2025-09-02:
I put a dummy value for Pin1. I had original put the same value in Pin1 and 2. I don't have further log errors.

[image]

aaronwmorris, 2025-09-02:
That should not make a difference, but I will continue to check.

illigtr, 2025-09-04:
@aaronwmorris I've been back and forth with developers at ESPHome to get the AS3935 to work under I2C. You may wish to review it [here](https://github.com/esphome/esphome/issues/10455#issuecomment-3250999545)

I think it is finally working. TL;DR the read register routine was not working correctly in I2C.

There are some other interesting nuggets in the discussion, including the caps installed on the CJMCU-3935. I have not tested mine, but my sensors work fine. 

I have installed my DFROBOT AS3935 into my indi-allsky. I also have the CJMCU-3935 running under ESPHome. We may have a storm this evening or tomorrow. I'll try to compare the two sensors.

Right now I do have fairly high noise counts that I am trying to address with the noise setting. 

Some things I've learned over the past week testing this device under Arduino IDE and ESPHome, both in SPI and I2C mode.

The sensor should always be re-calibrated upon power up. Either call the re-calibrate routine or put the sensor in sleep and then wake it up, which should force re-calibration. I have seen where the sensor will simply not work at all unless this is done on initialization. There is also a setting to do a soft reset, basically wiping out all settings in the sensor. I added this to my code in Arduino and ESPHome. 
 

[image]

illigtr, 2025-09-04:
@aaronwmorris I browsed your lightningSensorAs3935.py file in ~/indi-allsky/indi_allsky/devices/sensors. Hoping this is the right file.

From what I can see in the init sequence, no calibration is done. I can see that various settings are assigned, although not all are available in the GUI.

aaronwmorris, 2025-09-04:
This is the library used for managing the AS3935:  https://github.com/fourstix/Sparkfun_CircuitPython_QwiicAS3935/

That is the correct file.  I did not expose all settings in the GUI because it is difficult to determine which settings are actually useful to expose.  If something is useful, I can work on adding it.  (see my update below)  The antenna tuning has never been super useful for me in the past, but I could be convinced otherwise.

It looks like I have made a very poor assumption.  I assumed the library was automatically calibrating on startup, but it does not appear to be doing that.  I will push a change shortly to perform a calibration on startup.

illigtr, 2025-09-04:
@aaronwmorris Since strike, disturber and noise are interrupt counts, under what circumstances do these reset to zero? Reboot/shutdown? Restart of indi-allsky services (I don't think so, at least from what I can see). Perhaps a different graphical value over time slot might be more useful, at least for storm alerts. The disturber/noise are good just to debug the best values for the sensor.

This is the from the ESPHome sensor:

[image]

[image]

aaronwmorris, 2025-09-05:
I may change the null value to 0 since the lowest value reported for distance should be 1km (or "overhead").  I initially used -1 since I thought the smallest distance was 0.

The data for the AS3935 is stored as a 5 minute stack.  The sensors are read every 15 seconds.  indi-allsky stores the number of lightning strikes and distances in an array every 15 seconds and entries older then 5 minutes (+20 entries) are removed from the stack.

The challenge here is the sensors are read asynchronously from the exposures.  I have to account for the fact that exposures might be taken every 15, 30, 45, or 60 (etc) seconds.  This was the only way I could think of to represent multiple data reads that occur between exposures.

illigtr, 2025-09-05:
@aaronwmorris Ok. Perfect. I think that method works fine. So this means that if the array if full of values for say, disturbers or noise, and then there is no disturber or noise for 5+ minutes, then the graph would begin to show 0 values again. Correct?

aaronwmorris, 2025-09-05:
@illigtr Affirmative

aaronwmorris, 2025-09-06:
I have merged #2354 with the calibration routine and setting null values to 0

illigtr, 2025-09-06:
I found a minor error when trying to adjust spike rejection. I believe spike rejection can be from 1 to 15, at least in the SparkFun library I'm using with Arduino. I think the entry mask was copied from the Noise Level Threshold, as it is set to 1-7 (correctly).

[image]

illigtr, 2025-09-06:
So I now have both my as3935 sensors located outside. One under ESPHome control, the other under indi-allsky. Both are configured for I2C. I was hoping to have both use the same parameters to compare performance, but setting the indi-allsky one to OUTDOOR creates dozens of noise interrupts, regardless of setting, even up to setting 7. I think this has more to do with the placement of the sensor inside my waterproof enclosure for indi-allsky. I am designing a secondary child enclosure only to house my various sensors and connect them all on one i2c cable to my main waterproof enclosure. I'll allow for about 1 foot distance. Hoping this reduces the noise alerts and allows me to use the sensor at maximum gain. I should have some good data to share in 2-3 weeks.

aaronwmorris, 2025-09-06:
@illigtr Thank you for catching the spike rejection error.  The message was correct, the max value is 11, but I did not adjust the condition (copy/paste error).  Merged #2357

illigtr, 2025-09-16:
Some progress from my side. I have installed the BME680 and AS3935 in a separate waterproof ASA box approximately 18 inches away from the main indi-allsky installation. Using Outdoor = true, I am no longer am spammed by noise interrupts.  Autumn weather has been exceptional, so no storms yet, but now I have 2 lightning sensors, both with the same configuration and protocol (ESPHome, indi-allsky).

aaronwmorris, 2025-09-27:
I have yet to have a single detection from my sensor despite several thunderstorms.  The caveat is that this is sitting on my desk in my office.

I believe I previously mentioned I had a high amount of noise detections.  The noise sometimes disappears, usually when I am not in my office.  I think I have identified the source of the noise:  my Pi5 has a cooler with a fan that attaches to the onboard PWM fan controller of the Pi5.  The noise tends to happen when the fan activates and the fan is more active when I am in my office (probably because I have more things turned on generating heat when I am in my office).  That is my current top suspect, anyway.

I also noticed that sometimes the AS3935 fails to calibrate on start up.  I have merged #2411 which retries the calibration routine until it succeeds.