aaronwmorris/indi-allsky #2188: No image for 15 minutes

vineyardverde, 2025-06-18:
Hello, my camera has stopped taking images (it was fine all day, but is now saying No image for 15 minutes) - excerpts of the log below (I couldn't paste the whole log b/c of length).  Thank you!

/home/vineyard/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 8009116 kB

System: Raspberry Pi 4 Model B Rev 1.5

Linux allskypi 6.12.25+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64 GNU/Linux
Hostname
allskypi
allskypi.Home 

Time
Wed Jun 18 08:59:48 PM WEST 2025

System timezone
Europe/Lisbon

Uptime
 20:59:48 up  7:35,  2 users,  load average: 1.12, 0.85, 1.05

Memory
               total        used        free      shared  buff/cache   available
Mem:         8009116     2094292     3526876      353268     2890572     5914824
Swap:         524284           0      524284

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
udev             3733332        0   3733332   0% /dev
tmpfs             800912    33112    767800   5% /run
/dev/mmcblk0p2  30082288 28535972         0 100% /
tmpfs            4004556      468   4004088   1% /dev/shm
tmpfs               5120       16      5104   1% /run/lock
/dev/mmcblk0p1    522230    70298    451932  14% /boot/firmware
tmpfs             800908       44    800864   1% /run/user/1000

systemctl runlevel
graphical.target

sysctl info
vm.swappiness = 1

Thermal info
cpu-thermal
41381

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

```

indi-allky log errors
```
2025-06-18T20:59:33.005857+01:00 allskypi [ERROR] Capture-1-9150/MainThread libcamera.getCcdExposureStatus() [314]: libcamera-still error: b'    SRGGB12_CSI2P,2028x1080/0 - Score: 8179.84\n'
2025-06-18T20:59:33.005981+01:00 allskypi [ERROR] Capture-1-9150/MainThread libcamera.getCcdExposureStatus() [314]: libcamera-still error: b'    SRGGB12_CSI2P,2028x1520/0 - Score: 7096\n'
2025-06-18T20:59:33.006081+01:00 allskypi [ERROR] Capture-1-9150/MainThread libcamera.getCcdExposureStatus() [314]: libcamera-still error: b'    SRGGB12_CSI2P,4056x3040/0 - Score: 0\n'
2025-06-18T20:59:33.006172+01:00 allskypi [ERROR] Capture-1-9150/MainThread libcamera.getCcdExposureStatus() [314]: libcamera-still error: b'[7:35:09.380691260] [30704] \x1b[1;32m INFO \x1b[1;37mCamera \x1b[1;34mcamera.cpp:1205 \x1b[0mconfiguring streams: (0) 4056x3040-YUV420 (1) 4056x3040-SBGGR12_CSI2P\n'
2025-06-18T20:59:33.006257+01:00 allskypi [ERROR] Capture-1-9150/MainThread libcamera.getCcdExposureStatus() [314]: libcamera-still error: b'[7:35:09.381061145] [30710] \x1b[1;32m INFO \x1b[1;37mRPI \x1b[1;34mvc4.cpp:622 \x1b[0mSensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 4056x3040-SBGGR12_1X12 - Selected unicam format: 4056x3040-pBCC\n'
2025-06-18T20:59:33.006384+01:00 allskypi [ERROR] Capture-1-9150/MainThread libcamera.getCcdExposureStatus() [314]: libcamera-still error: b'Still capture image received\n'
2025-06-18T20:59:33.006454+01:00 allskypi [ERROR] Capture-1-9150/MainThread libcamera.getCcdExposureStatus() [314]: libcamera-still error: b'ERROR: *** failed to write file - output probably corrupt ***\n'
2025-06-18T20:59:33.006507+01:00 allskypi [ERROR] Capture-1-9150/MainThread libcamera._processMetadata() [334]: Metadata file not found: [Errno 2] No such file or directory: '/tmp/tmphttg7hn0.json'
2025-06-18T20:59:33.006565+01:00 allskypi [ERROR] Capture-1-9150/MainThread libcamera._processMetadata() [357]: libcamera camera temperature key not found
2025-06-18T20:59:33.006617+01:00 allskypi [ERROR] Capture-1-9150/MainThread libcamera._processMetadata() [404]: libcamera sensor AWB key not found
2025-06-18T20:59:33.006667+01:00 allskypi [ERROR] Capture-1-9150/MainThread libcamera._processMetadata() [436]: libcamera sensor black level key not found
2025-06-18T20:59:33.007132+01:00 allskypi [ERROR] Image-2-24590/MainThread image.processImage() [298]: Frame is empty: /tmp/tmpf0be23nm.jpg
2025-06-18T20:59:48.745696+01:00 allskypi [ERROR] Capture-1-9150/MainThread libcamera.getCcdExposureStatus() [314]: libcamera-still error: b'[7:35:24.872696652] [30718] \x1b[1;32m INFO \x1b[1;37mCamera \x1b[1;34mcamera_manager.cpp:326 \x1b[0mlibcamera v0.5.0+59-d83ff0a4\n'
2025-06-18T20:59:48.745862+01:00 allskypi [ERROR] Capture-1-9150/MainThread libcamera.getCcdExposureStatus() [314]: libcamera-still error: b'[7:35:24.969757014] [30736] \x1b[1;33m WARN \x1b[1;37mRPiSdn \x1b[1;34msdn.cpp:40 \x1b[0mUsing legacy SDN tuning - please consider moving SDN inside rpi.denoise\n'
2025-06-18T20:59:48.745956+01:00 allskypi [ERROR] Capture-1-9150/MainThread libcamera.getCcdExposureStatus() [314]: libcamera-still error: b'[7:35:24.972255284] [30736] \x1b[1;32m INFO \x1b[1;37mRPI \x1b[1;34mvc4.cpp:447 \x1b[0mRegistered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media1 and ISP device /dev/media2\n'
2025-06-18T20:59:48.746025+01:00 allskypi [ERROR] Capture-1-9150/MainThread libcamera.getCcdExposureStatus() [314]: libcamera-still error: b"[7:35:24.972330301] [30736] \x1b[1;32m INFO \x1b[1;37mRPI \x1b[1;34mpipeline_base.cpp:1121 \x1b[0mUsing configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'\n"
2025-06-18T20:59:48.746082+01:00 allskypi [ERROR] Capture-1-9150/MainThread libcamera.getCcdExposureStatus() [314]: libcamera-still error: b'Mode selection for 4056:3040:12:P\n'
2025-06-18T20:59:48.746133+01:00 allskypi [ERROR] Capture-1-9150/MainThread libcamera.getCcdExposureStatus() [314]: libcamera-still error: b'    SRGGB10_CSI2P,1332x990/0 - Score: 10552.2\n'
2025-06-18T20:59:48.746185+01:00 allskypi [ERROR] Capture-1-9150/MainThread libcamera.getCcdExposureStatus() [314]: libcamera-still error: b'    SRGGB12_CSI2P,2028x1080/0 - Score: 8179.84\n'
2025-06-18T20:59:48.746236+01:00 allskypi [ERROR] Capture-1-9150/MainThread libcamera.getCcdExposureStatus() [314]: libcamera-still error: b'    SRGGB12_CSI2P,2028x1520/0 - Score: 7096\n'
2025-06-18T20:59:48.747173+01:00 allskypi [ERROR] Capture-1-9150/MainThread libcamera.getCcdExposureStatus() [314]: libcamera-still error: b'    SRGGB12_CSI2P,4056x3040/0 - Score: 0\n'
2025-06-18T20:59:48.747788+01:00 allskypi [ERROR] Capture-1-9150/MainThread libcamera.getCcdExposureStatus() [314]: libcamera-still error: b'[7:35:24.978326423] [30718] \x1b[1;32m INFO \x1b[1;37mCamera \x1b[1;34mcamera.cpp:1205 \x1b[0mconfiguring streams: (0) 4056x3040-YUV420 (1) 4056x3040-SBGGR12_CSI2P\n'
2025-06-18T20:59:48.748718+01:00 allskypi [ERROR] Capture-1-9150/MainThread libcamera.getCcdExposureStatus() [314]: libcamera-still error: b'[7:35:24.980322828] [30736] \x1b[1;32m INFO \x1b[1;37mRPI \x1b[1;34mvc4.cpp:622 \x1b[0mSensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 4056x3040-SBGGR12_1X12 - Selected unicam format: 4056x3040-pBCC\n'
2025-06-18T20:59:48.749007+01:00 allskypi [ERROR] Capture-1-9150/MainThread libcamera.getCcdExposureStatus() [314]: libcamera-still error: b'Still capture image received\n'
2025-06-18T20:59:48.749270+01:00 allskypi [ERROR] Capture-1-9150/MainThread libcamera.getCcdExposureStatus() [314]: libcamera-still error: b'ERROR: *** failed to write file - output probably corrupt ***\n'
2025-06-18T20:59:48.749441+01:00 allskypi [ERROR] Capture-1-9150/MainThread libcamera._processMetadata() [334]: Metadata file not found: [Errno 2] No such file or directory: '/tmp/tmpbnnsy7cz.json'
2025-06-18T20:59:48.749552+01:00 allskypi [ERROR] Capture-1-9150/MainThread libcamera._processMetadata() [357]: libcamera camera temperature key not found
2025-06-18T20:59:48.749692+01:00 allskypi [ERROR] Capture-1-9150/MainThread libcamera._processMetadata() [404]: libcamera sensor AWB key not found
2025-06-18T20:59:48.749786+01:00 allskypi [ERROR] Capture-1-9150/MainThread libcamera._processMetadata() [436]: libcamera sensor black level key not found
2025-06-18T20:59:48.750544+01:00 allskypi [ERROR] Image-2-24590/MainThread image.processImage() [298]: Frame is empty: /tmp/tmpok7fowen.jpg
```

#################################
###     end support info      ###
#################################

aaronwmorris, 2025-06-18:
You are out of space on your `/` (root) filesystem.

vineyardverde, 2025-06-18:
Sorry I have no idea why it pasted the original thing again.  When I try and execute the expire code, it's returning error message:

Traceback (most recent call last):
  File "/home/vineyard/indi-allsky/./misc/expire_images.py", line 24, in <module>
    from sqlalchemy.orm.exc import NoResultFound
  File "/home/vineyard/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/sqlalchemy/__init__.py", line 12, in <module>
    from . import util as _util
  File "/home/vineyard/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/sqlalchemy/util/__init__.py", line 15, in <module>
    from ._collections import coerce_generator_arg as coerce_generator_arg
  File "/home/vineyard/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/sqlalchemy/util/_collections.py", line 39, in <module>
    from .typing import is_non_string_iterable
  File "/home/vineyard/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/sqlalchemy/util/typing.py", line 59, in <module>
    from typing_extensions import TypeAliasType as TypeAliasType  # 3.12
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ImportError: cannot import name 'TypeAliasType' from 'typing_extensions' (/usr/lib/python3/dist-packages/typing_extensions.py)

vineyardverde, 2025-06-18:
Thanks Aaron.  I tried that and it started doing something but still came back with an error as per below.

Just out of curiosity, shouldn't old files be wiped automatically as and when the SD card gets full?

[INFO]: Cutoff for images:       30 days
[INFO]: Cutoff for raw images:   10 days
[INFO]: Cutoff for fits images:  10 days
[INFO]: Cutoff for videos:       365 days
[WARNING]: Found 24565 expired images to delete
[WARNING]: Found 0 expired Panorama images to delete
[WARNING]: Found 0 expired FITS images to delete
[WARNING]: Found 0 expired RAW images to delete
[WARNING]: Found 0 expired videos to delete
[WARNING]: Found 0 expired mini videos to delete
[WARNING]: Found 0 expired keograms to delete
[WARNING]: Found 0 expired star trails to delete
[WARNING]: Found 0 expired star trail videos to delete
[WARNING]: Found 0 expired panorama videos to delete
[INFO]: Proceeding in 10 seconds
[WARNING]: Deleting...
[INFO]: Removing old IndiAllSkyDbImageTable entry: ccd_3edce882-446e-4f90-a5b8-f008d07ed9bc/exposures/20250421/day/21_16/ccd1_20250421_164636.jpg
Traceback (most recent call last):
  File "/home/vineyard/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1142, in _commit_impl
    self.engine.dialect.do_commit(self.connection)
  File "/home/vineyard/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/sqlalchemy/engine/default.py", line 706, in do_commit
    dbapi_connection.commit()
sqlite3.OperationalError: database or disk is full

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/vineyard/indi-allsky/./misc/expire_images.py", line 350, in <module>
    ei.main()
  File "/home/vineyard/indi-allsky/./misc/expire_images.py", line 253, in main
    delete_count += self._deleteAssets(asset_table, id_list)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/vineyard/indi-allsky/./misc/expire_images.py", line 285, in _deleteAssets
    entry.deleteAsset()
  File "/home/vineyard/indi-allsky/indi_allsky/flask/models.py", line 205, in deleteAsset
    db.session.commit()
  File "/home/vineyard/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/sqlalchemy/orm/scoping.py", line 599, in commit
    return self._proxied.commit()
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/vineyard/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/sqlalchemy/orm/session.py", line 2032, in commit
    trans.commit(_to_root=True)
  File "<string>", line 2, in commit
  File "/home/vineyard/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/sqlalchemy/orm/state_changes.py", line 139, in _go
    ret_value = fn(self, *arg, **kw)
                ^^^^^^^^^^^^^^^^^^^^
  File "/home/vineyard/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/sqlalchemy/orm/session.py", line 1320, in commit
    trans.commit()
  File "/home/vineyard/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 2629, in commit
    self._do_commit()
  File "/home/vineyard/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 2734, in _do_commit
    self._connection_commit_impl()
  File "/home/vineyard/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 2705, in _connection_commit_impl
    self.connection._commit_impl()
  File "/home/vineyard/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1144, in _commit_impl
    self._handle_dbapi_exception(e, None, None, None, None)
  File "/home/vineyard/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 2352, in _handle_dbapi_exception
    raise sqlalchemy_exception.with_traceback(exc_info[2]) from e
  File "/home/vineyard/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1142, in _commit_impl
    self.engine.dialect.do_commit(self.connection)
  File "/home/vineyard/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/sqlalchemy/engine/default.py", line 706, in do_commit
    dbapi_connection.commit()
sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) database or disk is full
(Background on this error at: https://sqlalche.me/e/20/e3q8)

aaronwmorris, 2025-06-19:
It sounds like you will need to manually free some space before you are able to expire images.  It probably expands the database when changes are made (even deletes).

indi-allsky expires images twice a day, but the appropriate retention settings need to be entered.  A 32GB card is going to be very restrictive, especially for a Raspberry Pi HQ camera.  Even with a 128GB card, you would probably only be able to reasonably store 10 days of images.  A 32GB card would probably need to expire images after 2-3 days.

vinbew, 2025-06-19:
Thanks Aaron, yes that worked.  I found the image files eventually in the var/www/ subfolders and deleted a bunch manually.  And then ran the expire_image.py script.  It's down to 84% memory now (instead of full) and I've also reset the expiry date to 3 days (I don't really need them to kept for much longer) - will change the SD card eventually.

Thanks for your help with this.