aaronwmorris/indi-allsky #2200: Docker compose v2 build error

jmejco, 2025-06-26:
# Description

In a clean environment with no pre-existing images, when using `docker compose build` with Docker Compose v2, the build fails on any service that uses `FROM docker-indi.base`, with the following error:

```
target capture_indi_allsky: failed to solve: docker-indi.base: failed to resolve source metadata 
for docker.io/library/docker-indi.base:latest: pull access denied, repository does not exist 
or may require authorization: server message: insufficient_scope: authorization failed
```

This happens because Compose v2 attempts to build all images in parallel, and the image build for services like `capture.indi.allsky` start before the `indi.base` image is built and available locally. As a result, Docker incorrectly tries to pull `docker-indi.base` from Docker Hub instead. The Docker [docs](https://docs.docker.com/compose/how-tos/dependent-images/#use-another-services-image-as-the-base-image) explain this behavior and recommend declaring the dependent base image using `additional_context`.

# Workaround
I was able to workaround by running the base image build manually first:

```bash
docker compose build indi.base
docker compose build
```

# Environment

* Docker: 28.3.0
* Docker Compose: v2.37.3
* OS: Debian 12 (Raspberry Pi 5)

# Logs

[docker-build-error.log](https://github.com/user-attachments/files/20926452/docker-build-error.log)

aaronwmorris, 2025-06-27:
I tried to use the `additional_context` but it still failed (in a different spot).  I have updated the wiki to use the workaround you indicated.

jmejco, 2025-06-28:
I was able to run a full build with these changes: https://github.com/aaronwmorris/indi-allsky/compare/main...jmejco:indi-allsky:patch/docker-compose-v2-build?expand=1

Where did it fail for you?

aaronwmorris, 2025-06-28:
I will try again.

aaronwmorris, 2025-07-03:
I guess I did not have the correct keys for the additional contexts.  I tried various combinations previously, but it always returned an error.  I really appreciate the direction!

Merged #2227 with the updates (and a few other updates)

jmejco, 2025-07-04:
@aaronwmorris 

Awesome! Thank you for all the amazing work you do for maintaining this project.

I just managed to make mine go live today, can't wait to see what I'll get tonight!

https://allsky.jamespace.uk