aaronwmorris/indi-allsky #2201: Youtube upload error

omeganon, 2025-06-26:
Hey Aaron,

I'm setting up a second camera and Youtube upload is giving me some issues. I'm using the same project as my other allsky camera, but each camera has its own credentials. The first continues to work fine, but the second (mostly?) generates this traceback -

```
   File "/home/admin/indi-allsky/indi_allsky/uploader.py", line 108, in run
     self.saferun()
   File "/home/admin/indi-allsky/indi_allsky/uploader.py", line 139, in saferun
     self.processUpload(u_dict)
   File "/home/admin/indi-allsky/indi_allsky/uploader.py", line 472, in processUpload
     response = client.put(**put_kwargs)
                ^^^^^^^^^^^^^^^^^^^^^^^^
   File "/home/admin/indi-allsky/indi_allsky/filetransfer/youtube_oauth2.py", line 116, in put
     response = self.resumable_upload(insert_request)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   File "/home/admin/indi-allsky/indi_allsky/filetransfer/youtube_oauth2.py", line 139, in resumable_upload
     status, response = insert_request.next_chunk()
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^
   File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/googleapiclient/_helpers.py", line 130, in positional_wrapper
     return wrapped(*args, **kwargs)
            ^^^^^^^^^^^^^^^^^^^^^^^^
   File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/googleapiclient/http.py", line 1007, in next_chunk
     resp, content = _retry_request(
                     ^^^^^^^^^^^^^^^
   File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/googleapiclient/http.py", line 191, in _retry_request
     resp, content = http.request(uri, method, *args, **kwargs)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/google_auth_httplib2.py", line 245, in request
     self.credentials.refresh(self._request)
   File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/google/oauth2/credentials.py", line 397, in refresh
     raise exceptions.RefreshError(
 The credentials do not contain the necessary fields need to refresh the access token. You must specify refresh_token, token_uri, client_id, and client_secret.
```

I say 'mostly', because it _has_ successfully performed an upload when manually initiated [three](https://www.youtube.com/watch?v=y4MyYZ9juIk) [times](https://www.youtube.com/watch?v=7Uwsgg_A24A) [before](https://www.youtube.com/watch?v=vowHXBVHx1I). All were manual though and this issue seems to have cropped up once I turned on automatic uploads, but that may be coincidental.

Any pointers?

omeganon, 2025-06-26:
```
/home/admin/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 8257680 kB

System: Raspberry Pi 5 Model B Rev 1.0

Linux allsky2 6.6.74+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux
Hostname
allsky2
allsky2.home.local 

Time
Thu 26 Jun 14:27:06 CDT 2025

System timezone
America/Chicago

Uptime
 14:27:06 up 2 days, 23:16,  3 users,  load average: 0.43, 0.46, 0.45

Memory
               total        used        free      shared  buff/cache   available
Mem:         8257680     2087120      985152      167824     5836432     6170560
Swap:         524272      233712      290560

Filesystems
Filesystem      1K-blocks      Used  Available Use% Mounted on
udev              4087040         0    4087040   0% /dev
tmpfs              825776     83328     742448  11% /run
/dev/nvme0n1p2 1921205552 217622160 1605917700  12% /
tmpfs             4128832      1024    4127808   1% /dev/shm
tmpfs                5120        48       5072   1% /run/lock
/dev/nvme0n1p1     523248     70004     453244  14% /boot/firmware
tmpfs              524288     34288     490000   7% /tmp
tmpfs              825760       208     825552   1% /run/user/1000
/dev/mmcblk0p2  122624044  22714140   93661328  20% /media/admin/rootfs1
/dev/mmcblk0p1     522230     69378     452852  14% /media/admin/bootfs1

systemctl runlevel
multi-user.target

sysctl info
vm.swappiness = 1

Thermal info
cpu-thermal
80950

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

Locale info
LANG=en_GB.UTF-8
LANGUAGE=
LC_CTYPE="en_GB.UTF-8"
LC_NUMERIC="en_GB.UTF-8"
LC_TIME="en_GB.UTF-8"
LC_COLLATE="en_GB.UTF-8"
[... 1718 lines cut ...]
2025-06-26T14:02:14.123035-05:00 allsky2 [ERROR] MainProcess-94885/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2025-06-26T14:02:14.123065-05:00 allsky2 [ERROR] MainProcess-94885/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:   File "/home/admin/indi-allsky/indi_allsky/filetransfer/youtube_oauth2.py", line 139, in resumable_upload
2025-06-26T14:02:14.123086-05:00 allsky2 [ERROR] MainProcess-94885/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:     status, response = insert_request.next_chunk()
2025-06-26T14:02:14.123106-05:00 allsky2 [ERROR] MainProcess-94885/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^
2025-06-26T14:02:14.123126-05:00 allsky2 [ERROR] MainProcess-94885/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:   File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/googleapiclient/_helpers.py", line 130, in positional_wrapper
2025-06-26T14:02:14.123147-05:00 allsky2 [ERROR] MainProcess-94885/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:     return wrapped(*args, **kwargs)
2025-06-26T14:02:14.123167-05:00 allsky2 [ERROR] MainProcess-94885/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:            ^^^^^^^^^^^^^^^^^^^^^^^^
2025-06-26T14:02:14.123186-05:00 allsky2 [ERROR] MainProcess-94885/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:   File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/googleapiclient/http.py", line 1007, in next_chunk
2025-06-26T14:02:14.123206-05:00 allsky2 [ERROR] MainProcess-94885/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:     resp, content = _retry_request(
2025-06-26T14:02:14.123225-05:00 allsky2 [ERROR] MainProcess-94885/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:                     ^^^^^^^^^^^^^^^
2025-06-26T14:02:14.123244-05:00 allsky2 [ERROR] MainProcess-94885/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:   File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/googleapiclient/http.py", line 191, in _retry_request
2025-06-26T14:02:14.123264-05:00 allsky2 [ERROR] MainProcess-94885/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:     resp, content = http.request(uri, method, *args, **kwargs)
2025-06-26T14:02:14.123284-05:00 allsky2 [ERROR] MainProcess-94885/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2025-06-26T14:02:14.123303-05:00 allsky2 [ERROR] MainProcess-94885/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:   File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/google_auth_httplib2.py", line 245, in request
2025-06-26T14:02:14.123323-05:00 allsky2 [ERROR] MainProcess-94885/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:     self.credentials.refresh(self._request)
2025-06-26T14:02:14.123342-05:00 allsky2 [ERROR] MainProcess-94885/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:   File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/google/oauth2/credentials.py", line 397, in refresh
[... cut ...]
2025-06-26T14:02:14.123381-05:00 allsky2 [ERROR] MainProcess-94885/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception: google.auth.exceptions.RefreshError: The credentials do not contain the necessary fields need to refresh the access token. You must specify refresh_token, token_uri, client_id, and client_secret.
2025-06-26T14:02:14.123402-05:00 allsky2 [ERROR] MainProcess-94885/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception: 
```

#################################
###     end support info      ###
#################################
```

aaronwmorris, 2025-06-27:
You may want to Revoke and retry the authorization.  It is not really clear to me what is going on.

omeganon, 2025-06-27:
I actually tried that but it fails. I tried with all possible aliases for the box, including the one I originally set up YouTube with --

https://fakeallsky2.com/indi-allsky/youtube/oauth2revoke

```
Bad Request

Something went wrong
```

I can't find any log that actually says why. Is gunicorn/flask logging somewhere that I haven't found? Nothing is logged to syslog, apache, or indi-allsky logs.

Is there a manual way to do it? If it's helpful, I'm familiar with SQL and know how to access the SQLite database...

aaronwmorris, 2025-06-27:
You may be able to alter the code to do this:

```
        status_code = getattr(revoke, 'status_code')
        if status_code != 200:
            abort(400, 'Something went wrong: {0}'.format(revoke.text.encode()))
```

omeganon, 2025-06-27:
Interesting...

```
Something went wrong: b'{\n "error": "invalid_token",\n "error_description": "Token is not revocable."\n}'
```

I'll just delete the credential in the admin console, create a new one and apply it...

omeganon, 2025-06-28:
Meh. Automatic day and manual uploads failing with the same error. 

```

2025-06-27T21:51:20.647221-05:00 allsky2 [ERROR] MainProcess-1165/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception: 
2025-06-27T21:51:20.647189-05:00 allsky2 [ERROR] MainProcess-1165/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception: google.auth.exceptions.RefreshError: The credentials do not contain the necessary fields need to refresh the access token. You must specify refresh_token, token_uri, client_id, and client_secret.
2025-06-27T21:51:20.647158-05:00 allsky2 [ERROR] MainProcess-1165/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:     raise exceptions.RefreshError(
2025-06-27T21:51:20.647122-05:00 allsky2 [ERROR] MainProcess-1165/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:   File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/google/oauth2/credentials.py", line 397, in refresh
2025-06-27T21:51:20.647092-05:00 allsky2 [ERROR] MainProcess-1165/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:     self.credentials.refresh(self._request)
2025-06-27T21:51:20.647061-05:00 allsky2 [ERROR] MainProcess-1165/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:   File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/google_auth_httplib2.py", line 245, in request
2025-06-27T21:51:20.647031-05:00 allsky2 [ERROR] MainProcess-1165/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2025-06-27T21:51:20.646996-05:00 allsky2 [ERROR] MainProcess-1165/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:     resp, content = http.request(uri, method, *args, **kwargs)
2025-06-27T21:51:20.646930-05:00 allsky2 [ERROR] MainProcess-1165/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:   File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/googleapiclient/http.py", line 191, in _retry_request
2025-06-27T21:51:20.646899-05:00 allsky2 [ERROR] MainProcess-1165/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:                     ^^^^^^^^^^^^^^^
2025-06-27T21:51:20.646867-05:00 allsky2 [ERROR] MainProcess-1165/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:     resp, content = _retry_request(
2025-06-27T21:51:20.646835-05:00 allsky2 [ERROR] MainProcess-1165/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:   File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/googleapiclient/http.py", line 1007, in next_chunk
2025-06-27T21:51:20.646804-05:00 allsky2 [ERROR] MainProcess-1165/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:            ^^^^^^^^^^^^^^^^^^^^^^^^
2025-06-27T21:51:20.646773-05:00 allsky2 [ERROR] MainProcess-1165/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:     return wrapped(*args, **kwargs)
2025-06-27T21:51:20.646742-05:00 allsky2 [ERROR] MainProcess-1165/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:   File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/googleapiclient/_helpers.py", line 130, in positional_wrapper
2025-06-27T21:51:20.646710-05:00 allsky2 [ERROR] MainProcess-1165/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^
2025-06-27T21:51:20.646680-05:00 allsky2 [ERROR] MainProcess-1165/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:     status, response = insert_request.next_chunk()
2025-06-27T21:51:20.646648-05:00 allsky2 [ERROR] MainProcess-1165/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:   File "/home/admin/indi-allsky/indi_allsky/filetransfer/youtube_oauth2.py", line 139, in resumable_upload
2025-06-27T21:51:20.646618-05:00 allsky2 [ERROR] MainProcess-1165/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2025-06-27T21:51:20.646588-05:00 allsky2 [ERROR] MainProcess-1165/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:     response = self.resumable_upload(insert_request)
2025-06-27T21:51:20.646557-05:00 allsky2 [ERROR] MainProcess-1165/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:   File "/home/admin/indi-allsky/indi_allsky/filetransfer/youtube_oauth2.py", line 116, in put
2025-06-27T21:51:20.646526-05:00 allsky2 [ERROR] MainProcess-1165/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:                ^^^^^^^^^^^^^^^^^^^^^^^^
2025-06-27T21:51:20.646495-05:00 allsky2 [ERROR] MainProcess-1165/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:     response = client.put(**put_kwargs)
2025-06-27T21:51:20.646453-05:00 allsky2 [ERROR] MainProcess-1165/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:   File "/home/admin/indi-allsky/indi_allsky/uploader.py", line 472, in processUpload
2025-06-27T21:51:20.646422-05:00 allsky2 [ERROR] MainProcess-1165/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:     self.processUpload(u_dict)
[... cut ...]
2025-06-27T21:51:20.646357-05:00 allsky2 [ERROR] MainProcess-1165/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:     self.saferun()
2025-06-27T21:51:20.646321-05:00 allsky2 [ERROR] MainProcess-1165/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception:   File "/home/admin/indi-allsky/indi_allsky/uploader.py", line 108, in run
2025-06-27T21:51:20.646235-05:00 allsky2 [ERROR] MainProcess-1165/MainThread allsky._fileUploadWorkerStart() [595]: Upload worker exception: Traceback (most recent call last
```

omeganon, 2025-06-28:
Something has to be broken about that refresh token. I’ll do some debugging tomorrow to see what it looks like.

aaronwmorris, 2025-06-28:
Try revoking the token from https://myaccount.google.com/u/0/permissions and re-adding it.

I agree something funky is going on.

omeganon, 2025-06-28:
Before trying to revoke the credential as above, refresh_token is definitely null at https://github.com/aaronwmorris/indi-allsky/blob/main/indi_allsky/filetransfer/youtube_oauth2.py#L47-L49

```
cat /tmp/credentials.json 
{
    "token": "ya29.a0AS3H6Ny-7<redacted>Lew0175",
    "refresh_token": null,
    "token_uri": "https://oauth2.googleapis.com/token",
    "client_id": "74187<redacted>.apps.googleusercontent.com",
    "client_secret": "GOCSPX<redacted>",
    "scopes": [
        "https://www.googleapis.com/auth/youtube.upload"
    ]
}
```

Revoked access and a re-auth gets me a 500 so I'll go through the full setup again.

omeganon, 2025-06-28:
After re-auth, refresh_token _is_ populated at the point of upload. Will check again tomorrow.

```
cat /tmp/credentials.json 
{
    "token": "ya29.<redacted>0g0175",
    "refresh_token": "1//01hIP<redacted>AQ",
    "token_uri": "https://oauth2.googleapis.com/token",
    "client_id": "74187<redacted>.apps.googleusercontent.com",
    "client_secret": "GOCSPX<redacted>",
    "scopes": [
        "https://www.googleapis.com/auth/youtube.upload"
    ]
}
```

omeganon, 2025-06-29:
Now the credentials are failing on the other camera. It seems like the first to authorize works, the second won't at refresh. Current Google cloud configuration is both credentials under one project. I'll now try distinct projects for each.

[image]

omeganon, 2025-06-29:
Seems like multiple Client IDs under one project _should_ have worked if we consider each camera a different platform... https://cloud.google.com/application-integration/docs/configure-authentication-profiles#:~:text=Create%20OAuth%202.&text=A%20client%20ID%20is%20used,to%20APIs%20&%20Services%20%3E%20Credentials.&text=Click%20+%20Create%20Credentials%20and%20select,See%20more%20code%20actions.&text=Click%20Create.,client%20ID%20is%20created%20successfully.

Google AI summary says 

```
Multiple Tokens and Scopes:
While multiple access tokens aren't the standard for a single application, you might encounter situations where you need to handle different scopes (permissions). Google recommends incremental authorization, requesting scopes as needed, rather than all at once. If you need to access different sets of resources with different permissions, you might consider using separate OAuth clients or handling multiple authorization flows, but this is less common than managing access with a single token and refresh token. 

Managing Multiple Clients:
If you have multiple web applications or platforms (e.g., web and mobile), each should have its own OAuth client ID. Each client will have its own set of tokens (access and refresh) associated with it. 
```

omeganon, 2025-06-30:
Successful automatic uploads from both this morning. It seems like there is _something_ that is common between those two sets of credentials in a single project that causes the refresh_token for one of them to be discarded.

Could be an interesting caveat for the Youtube setup wiki.

aaronwmorris, 2025-06-30:
Is the only recommendation to setup the credentials in separate projects in GCP?

omeganon, 2025-06-30:
Yes, that seems sufficient.

closed 2025-07-01 by omeganon

aaronwmorris, 2025-07-01:
I have updated the wiki.  Thank you for all the testing you performed!