aaronwmorris/indi-allsky #2228: ASI715MC not detected

p088gll, 2025-07-03:
I have been using indi-allsky with an ASI120MC-S for quite a while. Recently I got hold of an ASI715MC and since the chip IMX715  is mentioned  here https://github.com/aaronwmorris/indi-allsky/wiki/Camera-Information, I though it might be supported. However, even after re-running ./setup.sh I am getting a traceback of /home/pi/indi-allsky/indi_allsky/camera/indi.py and the message 'No cameras found':

```
Jul  3 22:22:17 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception: Traceback (most recent call last):
Jul  3 22:22:17 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:   File "/home/pi/indi-allsky/indi_allsky/camera/indi.py", line 645, in findCcd
Jul  3 22:22:17 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:     self.ccd_device = ccd_list[0]
Jul  3 22:22:17 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception: IndexError: list index out of range
Jul  3 22:22:17 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:
Jul  3 22:22:17 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception: During handling of the above exception, another exception occurred:
Jul  3 22:22:17 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:
Jul  3 22:22:17 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception: Traceback (most recent call last):
Jul  3 22:22:17 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:   File "/home/pi/indi-allsky/indi_allsky/capture.py", line 254, in run
Jul  3 22:22:17 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:     self.saferun()
Jul  3 22:22:17 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:   File "/home/pi/indi-allsky/indi_allsky/capture.py", line 266, in saferun
Jul  3 22:22:17 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:     self._initialize()
Jul  3 22:22:17 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:   File "/home/pi/indi-allsky/indi_allsky/capture.py", line 713, in _initialize
Jul  3 22:22:17 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:     self.indiclient.findCcd(camera_name=self.config.get('INDI_CAMERA_NAME'))
Jul  3 22:22:17 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:   File "/home/pi/indi-allsky/indi_allsky/camera/indi.py", line 647, in findCcd
Jul  3 22:22:17 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:     raise CameraException('No cameras found')
Jul  3 22:22:17 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception: indi_allsky.exceptions.CameraException: No cameras found
Jul  3 22:22:17 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:
Jul  3 22:22:17 sky [INFO] MainProcess-4129/MainThread allsky._startCaptureWorker() [363]: Starting Capture-10 worker
```

So I wonder if the AASI715MC is supposed to work. At least it is connected and found by Debian OS.
```
/home/pi/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 11
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 3885396 kB

System: Raspberry Pi 4 Model B Rev 1.2

Linux sky 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
Hostname
sky

Time
Thu 03 Jul 2025 10:36:13 PM CEST

System timezone
Europe/Berlin

Uptime
 22:36:13 up  2:37,  1 user,  load average: 0.00, 0.00, 0.05

Memory
               total        used        free      shared  buff/cache   available
Mem:         3885396      323880     1725544       18132     1835972     3466168
Swap:         614396           0      614396

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/root       61330152 33987428  24817888  58% /
devtmpfs         1678472        0   1678472   0% /dev
tmpfs            1942696      156   1942540   1% /dev/shm
tmpfs             777080    17484    759596   3% /run
tmpfs               5120        4      5116   1% /run/lock
/dev/mmcblk0p1    261108    31374    229734  13% /boot
tmpfs             388536        0    388536   0% /run/user/1000

[... 1341 lines cut ...]
Jul  3 22:35:21 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception: During handling of the above exception, another exception occurred:
Jul  3 22:35:21 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception: 
Jul  3 22:35:21 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception: Traceback (most recent call last):
Jul  3 22:35:21 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:   File "/home/pi/indi-allsky/indi_allsky/capture.py", line 254, in run
Jul  3 22:35:21 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:     self.saferun()
Jul  3 22:35:21 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:   File "/home/pi/indi-allsky/indi_allsky/capture.py", line 266, in saferun
Jul  3 22:35:21 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:     self._initialize()
Jul  3 22:35:21 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:   File "/home/pi/indi-allsky/indi_allsky/capture.py", line 713, in _initialize
Jul  3 22:35:21 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:     self.indiclient.findCcd(camera_name=self.config.get('INDI_CAMERA_NAME'))
Jul  3 22:35:21 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:   File "/home/pi/indi-allsky/indi_allsky/camera/indi.py", line 647, in findCcd
Jul  3 22:35:21 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:     raise CameraException('No cameras found')
Jul  3 22:35:21 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception: indi_allsky.exceptions.CameraException: No cameras found
Jul  3 22:35:21 sky [ERROR] MainProcess-4129/MainThread allsky._startCaptureWorker() [356]: Capture worker exception: 
Jul  3 22:35:26 sky [ERROR] Capture-22-4278/MainThread capture._initialize() [715]: Camera error: !!! NO CAMERAS FOUND !!!
```

#################################
###     end support info      ###
#################################

```

I am more than happy to provide more debug information if thats helpful.

Goetz

aaronwmorris, 2025-07-04:
I suspect the version of indilib you have installed it too old (1.9.7) to support the ASI715MC.  You will need to remove your OS indilib packages and build indilib from source.

These links might help:

https://github.com/aaronwmorris/indi-allsky/wiki/Uninstall-indi-repository

https://github.com/aaronwmorris/indi-allsky/wiki/Getting-Started#compiling-indi

And once you have a newer version of indilib, you should rebuild the pyindi-client python module.

```
./misc/rebuild_pyindi-client.sh
```

p088gll, 2025-07-04:
Thanks for the suggestion. Additionally I had to change the path of the indiserver in /home/pi/.config/systemd/user/indiserver.service from /usr/bin to /usr/local/bin.
But still no change, the errors remains:
```
2025-07-05T00:03:03.266016+02:00 sky [INFO] MainProcess-830/MainThread allsky._startCaptureWorker() [363]: Starting Capture-17 worker
2025-07-05T00:03:03.298720+02:00 sky [INFO] Capture-17-1350/MainThread indi.__init__() [129]: creating an instance of IndiClient
2025-07-05T00:03:03.299550+02:00 sky [INFO] Capture-17-1350/MainThread indi.__init__() [136]: PyIndi version: 2.1.4
2025-07-05T00:03:03.300833+02:00 sky [INFO] Capture-17-1350/MainThread capture._initialize() [690]: Connecting to indiserver
2025-07-05T00:03:03.301462+02:00 sky [INFO] Capture-17-1350/MainThread indi.serverConnected() [352]: Server connected (localhost:7624)
2025-07-05T00:03:03.308538+02:00 sky [INFO] Capture-17-1350/Dummy-2 indi.newDevice() [224]: new device Telescope Simulator
2025-07-05T00:03:08.303956+02:00 sky [INFO] Capture-17-1350/MainThread indi._findCcds() [611]: Searching for available cameras
2025-07-05T00:03:08.304823+02:00 sky [INFO] Capture-17-1350/MainThread indi.findCcd() [631]: Found 0 CCDs
2025-07-05T00:03:08.305349+02:00 sky [ERROR] Capture-17-1350/MainThread capture._initialize() [715]: Camera error: !!! NO CAMERAS FOUND !!!
2025-07-05T00:03:08.395253+02:00 sky [WARNING] Capture-17-1350/MainThread miscDb.addNotification() [1018]: Not adding existing notification

2025-07-05T00:05:32.763018+02:00 sky [ERROR] MainProcess-830/MainThread allsky._startCaptureWorker() [356]: Capture worker exception: Traceback (most recent call last):
2025-07-05T00:05:32.763725+02:00 sky [ERROR] MainProcess-830/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:   File "/home/pi/indi-allsky/indi_allsky/camera/indi.py", line 645, in findCcd
2025-07-05T00:05:32.764274+02:00 sky [ERROR] MainProcess-830/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:     self.ccd_device = ccd_list[0]
2025-07-05T00:05:32.764717+02:00 sky [ERROR] MainProcess-830/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:                       ~~~~~~~~^^^
2025-07-05T00:05:32.765147+02:00 sky [ERROR] MainProcess-830/MainThread allsky._startCaptureWorker() [356]: Capture worker exception: IndexError: list index out of range
2025-07-05T00:05:32.765570+02:00 sky [ERROR] MainProcess-830/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:
2025-07-05T00:05:32.766135+02:00 sky [ERROR] MainProcess-830/MainThread allsky._startCaptureWorker() [356]: Capture worker exception: During handling of the above exception, another exception occurred:
2025-07-05T00:05:32.766607+02:00 sky [ERROR] MainProcess-830/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:
2025-07-05T00:05:32.767120+02:00 sky [ERROR] MainProcess-830/MainThread allsky._startCaptureWorker() [356]: Capture worker exception: Traceback (most recent call last):
2025-07-05T00:05:32.767642+02:00 sky [ERROR] MainProcess-830/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:   File "/home/pi/indi-allsky/indi_allsky/capture.py", line 254, in run
2025-07-05T00:05:32.768150+02:00 sky [ERROR] MainProcess-830/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:     self.saferun()
2025-07-05T00:05:32.768792+02:00 sky [ERROR] MainProcess-830/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:   File "/home/pi/indi-allsky/indi_allsky/capture.py", line 266, in saferun
2025-07-05T00:05:32.769441+02:00 sky [ERROR] MainProcess-830/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:     self._initialize()
2025-07-05T00:05:32.770145+02:00 sky [ERROR] MainProcess-830/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:   File "/home/pi/indi-allsky/indi_allsky/capture.py", line 713, in _initialize
2025-07-05T00:05:32.770723+02:00 sky [ERROR] MainProcess-830/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:     self.indiclient.findCcd(camera_name=self.config.get('INDI_CAMERA_NAME'))
2025-07-05T00:05:32.771279+02:00 sky [ERROR] MainProcess-830/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:   File "/home/pi/indi-allsky/indi_allsky/camera/indi.py", line 647, in findCcd
2025-07-05T00:05:32.771814+02:00 sky [ERROR] MainProcess-830/MainThread allsky._startCaptureWorker() [356]: Capture worker exception:     raise CameraException('No cameras found')
2025-07-05T00:05:32.772313+02:00 sky [ERROR] MainProcess-830/MainThread allsky._startCaptureWorker() [356]: Capture worker exception: indi_allsky.exceptions.CameraException: No cameras found
```

So maybe I should give up at this point ?

aaronwmorris, 2025-07-04:
I think it is still running against the old version of Indi.  The path to indiserver is /usr/bin instead of /usr/local/bin.  Try re-running setup.sh and reconfigure the indiserver service.

aaronwmorris, 2025-07-04:
It could be picking up the Indi drivers from /usr/bin.  You may want to manually delete them. 

```
sudo mv /usr/bin/indi_asi_ccd /usr/bin/indi_asi_ccd.old

p088gll, 2025-07-05:
Nope, I removed the indi packages, so there was nothing left like /usr/bin/indi*
But anyway, I  reinstalled the raspberry with bookworm, compiled INDI and ran ./setup.sh and now the ASI 715MC camera is detected.

closed 2025-07-05 by p088gll