aaronwmorris/indi-allsky #2252: libcamera '--immediate' breaks 'libcamera extra options' --autofocus-mode

warwound, 2025-07-21:
I've just clean installed Bookworm on my 4GB Pi4, installed indi-allsky then installed the driver files for the Arducam Hawkeye .
Everything works except for the Hawkeye auto focus commands, they seem to do nothing.

For day and night libcamera extra options I have tried:
--autofocus-mode continuous
--autofocus-mode auto
--autofocus-mode manual --lens-position 0.0
--autofocus-mode manual --lens-position 1.0
and many more...

They all seem to silently fail.
I found the indi-allsky libcamera capture command in the logs and experimented with it manually in a terminal.
That (now modified) command is:

rpicam-still --immediate --nopreview --camera 0 --encoding jpg --quality 95 --gain 1 --shutter 4219 --awb auto --autofocus-mode auto --output ~/Desktop/capture.jpg

And again it fails, however if I remove the --immediate option the command works and I get a perfectly focused picture:
rpicam-still --nopreview --camera 0 --encoding jpg --quality 95 --gain 1 --shutter 4219 --awb auto --autofocus-mode auto --output ~/Desktop/capture.jpg

The Raspberry Pi docs state: "Because long exposures take plenty of time already, it often makes sense to skip the preview phase entirely with the [immediate](https://www.raspberrypi.com/documentation/computers/camera_software.html#immediate) option."

I assume that's why indi-allsky always uses the immediate option?
Is there a way for me to disable the immediate option?

Thanks

aaronwmorris, 2025-07-21:
You would just need to edit the code to remove the `--immediate` flag.

https://github.com/aaronwmorris/indi-allsky/blob/main/indi_allsky/camera/libcamera.py#L170

I guessing that you are running this during the day and autofocus will likely work while there is sufficient illumination, however, it will probably fail at night.  You could use the autofocus during the day to find the correct focus and just use that value with manual focus after the fact.

warwound, 2025-07-22:
Removing the 'immediate' option from the libcamera command in libcamera.py works perfectly and my Hawkeye now focuses in my daytime tests.

For night time usage I'm hoping to use:
--autofocus-mode manual --lens-position 0.0
Which should pre-focus to infinity.

I wonder if using the 'immediate' option could be made into a configuration setting in indi-allsky?
Allow the user to enable or disable the 'immediate' option instead of having to edit the libcamera.py file.

Thanks for your help.

aaronwmorris, 2025-07-23:
Merged #2263 to allow `--immediate` to be disabled.

warwound, 2025-07-23:
Thanks, I'll now close this issue.

closed 2025-07-23 by warwound