aaronwmorris/indi-allsky #2271: indi-pylibcamera darks tempsigmaclip issue

masterzen, 2025-07-27:
In the spirit of https://github.com/aaronwmorris/indi-allsky/issues/1130, when using indi-pylibcamera as the  driver for a Raspberry Pi HQ Camera (I know, I like complexity), the temperature is only refreshed when taking a picture.

This means:
* the first value of `next_temp_thold` (set at https://github.com/aaronwmorris/indi-allsky/blob/2084ec27015d2e13ae8685be98529e511345ca47/indi_allsky/darks.py#L650) is negative because `self.sensors_temp_av[0]` is 0 - this prevents taking pictures at temperature above -5ºC
* any call to `getCcdTemperature` always returns the same value, _except_ after taking dark frames. The problem is that indi-pylibcamera only sets the temperature in the metadata after taking a shot (see https://github.com/scriptorron/indi_pylibcamera/blob/75b8126742702295d41c5d87cc0b4bb427276c0f/src/indi_pylibcamera/CameraControl.py#L923). This has the unfortunate effect of not taking any more dark frames after the first shot, because even if the camera temperature decreases it's still reporting the same value.

I was able to work around this by forcing taking a 0.5s shot before calling `getCcdTemperature`, which is probably not what you want to do for all kind of cameras (this seems to be very specific to indi-pylibcamera).

It would be great to have an option to fire such "metadata only" shot for indi-pylibcamera. I'm not sure it's easy to detect that the driver is indi-pylibcamera because on my setup the driver is reported as the name of the camera I set in `~/.indi_pylibcamera/indi_pylibcamera.ini`.

Here's my local patch if you want to take inspiration for a real fix:

```diff
diff --git a/indi_allsky/darks.py b/indi_allsky/darks.py
index 5cca4718..61059884 100644
--- a/indi_allsky/darks.py
+++ b/indi_allsky/darks.py
@@ -617,23 +617,29 @@ class IndiAllSkyDarks(object):

         self._pre_run_tasks()

+        self._preshoot()
         self.getCcdTemperature()
+        logger.info('Camera temperature: %0.2f', self.sensors_temp_av[0])
         next_temp_thold = self.sensors_temp_av[0] - self.temp_delta
+        logger.info('Next threshold: %0.2f', next_temp_thold)

         # get first set of images
         self._run(IndiAllSkyDarksSigmaClip)

         while True:
             # This loop will run forever, it is up to the user to cancel
+            self._preshoot()
+            logger.info('Camera temperature: %0.2f', self.sensors_temp_av[0])
             self.getCcdTemperature()

+            logger.info('Camera temperature: %0.2f', self.sensors_temp_av[0])
             logger.info('Next temperature threshold: %0.1f', next_temp_thold)

             if self.sensors_temp_av[0] > next_temp_thold:
-                time.sleep(20.0)
+                time.sleep(60.0)
                 continue

-            logger.warning('Acheived next temperature threshold')
+            logger.warning('Achieved next temperature threshold: %0.1f | %0.1f', next_temp_thold, self.temp_delta)
             next_temp_thold -= self.temp_delta

             self._run(IndiAllSkyDarksSigmaClip)
@@ -682,6 +688,19 @@ class IndiAllSkyDarks(object):

             filename.unlink()  # no longer need the original file

+    def _preshoot(self):
+        # Tasks that need to be run before the main program loop
+
+        logger.info('Taking throw away exposure for rpicam')
+        self.shoot(0.5, sync=True, timeout=20.0)
+        i_dict = self.image_q.get(timeout=10)
+        filename = Path(i_dict['filename'])
+        if not filename.exists():
+            #task.setFailed('Frame not found: {0:s}'.format(str(filename)))
+            raise Exception('Frame not found {0:s}'.format(str(filename)))
+
+        filename.unlink()  # no longer need the original file
+

     def _pre_shoot_reconfigure(self):
         if self.camera_server in ['indi_asi_ccd']:
```

And last point, indi-pylibcamera doesn't return any value for `CCD_CFA`, and the darks.py output mention thinking it's going to assume the images are mono (which is not the case). I haven't yet checked the produced FITS image, but is it going to create an issue for dark processing down the line?

Thanks!

aaronwmorris, 2025-07-28:
I understand the temperature issue.  I will take a look at this.

You can ignore the message about CCD_CFA.  In the early days of indi-allsky, I used the data from the CCD_CFA control as input for debayering, but that is no longer the case.  The `CFA_PATTERN` FITS header is always used now.

aaronwmorris, 2025-08-13:
Merged #2305 to take throw away frames to update the temperature of libcamera managed cameras.