aaronwmorris/indi-allsky #2274: startrail problem...

marco050969, 2025-07-28:
no startrails more after three days...

[indi-allsky_support_info_2025728_12843.txt](https://github.com/user-attachments/files/21466641/indi-allsky_support_info_2025728_12843.txt)

marco050969, 2025-07-28:
Just for your information!!

I updated indi allsky on July 25, 2025, and since then, no star trails or star trail timelapses have been generated. Everything worked fine before.

aaronwmorris, 2025-07-28:
All of your images were excluded from the startrails for being too bright (ADU).

Can you post your Support Info?

aaronwmorris, 2025-07-28:
You know... I should have seen that.  I even read through it before I posted that.. :-D

Your target adu at night is 100, but the startrails max adu is 65.  I would increase the startrails max adu to 90-100.

marco050969, 2025-07-28:
Okay I do and test this night…

aaronwmorris, 2025-07-28:
You can always delete and try to regenerate a previous night's trails in the meantime.

marco050969, 2025-07-29:
Don’t work….

[image]

aaronwmorris, 2025-07-31:
I have a quick question, can you go into the gallery an look at the original frames to see if it is in the original picture?

marco050969, 2025-07-31:
The original frames are free of stripes…also black…

marco050969, 2025-07-31:
[image]

aaronwmorris, 2025-07-31:
I will keep looking....

aaronwmorris, 2025-07-31:
Would you be able to send me the original image (that ends up with the lines) and the original keogram that were used?  I might be able to figure out what is happening with the specific files that have the issue.

aaronwmorris, 2025-08-01:
I have done some research and I may have discovered the cause.  Since I cannot recreate the issue, I will have to rely on your testing to see if the issue is resolved.

Merged #2284 

Basically, what I found is that some functions of OpenCV do not initialize the contents of an image outside the visible area.  Therefore, you can end up with random data in the buffer that leaves trash after the transformation.  The transformation that wraps the keogram around the image effectively creates some blank areas that were not part of the original image.  I added that should initialize that data as zeros (black, which would be transparent).

marco050969, 2025-08-01:
> Would you be able to send me the original image (that ends up with the lines) and the original keogram that were used? I might be able to figure out what is happening with the specific files that have the issue.

Here is the link to my Cam: https://ekbyasodr61r2xgg.myfritz.net:8484/indi-allsky/

 I don’t know what you mean with pictures with lines. The original pictures ara all without lines in the black area. The stripes comes, when keogram is produced.

marco050969, 2025-08-02:
Great Job Aaron. The Problem with Stripes is gone...!!!!

[image]

closed 2025-08-24 by aaronwmorris