aaronwmorris/indi-allsky #2279: pyindi displayed in indi-allsky/system webpage

leforet, 2025-07-29:
not sure if I'm looking in the right place.  I've noticed that at  https://192.168.1.232/indi-allsky/system whenn looking at the versions of the modules installed.    I noticed thata pyindi release is set to 2.0.7, but this is different than the installed version of 2.1.3 (see below)

System
Python
3.11.2
gunicorn
22.0.0
cryptography
44.0.1
**pyindi
2.0.7**

This is after the pyindi-client was updated to 2.1.3 via the instructions at https://github.com/aaronwmorris/indi-allsky/wiki/Getting-Started

when I updated  the version of indisky via the instructions at https://github.com/aaronwmorris/indi-allsky/wiki/Updating-indi-allsky

while the setup.sh script ran, there was this line

> Detected **INDI version: 2.1.3**
> Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
> Collecting pyindi-client
>   Cloning https://github.com/indilib/pyindi-client.git (to revision d8ad88f) to /tmp/pip-install-91ztavxy/pyindi-client_a410c7a0e4df455ca2b08f862b167d5f
>   Running command git clone --filter=blob:none --quiet https://github.com/indilib/pyindi-client.git /tmp/pip-install-91ztavxy/pyindi-client_a410c7a0e4df455ca2b08f862b167d5f
>   WARNING: Did not find branch or tag 'd8ad88f', assuming revision or ref.
>   Running command git checkout -q d8ad88f
>   Resolved https://github.com/indilib/pyindi-client.git to commit d8ad88f
>   Preparing metadata (setup.py) ... done

it seems that indi-client version is 2.1.3 and client displayed on https://192.168.1.232/indi-allsky/system is 2.0.7?

Am I missing something?

aaronwmorris, 2025-07-29:
The "version" reported by pyindi-client is the version of INDI it was compiled against.  There is a rebuild script available to update it.

```
./misc/rebuild_pyindi-client.sh
```

leforet, 2025-07-29:
thanks, I'm now on a newer version
Can I make a suggestion to add a paragraph to https://github.com/aaronwmorris/indi-allsky/wiki/Updating-indi-allsky about also how to update the pyindi-client  and indi with the script?

closed 2025-08-24 by aaronwmorris