aaronwmorris/indi-allsky #2293: Change URL path where indi-allysky is served

campbellbartlett, 2025-08-09:
Hey,

Loving indi-allsky so far, you've put in a lot of work and it's really impressive.

Is it possible to change the path where indi-allsky is served? I have a vanilla install on a new raspberry pi. I can see that it's using an apache2 webserver.

I'm running a reverse proxy on my local network (nginx proxy manager) to proxy allsky to `https://allsky.myDomain.tld/` and some devices are unable to load anything from the `/static/` directory when I access indi-allsky via this address. I'm assuming it has something to do with the path, all the other applications that I host (on a different machine) are served at `/` on their respective ports and they work fine.

I can't work out how to change it, I did manage to disable the redirect from `/` to `/indi-allysky/` in the apache2 config (in case that was causing the issue, but no luck).

Another theory I have is that it could be the self signed certificate that's causing problems. I don't need that (or enforcement of https) because the reverse proxy provides a wildcard cert for my domain. Is it possible to switch this off and serve the application over http (or even choose my own port)?

I don't have a lot of experience working with apache2 or gunicorn/flask etc so I'm not really sure where to start or what to look at. 

Thank you,
Campbell

aaronwmorris, 2025-08-09:
Sure, this is fairly simple.  I have a page that describes how to change the context root.

https://github.com/aaronwmorris/indi-allsky/wiki/Change-root-context

campbellbartlett, 2025-08-10:
Nice thank you for your help. I managed to get that working and to disable HTTPS.

Interestingly it still wasn't working on iOS, so I also had to disable HTTP/2 support in Nginx proxy manager too. This seems to be a bug with nginx, not something broken with indi-allsky.

evripidis, 2025-08-10:
> Hey,
>
> Loving indi-allsky so far, you've put in a lot of work and it's really impressive.
>
> Is it possible to change the path where indi-allsky is served? I have a vanilla install on a new raspberry [...]

Hello @campbellbartlett 

Would  you mind mention any specific tip about that? I have a NPM setup too (not on the same network as allsky), I already run 9 others proxy hosts in there with no issue. I do have an active proxy host for the specific allsky machine too (for my nomachine connection). I even tried a domain name I own to redirect back on the allsky setup. All failed! The only thing that works outside the network is "DDNS url:PORT/indi-allsky/"

evripidis, 2025-08-11:
@campbellbartlett Thank you very much for your details! That thing was driving me crazy!

campbellbartlett, 2025-08-12:
Yep - however if you're using port 80 or 443, you don't need to specify the port - http defaults to port 80 and https defaults to port 443.

aaronwmorris, 2025-08-24:
Closing this for now.  Feel free to re-open if more info is needed.

closed 2025-08-24 by aaronwmorris