aaronwmorris/indi-allsky #2302: Temporary files not getting deleted

mikaelnousiainen, 2025-08-12:
I seem to have some kind of an issue with indi-allsky *randomly* not deleting temporary files from the TMPDIR tmpfs ramdisk. The disk gets eventually full and indi-allsky stops working. A full OS restart - or cleanup of the dir + restart of indi-allsky - will fix the issue, of course.

I've recently (a week ago) updated to the latest indi-allsky and it didn't seem to help at all.

Most of the temporary files do get deleted, but many of the temperature sensor JSON files get left behind and sometimes the raw DNG images too - which fill up the disk fast.

Here's an example:

```
-rw-r--r-- 1 indiallsky indiallsky     2679 Aug 12 12:22 tmpsp9o6ung.json
-rw-r--r-- 1 indiallsky indiallsky     2680 Aug 12 12:22 tmp6a_w15xt.json
-rw-r--r-- 1 indiallsky indiallsky     2677 Aug 12 12:23 tmpv4ied7f3.json
-rw-r--r-- 1 indiallsky indiallsky 24805452 Aug 12 12:23 tmpruwuqkvm.dng
-rw-r--r-- 1 indiallsky indiallsky 24805452 Aug 12 12:23 tmp48l7vmd4.dng
-rw-r--r-- 1 indiallsky indiallsky 24805452 Aug 12 12:23 tmppa1r8xun.dng
-rw-r--r-- 1 indiallsky indiallsky 24805452 Aug 12 12:24 tmpjgn9h5u_.dng
-rw-r--r-- 1 indiallsky indiallsky 24805452 Aug 12 12:24 tmp3_qu76st.dng
-rw-r--r-- 1 indiallsky indiallsky 24805452 Aug 12 12:24 tmpbccdf4cq.dng
-rw-r--r-- 1 indiallsky indiallsky 24805452 Aug 12 12:25 tmp6_vb5dwk.dng
-rw-r--r-- 1 indiallsky indiallsky 19005440 Aug 12 12:25 tmpba6irzqa.dng
-rw-r--r-- 1 indiallsky indiallsky        0 Aug 12 12:25 tmp0f5zmgl_.json
-rw-r--r-- 1 indiallsky indiallsky        0 Aug 12 12:25 tmpxyroh205.json
-rw-r--r-- 1 indiallsky indiallsky        0 Aug 12 12:26 tmpuv18db2k.dng
-rw-r--r-- 1 indiallsky indiallsky        0 Aug 12 12:26 tmpmbqp0ujt.json
```

Have you seen this happen - what could be the reason?

mikaelnousiainen, 2025-08-12:
I've seen image queue backups happen (logged) only when/after the disk space on the temp volume runs out.

Here's the support info:

```
/home/indiallsky/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: raspbian
Release: 11
Arch: armv7l
Bits: 32

CPUs: 4
Memory: 1863952 kB

System: Raspberry Pi 4 Model B Rev 1.5

Linux vascam1 6.1.21-v7l+ #1642 SMP Mon Apr  3 17:22:30 BST 2023 armv7l GNU/Linux
Hostname
vascam1

Time
Wed 13 Aug 00:21:48 EEST 2025

System timezone
Europe/Helsinki

Uptime
 00:21:48 up 1 day,  6:15,  1 user,  load average: 1.17, 0.96, 0.93

Memory
               total        used        free      shared  buff/cache   available
Mem:         1863952      745024      331888       43800      787040      984736
Swap:              0           0           0

Filesystems
Filesystem     1K-blocks      Used Available Use% Mounted on
/dev/root      122962632 106363596  11576692  91% /
devtmpfs          668808         0    668808   0% /dev
tmpfs             931976       188    931788   1% /dev/shm
tmpfs             372792     43068    329724  12% /run
tmpfs               5120         4      5116   1% /run/lock
tmpfs             196608        28    196580   1% /opt/indi-allsky-tmp
/dev/mmcblk0p1    261108     51636    209472  20% /boot
tmpfs             186392         0    186392   0% /run/user/1002

systemctl runlevel
multi-user.target

sysctl info
vm.swappiness = 1

Thermal info
cpu-thermal
36511

system python: Python 3.9.2

indiserver: /usr/local/bin/indiserver

Locale info
LANG=en_GB.UTF-8
LANGUAGE=
LC_CTYPE="en_GB.UTF-8"
LC_NUMERIC="en_GB.UTF-8"
LC_TIME="en_GB.UTF-8"
[... 1516 lines cut ...]
Aug 13 00:19:09 vascam1 [ERROR] Image-1-27476/MainThread processing.update_astrometric_data() [2292]: HST next pass error: that satellite seems to stay always below your horizon
Aug 13 00:19:26 vascam1 [ERROR] Image-1-27476/MainThread processing.update_astrometric_data() [2292]: HST next pass error: that satellite seems to stay always below your horizon
Aug 13 00:19:44 vascam1 [ERROR] Image-1-27476/MainThread processing.update_astrometric_data() [2292]: HST next pass error: that satellite seems to stay always below your horizon
Aug 13 00:20:02 vascam1 [ERROR] Image-1-27476/MainThread processing.update_astrometric_data() [2292]: HST next pass error: that satellite seems to stay always below your horizon
Aug 13 00:20:19 vascam1 [ERROR] Image-1-27476/MainThread processing.update_astrometric_data() [2292]: HST next pass error: that satellite seems to stay always below your horizon
Aug 13 00:20:37 vascam1 [ERROR] Image-1-27476/MainThread processing.update_astrometric_data() [2292]: HST next pass error: that satellite seems to stay always below your horizon
Aug 13 00:20:54 vascam1 [ERROR] Image-1-27476/MainThread processing.update_astrometric_data() [2292]: HST next pass error: that satellite seems to stay always below your horizon
Aug 13 00:21:12 vascam1 [ERROR] Image-1-27476/MainThread processing.update_astrometric_data() [2292]: HST next pass error: that satellite seems to stay always below your horizon
Aug 13 00:21:29 vascam1 [ERROR] Image-1-27476/MainThread processing.update_astrometric_data() [2292]: HST next pass error: that satellite seems to stay always below your horizon
Aug 13 00:21:46 vascam1 [ERROR] Image-1-27476/MainThread processing.update_astrometric_data() [2292]: HST next pass error: that satellite seems to stay always below your horizon
```
starTrail errors
```
```
keogram errors
```
Aug 13 00:01:29 vascam1 [INFO] Image-1-27476/MainThread processing.realtimeKeogramDataSave() [3517]: Storing realtime keogram data
Aug 13 00:08:45 vascam1 [INFO] Image-1-27476/MainThread processing.realtimeKeogramDataSave() [3517]: Storing realtime keogram data
Aug 13 00:16:02 vascam1 [INFO] Image-1-27476/MainThread processing.realtimeKeogramDataSave() [3517]: Storing realtime keogram data
```

#################################
###     end support info      ###
#################################
``

mikaelnousiainen, 2025-08-12:
Also just want to point out that the temporary JSON files from the temperature reading script seem to get left out almost every time the script is executed now. No errors/warnings logged about anything else than the satellite pass.

aaronwmorris, 2025-08-14:
I found a condition where the temp JSON files would not be deleted if the script exited abnormally.

Merged #2304 to fix.

mikaelnousiainen, 2025-08-14:
@aaronwmorris Cool, I'll give it a try.

Could there be a similar condition for the temporary image (.dng) files not getting deleted? Those are the ones filling the disk fast.

mikaelnousiainen, 2025-08-14:
Not sure how I should read this: (the long period there on the 3rd entry was a reboot after I pulled the latest changes)

```
+-------+---------------------+------------+---------+--------+--------+------------+
|   ID  |         Date        |  Exposure  | Elapsed | Delta  | Period | Processing |
+-------+---------------------+------------+---------+--------+--------+------------+
| 76110 | 2025-08-14 23:45:13 | 15.0000000 |  17.42  | +2.424 |   17   |    6.06    |
| 76109 | 2025-08-14 23:44:56 | 15.0000000 |  17.36  | +2.358 |   18   |    6.05    |
| 76108 | 2025-08-14 23:44:38 | 15.0000000 |  18.22  | +3.216 |  183   |    8.46    |
| 76107 | 2025-08-14 23:41:35 | 15.0000000 |  17.36  | +2.360 |   17   |    6.56    |
| 76106 | 2025-08-14 23:41:18 | 15.0000000 |  17.45  | +2.449 |   18   |    5.88    |
| 76105 | 2025-08-14 23:41:00 | 15.0000000 |  17.52  | +2.522 |   17   |    5.88    |
| 76104 | 2025-08-14 23:40:43 | 15.0000000 |  17.49  | +2.486 |   18   |    5.82    |
| 76103 | 2025-08-14 23:40:25 | 15.0000000 |  17.39  | +2.387 |   17   |    5.82    |
| 76102 | 2025-08-14 23:40:08 | 15.0000000 |  17.49  | +2.487 |   17   |    5.82    |
| 76101 | 2025-08-14 23:39:51 | 15.0000000 |  17.40  | +2.404 |   18   |    6.23    |
| 76100 | 2025-08-14 23:39:33 | 15.0000000 |  17.41  | +2.411 |   17   |    5.88    |
| 76099 | 2025-08-14 23:39:16 | 15.0000000 |  17.43  | +2.429 |   18   |    5.85    |
| 76098 | 2025-08-14 23:38:58 | 15.0000000 |  17.43  | +2.432 |   17   |    5.87    |
| 76097 | 2025-08-14 23:38:41 | 15.0000000 |  17.33  | +2.329 |   18   |    5.85    |
| 76096 | 2025-08-14 23:38:23 | 15.0000000 |  17.53  | +2.534 |   17   |    5.90    |
``

aaronwmorris, 2025-08-14:
The definitions are:

* Exposure - exposure :-)
* Delta - Difference between the amount of time the exposure required to execute and the requested exposure.  This generally tells you how much time it takes to download the exposure from the camera once it completed.
* Period - The number of seconds between the start of one exposure and the previous one (integer, therefore whole seconds)
* Processing - Amount of time the processing pipeline spent processing the image (stretching, rotating, star detection, etc)

Lets try this for some additional details:

```
sqlite3 -header -table /var/lib/indi-allsky/indi-allsky.sqlite "SELECT id,exposure,process_elapsed FROM image WHERE process_elapsed > 10.0 LIMIT 30;"
```

mikaelnousiainen, 2025-08-14:
Looks like there are only a handful of these:
```
$ sqlite3 -header -table /var/lib/indi-allsky/indi-allsky.sqlite "SELECT id,exposure,process_elapsed FROM image WHERE process_elapsed > 10.0 LIMIT 30;"
+-------+----------------------+------------------+
|  id   |       exposure       | process_elapsed  |
+-------+----------------------+------------------+
| 26007 | 0.00121034414041787  | 10.0267586708069 |
| 28175 | 0.000809002202004194 | 10.3011956214905 |
| 28176 | 0.000809002202004194 | 10.4879331588745 |
| 35220 | 7.75466442108154     | 20.0413475036621 |
| 35222 | 9.17384910583496     | 19.5464754104614 |
| 35223 | 14.2067384719849     | 24.8046326637268 |
| 35224 | 14.2067384719849     | 11.1833469867706 |
| 35225 | 15.0                 | 11.3837237358093 |
| 35227 | 15.0                 | 11.4884641170502 |
| 35228 | 15.0                 | 11.7733154296875 |
| 35230 | 15.0                 | 10.3501143455505 |
| 35231 | 15.0                 | 10.5523021221161 |
| 35607 | 15.0                 | 13.4998214244843 |
| 35875 | 1.78470528125763     | 14.7766499519348 |
| 37969 | 0.000742634641937912 | 10.6888341903687 |
| 38714 | 0.000412247580243275 | 10.9559996128082 |
| 42964 | 0.000814242288470268 | 10.2900466918945 |
| 44172 | 0.000404573511332274 | 10.5843729972839 |
| 55603 | 0.000963164959102869 | 10.2429738044739 |
| 56871 | 0.000479029200505465 | 10.5065004825592 |
| 58076 | 0.000601277803070843 | 10.6618361473084 |
| 58539 | 0.025447828695178    | 10.9061236381531 |
| 61724 | 0.0281017124652863   | 13.3408524990082 |
| 65449 | 0.000847337942104787 | 10.4600665569305 |
| 69674 | 0.174516901373863    | 10.2254383563995 |
+-------+----------------------+------------------+
```

aaronwmorris, 2025-08-14:
It does help.  Those are not related to the external temp script.  It is something else.

mikaelnousiainen, 2025-08-17:
It's odd, the issue keeps repeating now more often after latest upgrade. Previously I was using version 2024.05.1 and I had to clean up files / restart maybe once a week. Now the disk gets full every day. Additionally, keograms and timelapses don't get generated anymore - I just see a message they start to get generated and nothing more - no errors, nothing.

I wonder if everything is related somehow? I followed the upgrade process and ran the setup again after the latest upgrade last week.

mikaelnousiainen, 2025-08-17:
Interesting, I just went to check for the timelapse/keogram generation messages and for whatever reason NOW I see another error:

```
Aug 17 00:06:21 vascam1 [ERROR] MainProcess-916/MainThread allsky.unhandled_exception() [36]: An uncaught exception occurred:
Aug 17 00:06:21 vascam1 [ERROR] MainProcess-916/MainThread allsky.unhandled_exception() [37]: Type: <class 'ModuleNotFoundError'>
Aug 17 00:06:21 vascam1 [ERROR] MainProcess-916/MainThread allsky.unhandled_exception() [38]: Value: No module named 'simplejpeg'
Aug 17 00:06:21 vascam1 [ERROR] MainProcess-916/MainThread allsky.unhandled_exception() [43]: '  File "/home/indiallsky/indi-allsky/allsky.py", line 105, in <module>\n    action_func(*args_list, **kwargs_dict)\n'
Aug 17 00:06:21 vascam1 [ERROR] MainProcess-916/MainThread allsky.unhandled_exception() [43]: '  File "/home/indiallsky/indi-allsky/indi_allsky/allsky.py", line 711, in run\n    self._startVideoWorker()\n'
Aug 17 00:06:21 vascam1 [ERROR] MainProcess-916/MainThread allsky.unhandled_exception() [43]: '  File "/home/indiallsky/indi-allsky/indi_allsky/allsky.py", line 470, in _startVideoWorker\n    from .video import VideoWorker\n'
Aug 17 00:06:21 vascam1 [ERROR] MainProcess-916/MainThread allsky.unhandled_exception() [43]: '  File "/home/indiallsky/indi-allsky/indi_allsky/video.py", line 22, in <module>\n    from .timelapse import TimelapseGenerator\n'
Aug 17 00:06:21 vascam1 [ERROR] MainProcess-916/MainThread allsky.unhandled_exception() [43]: '  File "/home/indiallsky/indi-allsky/indi_allsky/timelapse.py", line 7, in <module>\n    from . import timelapse_preprocessor\n'
Aug 17 00:06:21 vascam1 [ERROR] MainProcess-916/MainThread allsky.unhandled_exception() [43]: '  File "/home/indiallsky/indi-allsky/indi_allsky/timelapse_preprocessor/__init__.py", line 2, in <module>\n    from .preProcessorWrapKeogram import PreProcessorWrapKeogram as wrap_keogram\n'
Aug 17 00:06:21 vascam1 [ERROR] MainProcess-916/MainThread allsky.unhandled_exception() [43]: '  File "/home/indiallsky/indi-allsky/indi_allsky/timelapse_preprocessor/preProcessorWrapKeogram.py", line 7, in <module>\n    import simplejpeg\n'
```

This might be the reason for at least some of my issues... I wonder why the module doesn't get installed?

aaronwmorris, 2025-08-17:
Looks like I did miss the simplejpeg requirement in some of the python requirements files.  Merged #2321 to fix.

You can manually fix this by running the following (or just rerunning setup.sh)

```
source virtualenv/indi-allsky/bin/activate

pip3 install simplejpeg
```

mikaelnousiainen, 2025-08-17:
Thanks. However, of course it can't be this easy. I'm running Raspberry Pi OS based on Debian 10 Bullseye  (though with latest updates) and simplejpeg depenendency for libjpeg-turbo fails.

Is Debian 10 still supported?

Anyway, it looks like they have fixed a similar issue in the past (see https://github.com/libjpeg-turbo/libjpeg-turbo/issues/523#event-12141962595), but somehow the build is still failing because of compiler options regarding floating point handling / ARM NEON instructions: 

```
      -- The C compiler identification is GNU 10.2.1
      -- Detecting C compiler ABI info
      -- Detecting C compiler ABI info - done
      -- Check for working C compiler: /usr/bin/cc - skipped
      -- Detecting C compile features
      -- Detecting C compile features - done
      -- CMAKE_BUILD_TYPE = Release
      -- VERSION = 3.1.0, BUILD = 20250817
      -- 32-bit build (arm)
      -- CMAKE_INSTALL_PREFIX = /opt/libjpeg-turbo
      -- CMAKE_INSTALL_BINDIR = bin (/opt/libjpeg-turbo/bin)
      -- CMAKE_INSTALL_DATAROOTDIR =  (/opt/libjpeg-turbo)
      -- CMAKE_INSTALL_DOCDIR = doc (/opt/libjpeg-turbo/doc)
      -- CMAKE_INSTALL_INCLUDEDIR = include (/opt/libjpeg-turbo/include)
      -- CMAKE_INSTALL_LIBDIR = lib32 (/opt/libjpeg-turbo/lib32)
      -- CMAKE_INSTALL_MANDIR = man (/opt/libjpeg-turbo/man)
      -- Shared libraries disabled (ENABLE_SHARED = 0)
      -- Static libraries enabled (ENABLE_STATIC = 1)
      -- Arithmetic decoding support enabled (WITH_ARITH_DEC = 1)
      -- Arithmetic encoding support enabled (WITH_ARITH_ENC = 1)
      -- TurboJPEG API library enabled (WITH_TURBOJPEG = 1)
      -- TurboJPEG Java wrapper disabled (WITH_JAVA = 0)
      -- Emulating libjpeg API/ABI v6.2 (WITH_JPEG7 = 0, WITH_JPEG8 = 0)
      -- libjpeg API shared library version = 62.4.0
      -- Compiler flags = -flto -O3 -DNDEBUG
      -- Linker flags = -flto -Wl,--strip-all,--exclude-libs,ALL,--gc-sections
      -- Looking for sys/types.h
      -- Looking for sys/types.h - found
      -- Looking for stdint.h
      -- Looking for stdint.h - found
      -- Looking for stddef.h
      -- Looking for stddef.h - found
      -- Check size of size_t
      -- Check size of size_t - done
      -- Check size of unsigned long
      -- Check size of unsigned long - done
      -- Performing Test HAVE_BUILTIN_CTZL
      -- Performing Test HAVE_BUILTIN_CTZL - Success
      -- Performing Test RIGHT_SHIFT_IS_UNSIGNED
      -- Performing Test RIGHT_SHIFT_IS_UNSIGNED - Failed
      -- Performing Test HIDDEN_WORKS
      -- Performing Test HIDDEN_WORKS - Success
      -- HIDDEN = __attribute__((visibility("hidden")))
      -- Performing Test INLINE_WORKS
      -- Performing Test INLINE_WORKS - Success
      -- INLINE = __inline__ __attribute__((always_inline)) (FORCE_INLINE = 1)
      -- Performing Test HAVE_THREAD_LOCAL
      -- Performing Test HAVE_THREAD_LOCAL - Success
      -- THREAD_LOCAL = __thread
      -- Performing Test HAVE_VERSION_SCRIPT
      -- Performing Test HAVE_VERSION_SCRIPT - Success
      -- Linker supports GNU-style version scripts
      -- CMAKE_EXECUTABLE_SUFFIX =
      -- Performing Test NEED_SOFTFP_FOR_INTRINSICS
      -- Performing Test NEED_SOFTFP_FOR_INTRINSICS - Failed
      -- Performing Test HAVE_NEON
      -- Performing Test HAVE_NEON - Success
      -- Performing Test HAVE_VLD1_S16_X3
      -- Performing Test HAVE_VLD1_S16_X3 - Failed
      -- Performing Test HAVE_VLD1_U16_X2
      -- Performing Test HAVE_VLD1_U16_X2 - Failed
      -- Performing Test HAVE_VLD1Q_U8_X4
      -- Performing Test HAVE_VLD1Q_U8_X4 - Failed
[... 211 lines cut ...]
          cmd_obj.run()
        File "/tmp/pip-build-env-tt676rnr/overlay/lib/python3.9/site-packages/setuptools/_distutils/command/build.py", line 135, in run
          self.run_command(cmd_name)
        File "/tmp/pip-build-env-tt676rnr/overlay/lib/python3.9/site-packages/setuptools/_distutils/cmd.py", line 357, in run_command
          self.distribution.run_command(command)
        File "/tmp/pip-build-env-tt676rnr/overlay/lib/python3.9/site-packages/setuptools/dist.py", line 1102, in run_command
          super().run_command(command)
        File "/tmp/pip-build-env-tt676rnr/overlay/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 1021, in run_command
          cmd_obj.run()
        File "<string>", line 141, in run
        File "<string>", line 179, in build_cmake_dependencies
        File "<string>", line 203, in build_cmake_dependency
        File "/usr/lib/python3.9/subprocess.py", line 373, in check_call
          raise CalledProcessError(retcode, cmd)
      subprocess.CalledProcessError: Command '['cmake', '--build', '.', '--config', 'Release']' returned non-zero exit status 2.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for simplejpeg
Failed to build simplejpeg
error: failed-wheel-build-for-install

× Failed to build installable wheels for some pyproject.toml based projects
╰─> simplejpeg
```

mikaelnousiainen, 2025-08-19:
Ok, would the condition where the satellite is never visible affect the issue? Why is this condition logged as an ERROR anyway:

```
Aug 18 09:55:32 vascam1 [ERROR] Image-2-1365/MainThread processing.update_astrometric_data() [2319]: TIANGONG next pass error: that satellite seems to stay always below your horizon
```

aaronwmorris, 2025-08-20:
No, this is a common condition that should be caught by the current code.

mikaelnousiainen, 2025-08-20:
Also, I found yet another case where the image metadata JSON temporary file gets left behind - when an AWS S3 upload fails with a timeout:

```
Aug 20 16:30:02 vascam1 [ERROR] MainProcess-951/MainThread allsky._fileUploadWorkerStart() [599]: Upload worker exception: Traceback (most recent call last):
Aug 20 16:30:02 vascam1 [ERROR] MainProcess-951/MainThread allsky._fileUploadWorkerStart() [599]: Upload worker exception:   File "/home/indiallsky/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/urllib3/connectionpool.py", line 714, in urlopen
Aug 20 16:30:02 vascam1 [ERROR] MainProcess-951/MainThread allsky._fileUploadWorkerStart() [599]: Upload worker exception:     httplib_response = self._make_request(
Aug 20 16:30:02 vascam1 [ERROR] MainProcess-951/MainThread allsky._fileUploadWorkerStart() [599]: Upload worker exception:   File "/home/indiallsky/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/urllib3/connectionpool.py", line 415, in _make_request
Aug 20 16:30:02 vascam1 [ERROR] MainProcess-951/MainThread allsky._fileUploadWorkerStart() [599]: Upload worker exception:     conn.request(method, url, **httplib_request_kw)
Aug 20 16:30:02 vascam1 [ERROR] MainProcess-951/MainThread allsky._fileUploadWorkerStart() [599]: Upload worker exception:   File "/home/indiallsky/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/botocore/awsrequest.py", line 96, in request
Aug 20 16:30:02 vascam1 [ERROR] MainProcess-951/MainThread allsky._fileUploadWorkerStart() [599]: Upload worker exception:     rval = super().request(method, url, body, headers, *args, **kwargs)
Aug 20 16:30:02 vascam1 [ERROR] MainProcess-951/MainThread allsky._fileUploadWorkerStart() [599]: Upload worker exception:   File "/home/indiallsky/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/urllib3/connection.py", line 244, in request
Aug 20 16:30:02 vascam1 [ERROR] MainProcess-951/MainThread allsky._fileUploadWorkerStart() [599]: Upload worker exception:     super(HTTPConnection, self).request(method, url, body=body, headers=headers)
Aug 20 16:30:02 vascam1 [ERROR] MainProcess-951/MainThread allsky._fileUploadWorkerStart() [599]: Upload worker exception:   File "/usr/lib/python3.9/http/client.py", line 1259, in request
Aug 20 16:30:02 vascam1 [ERROR] MainProcess-951/MainThread allsky._fileUploadWorkerStart() [599]: Upload worker exception:     self._send_request(method, url, body, headers, encode_chunked)
Aug 20 16:30:02 vascam1 [ERROR] MainProcess-951/MainThread allsky._fileUploadWorkerStart() [599]: Upload worker exception:   File "/usr/lib/python3.9/http/client.py", line 1305, in _send_request
Aug 20 16:30:02 vascam1 [ERROR] MainProcess-951/MainThread allsky._fileUploadWorkerStart() [599]: Upload worker exception:     self.endheaders(body, encode_chunked=encode_chunked)
Aug 20 16:30:02 vascam1 [ERROR] MainProcess-951/MainThread allsky._fileUploadWorkerStart() [599]: Upload worker exception:   File "/usr/lib/python3.9/http/client.py", line 1254, in endheaders
Aug 20 16:30:02 vascam1 [ERROR] MainProcess-951/MainThread allsky._fileUploadWorkerStart() [599]: Upload worker exception:     self._send_output(message_body, encode_chunked=encode_chunked)
Aug 20 16:30:02 vascam1 [ERROR] MainProcess-951/MainThread allsky._fileUploadWorkerStart() [599]: Upload worker exception:   File "/home/indiallsky/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/botocore/awsrequest.py", line 130, in _send_output
Aug 20 16:30:02 vascam1 [ERROR] MainProcess-951/MainThread allsky._fileUploadWorkerStart() [599]: Upload worker exception:     self._handle_expect_response(message_body)
Aug 20 16:30:02 vascam1 [ERROR] MainProcess-951/MainThread allsky._fileUploadWorkerStart() [599]: Upload worker exception:   File "/home/indiallsky/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/botocore/awsrequest.py", line 176, in _handle_expect_response
Aug 20 16:30:02 vascam1 [ERROR] MainProcess-951/MainThread allsky._fileUploadWorkerStart() [599]: Upload worker exception:     self._send_message_body(message_body)
Aug 20 16:30:02 vascam1 [ERROR] MainProcess-951/MainThread allsky._fileUploadWorkerStart() [599]: Upload worker exception:   File "/home/indiallsky/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/botocore/awsrequest.py", line 209, in _send_message_body
Aug 20 16:30:02 vascam1 [ERROR] MainProcess-951/MainThread allsky._fileUploadWorkerStart() [599]: Upload worker exception:     self.send(message_body)
Aug 20 16:30:02 vascam1 [ERROR] MainProcess-951/MainThread allsky._fileUploadWorkerStart() [599]: Upload worker exception:   File "/home/indiallsky/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/botocore/awsrequest.py", line 223, in send
Aug 20 16:30:02 vascam1 [ERROR] MainProcess-951/MainThread allsky._fileUploadWorkerStart() [599]: Upload worker exception:     return super().send(str)
Aug 20 16:30:02 vascam1 [ERROR] MainProcess-951/MainThread allsky._fileUploadWorkerStart() [599]: Upload worker exception:   File "/usr/lib/python3.9/http/client.py", line 972, in send
Aug 20 16:30:02 vascam1 [ERROR] MainProcess-951/MainThread allsky._fileUploadWorkerStart() [599]: Upload worker exception:     self.sock.sendall(datablock)
Aug 20 16:30:02 vascam1 [ERROR] MainProcess-951/MainThread allsky._fileUploadWorkerStart() [599]: Upload worker exception:   File "/usr/lib/python3.9/ssl.py", line 1237, in sendall
[... cut ...]
Aug 20 16:30:02 vascam1 [ERROR] MainProcess-951/MainThread allsky._fileUploadWorkerStart() [599]: Upload worker exception:     raise ConnectionClosedError(
Aug 20 16:30:02 vascam1 [ERROR] MainProcess-951/MainThread allsky._fileUploadWorkerStart() [599]: Upload worker exception: botocore.exceptions.ConnectionClosedError: Connection was closed before we received a valid response from endpoint URL: "https://xxxxxxxx.s3.eu-central-1.amazonaws.com/ccd_xxx/exposures/20250820/day/20_16/ccd1_20250820_162937.jpg".
Aug 20 16:30:02 vascam1 [ERROR] MainProcess-951/MainThread allsky._fileUploadWorkerStart() [599]: Upload worker exception: 
```

aaronwmorris, 2025-08-20:
Merged #2327 to fix the next_pass() exceptions.

aaronwmorris, 2025-08-20:
Merged #2328 to catch some additional exceptions and ensure files are cleaned up on a filetransfer exception.

mikaelnousiainen, 2025-08-20:
Great, thanks! I've just updated my indi-allsky installation with your latest changes - hope the issues are finally gone!

aaronwmorris, 2025-08-22:
Thank you for hanging in there!