aaronwmorris/indi-allsky #2311: interactive authentication required. - Debian Proxmox VM

morganjrees, 2025-08-14:
org.freedesktop.DBus.Error.InteractiveAuthorizationRequired: Interactive authentication required.

Hi awesome people im getting this error when trying to reboot my all sky. Ive hosted this I a VM with Proxmox running Debian, it may be some sort of sudo issue, my user account running indi-allsky is in the sudoers list. and im able to sudo in terminal under this account and root account.

Any ideas would be appreciated... 

[image]

morganjrees, 2025-10-29:
root@allsky:~# sudo find /etc/polkit-1/
/etc/polkit-1/
/etc/polkit-1/rules.d
/etc/polkit-1/rules.d/90-indi-allsky.rules
root@allsky:~#

aaronwmorris, 2025-11-03:
100% memory utilization is a fairly normal condition.  If it is hanging, I would find it hard to believe it is due to memory, unless something other than indi-allsky is running there.

Do you receive any error messages  in the log when it locks up?