aaronwmorris/indi-allsky #2316: indi passive

sonny486, 2025-08-16:
How do I get indi passive to get images from one of my other all-sky servers?

Let me know.

Thanks

aaronwmorris, 2025-08-17:
All you have to do is change the `INDI Server` on the Camera tab to be the hostname or IP address of the indiserver for the other indi-allsky instance.

sonny486, 2025-08-17:
Yeah I have done that, and it didn't do anything.  I tried the hostname and IP and the interface on the client shows down.

the server is a working latest build indi-allsky and it is working well.

sonny486, 2025-08-17:
[image]
[image]
[image]

sonny486, 2025-08-17:
Same result with Ubuntu 22.04 with indi 2.1.5

sonny486, 2025-08-17:
TypeError: IndiClientPassive.__init__() missing 4 required positional arguments: 'dec_v', 'gain_v', 'bin_v', and 'night_v'

sonny486, 2025-08-17:
Aug 16 22:10:11 timelapse [INFO] MainProcess-3062/MainThread allsky._startCaptureWorker() [367]: Starting Capture-41 worker
Aug 16 22:10:11 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception: 
Aug 16 22:10:11 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception: TypeError: IndiClientPassive.__init__() missing 4 required positional arguments: 'dec_v', 'gain_v', 'bin_v', and 'night_v'
Aug 16 22:10:11 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception:                       ^^^^^^^^^^^^^^^^^
Aug 16 22:10:11 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception:     self.indiclient = camera_interface(
Aug 16 22:10:11 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception:   File "/home/orangepi/indi-allsky/indi_allsky/capture.py", line 740, in _initialize
Aug 16 22:10:11 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception:     self._initialize()
Aug 16 22:10:11 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception:   File "/home/orangepi/indi-allsky/indi_allsky/capture.py", line 328, in saferun
Aug 16 22:10:11 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception:     self.saferun()
Aug 16 22:10:11 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception:   File "/home/orangepi/indi-allsky/indi_allsky/capture.py", line 316, in run
Aug 16 22:10:11 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception: Traceback (most recent call last):
Aug 16 22:09:58 timelapse [INFO] MainProcess-3062/MainThread allsky._startCaptureWorker() [367]: Starting Capture-40 worker
Aug 16 22:09:58 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception: 
Aug 16 22:09:58 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception: TypeError: IndiClientPassive.__init__() missing 4 required positional arguments: 'dec_v', 'gain_v', 'bin_v', and 'night_v'
Aug 16 22:09:58 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception:                       ^^^^^^^^^^^^^^^^^
Aug 16 22:09:58 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception:     self.indiclient = camera_interface(
Aug 16 22:09:58 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception:   File "/home/orangepi/indi-allsky/indi_allsky/capture.py", line 740, in _initialize
Aug 16 22:09:58 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception:     self._initialize()
Aug 16 22:09:58 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception:   File "/home/orangepi/indi-allsky/indi_allsky/capture.py", line 328, in saferun
Aug 16 22:09:58 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception:     self.saferun()
Aug 16 22:09:58 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception:   File "/home/orangepi/indi-allsky/indi_allsky/capture.py", line 316, in run
Aug 16 22:09:58 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception: Traceback (most recent call last):
Aug 16 22:09:45 timelapse [INFO] MainProcess-3062/MainThread allsky._startCaptureWorker() [367]: Starting Capture-39 worker
Aug 16 22:09:45 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception: 
Aug 16 22:09:45 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception: TypeError: IndiClientPassive.__init__() missing 4 required positional arguments: 'dec_v', 'gain_v', 'bin_v', and 'night_v'
Aug 16 22:09:45 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception:                       ^^^^^^^^^^^^^^^^^
Aug 16 22:09:45 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception:     self.indiclient = camera_interface(
Aug 16 22:09:45 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception:   File "/home/orangepi/indi-allsky/indi_allsky/capture.py", line 740, in _initialize
Aug 16 22:09:45 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception:     self._initialize()
Aug 16 22:09:45 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception:   File "/home/orangepi/indi-allsky/indi_allsky/capture.py", line 328, in saferun
Aug 16 22:09:45 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception:     self.saferun()
[... cut ...]
Aug 16 22:09:45 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception: Traceback (most recent call last):
Aug 16 22:09:32 timelapse [INFO] MainProcess-3062/MainThread allsky._startCaptureWorker() [367]: Starting Capture-38 worker
Aug 16 22:09:32 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception: 
Aug 16 22:09:32 timelapse [ERROR] MainProcess-3062/MainThread allsky._startCaptureWorker() [360]: Capture worker exception: TypeError: IndiClientPassive.__init__() missing 4 required positional arguments: 'dec_v', 'gain_v', 'bin_v', and 'night_v'

aaronwmorris, 2025-08-17:
Looks like I refactored the camera classes a while back and just forgot about the passive camera.

Merged #2319 .  Let me know how that works.

sonny486, 2025-08-17:
It's getting exposures now, but not doing anything with them.

Aug 17 15:23:24 timelapse [INFO] Capture-16-19952/MainThread indi.getCcdTemperature() [892]: Camera temperature: 0.0
Aug 17 15:23:24 timelapse [INFO] Capture-16-19952/MainThread capture.detectNight() [1719]: Sun alt: 61.8, Moon alt: 3.7, phase 31.8%
Aug 17 15:23:24 timelapse [INFO] Capture-16-19952/MainThread capture.saferun() [372]: Exposure state: IDLE
Aug 17 15:23:24 timelapse [INFO] Capture-16-19952/MainThread capture.saferun() [371]: Camera last ready: 0.7s
Aug 17 15:23:23 timelapse [INFO] Capture-16-19952/MainThread capture.saferun() [605]: Exposure received in 0.1516s (+0.1416s)
Aug 17 15:23:23 timelapse [INFO] Capture-16-19952/MainThread capture.saferun() [728]: Total time since last exposure 15.0333 s
Aug 17 15:23:23 timelapse [INFO] Capture-16-19952/MainThread capture.shoot() [1914]: Taking 0.01000001 s exposure (gain 1)
Aug 17 15:23:14 timelapse [WARNING] MainProcess-19745/MainThread allsky._queueManualTasks() [1289]: Reload initiated
Aug 17 15:23:14 timelapse [INFO] MainProcess-19745/MainThread allsky._queueManualTasks() [1279]: Picked up MAIN task
Aug 17 15:23:13 timelapse [INFO] Capture-16-19952/MainThread indi.getCcdTemperature() [892]: Camera temperature: 0.0
Aug 17 15:23:12 timelapse [INFO] Capture-16-19952/MainThread capture.detectNight() [1719]: Sun alt: 61.8, Moon alt: 3.8, phase 31.8%
Aug 17 15:23:12 timelapse [INFO] Capture-16-19952/MainThread capture.saferun() [372]: Exposure state: IDLE
Aug 17 15:23:12 timelapse [INFO] Capture-16-19952/MainThread capture.saferun() [371]: Camera last ready: 4.5s
Aug 17 15:23:08 timelapse [INFO] Capture-16-19952/MainThread capture.saferun() [605]: Exposure received in 0.1515s (+0.1415s)
Aug 17 15:23:08 timelapse [INFO] Capture-16-19952/MainThread capture.saferun() [728]: Total time since last exposure 15.0309 s
Aug 17 15:23:08 timelapse [INFO] Capture-16-19952/MainThread capture.shoot() [1914]: Taking 0.01000001 s exposure (gain 1)
Aug 17 15:23:01 timelapse [INFO] Capture-16-19952/MainThread indi.getCcdTemperature() [892]: Camera temperature: 0.0
Aug 17 15:23:01 timelapse [INFO] Capture-16-19952/MainThread capture.detectNight() [1719]: Sun alt: 61.8, Moon alt: 3.8, phase 31.8%
Aug 17 15:23:01 timelapse [INFO] Capture-16-19952/MainThread capture.saferun() [372]: Exposure state: IDLE
Aug 17 15:23:01 timelapse [INFO] Capture-16-19952/MainThread capture.saferun() [371]: Camera last ready: 8.3s
Aug 17 15:22:53 timelapse [INFO] Capture-16-19952/MainThread capture.saferun() [605]: Exposure received in 0.1516s (+0.1416s)
Aug 17 15:22:53 timelapse [INFO] Capture-16-19952/MainThread capture.saferun() [728]: Total time since last exposure 15.0369 s
Aug 17 15:22:53 timelapse [INFO] Capture-16-19952/MainThread capture.shoot() [1914]: Taking 0.01000001 s exposure (gain 1)
Aug 17 15:22:50 timelapse [INFO] Capture-16-19952/MainThread indi.getCcdTemperature() [892]: Camera temperature: 0.0
Aug 17 15:22:50 timelapse [INFO] Capture-16-19952/MainThread capture.detectNight() [1719]: Sun alt: 61.9, Moon alt: 3.8, phase 31.8%
Aug 17 15:22:50 timelapse [INFO] Capture-16-19952/MainThread capture.saferun() [372]: Exposure state: IDLE
Aug 17 15:22:50 timelapse [INFO] Capture-16-19952/MainThread capture.saferun() [371]: Camera last ready: 12.1s
Aug 17 15:22:39 timelapse [INFO] Capture-16-19952/MainThread indi.getCcdTemperature() [892]: Camera temperature: 0.0
Aug 17 15:22:39 timelapse [INFO] Capture-16-19952/MainThread capture.detectNight() [1719]: Sun alt: 61.9, Moon alt: 3.9, phase 31.8%

sonny486, 2025-08-17:
Actually it looks like it's in some kind of simulator mode.  I have the primary indi system taking exposures every 20 seconds, however this one is taking photos every 15 seconds.

aaronwmorris, 2025-08-19:
I will track this down, just give me a bit of time.

sonny486, 2025-08-19:
No worries, thanks man!

sonny486, 2025-08-19:
AHHHH that makes perfect sense!  Ok I will rebuild it and let you know.

Thanks a bunch!  Paypal on the way

aaronwmorris, 2025-08-20:
indi_pylibcamera actually works fairly well if the in-repo driver cannot be bulit.  scriptorron implemented a full indi server/driver in python using pyindi-client.  He may be psychotic.  :-)  It uses the picamera2 python interface for libcamera.

The `make install` command should install it, but if you have to manually install, try `/usr/local/bin/`.  That is where the indi setup scripts will look for it if you used my build script previously.

sonny486, 2025-08-20:
Yeah I got it to work!  Building the indi_libcamera from the 3rdparty directly worked for me.

I am up and running. 

I have an orangepi 5 with 8 cores rendering the videos now (so I can render them much quicker than RPI5)

closed 2025-08-20 by sonny486