aaronwmorris/indi-allsky #2387: Dew Heater "Invert Output" ignored.

illigtr, 2025-09-18:
Please see discussion [here](https://github.com/aaronwmorris/indi-allsky/discussions/2377)

The invert output has no effect on either software or hardware PWM. I have an inverse logic dew heater (HIGH = OFF) built around an 817 opto-coupler and a MOSFET-N driving 10 watt resistors (32 ohms @ 12 VDC). When the dew heater should be off, it remains ON all the time, regardless of invert output, regardless of PWM type.

Finally did this as a work around....which works.

[image]

illigtr, 2025-09-20:
@aaronwmorris I looked at the new code, but I think IHMO this doesn't solve the issue. I noticed you kept the _state = -1 for each of the variants of dew heater. I like that...

but in sensor.py -->  init_dew_heater() you added 

```python
 # set initial state
  self.dew_heater.state = 0
```
This basically puts the problem back.

The problem now is that when set_dew_heater() is finally called, it is going to compare state 0 with default value (from GUI), also 0.... so an inverse logic heater will stay ON. set_dew_heater() requires the new state to be different from the current state to actually change the PWM or GPIO high/low.

Am I missing something?

aaronwmorris, 2025-09-20:
The real issue here was that the internal state variable `self._state` did not match the hardware pin state when inverse mode was active.

By setting `self.dew_heater.state = 0` this forces the variable and the pin state to be synchronized.  The logic for the inverted mode is in the `state` setter.  No comparison of the states are performed when setting the `state` property explicitly.

You may wonder why I even perform a comparison of the variable instead of just setting the pin state every time... When I looked at the PWM signal on an oscilloscope, I noticed there was jitter every time the PWM state was set, even if the same PWM setting was applied to the pin that was already set.

illigtr, 2025-09-20:
@aaronwmorris Ok. I saw the inverse logic in setter. However, the only time the inverse logic of setter will come into action is when set_dew_heater() is called *and* the states are different. I guess I'll push the new update to my device and check the result and report back.

illigtr, 2025-09-20:
Ok. I can confirm the result is good. I put back my modded line to who PWM value and it is set to 65535 at initialisation. 
```
pi@indi-allsky:~ $ grep "Set dew heater" /var/log/indi-allsky/indi-allsky.log | tail -n 100
2025-09-20T11:08:39.735538-04:00 indi-allsky [WARNING] Sensor-1-1166/MainThread dewHeaterPwm.state() [60]: Set dew heater state: 0%, PWM:65535
'''

Thank you for fixing this.

closed 2025-09-20 by illigtr